openwrt/openwrt #10230: 22.03 dnsmasq 'leaks' IPv6 addresses.

Mijzelf, 2022-07-09:
Have been using a dns based ad blocker for years. Dnsmasq loads a file as generated by https://pgl.yoyo.org/adservers/serverlist.php?hostformat=dnsmasq, containing 
```
address=/1-1ads.com/192.168.1.1
address=/101com.com/192.168.1.1
address=/101order.com/192.168.1.1
address=/123freeavatars.com/192.168.1.1
address=/180hits.de/192.168.1.1
address=/180searchassistant.com/192.168.1.1
... 
```
This worked fine. After installing 22.03-rc4 (on my ZyXEL P-2812-HNU-F1) suddenly I saw ads where I didn't see them before. It turns out that dnsmasq gives now the overruled ipv4 address of the ad domain, but also a real ipv6 address. 
I suppose this is a bug. When I add a domain like this, dnsmasq shouldn't upstream the request, and so dnsmasq shouldn't be able to offer the ipv6 address.

I found 2 work-arounds: 
- Also add an ipv6 address for each domain. This seriously adds to dnsmasq's memory usage.
- Add the domains as NXDOMAIN:
```
address=/1-1ads.com/#
address=/101com.com/#
```
This also works, but I don't know if it at client side always has the same effect. (a NXDOMAIN on a DNS request vs a 404 on an HTTP request)

dave14305, 2022-07-09:
You should probably consult the dnsmasq man page, then report upstream if you still think it’s a bug.

dave14305, 2022-07-10:
From the dnsmasq CHANGELOG:
```
Note in manpage the change in behaviour of -address. This behaviour
actually changed in v2.86, but was undocumented there. From 2.86 on,
(eg) --address=/example.com/1.2.3.4 ONLY applies to A queries. All other
types of query will be sent upstream. Pre 2.86, that would catch the
whole example.com domain and queries for other types would get
a local NODATA answer. The pre-2.86 behaviour is still available,
by configuring --address=/example.com/1.2.3.4 --local=/example.com/
```

Mijzelf, 2022-07-10:
Thanks for that! Both the online changelog and manpage (https://dnsmasq.org/CHANGELOG https://dnsmasq.org/docs/dnsmasq-man.html) don't tell that.
It's out of scope here, but I wonder when ever you would want this behavior. I can't think of a scenario where you would want to intercept an A query, and leave all other queries.
Anyway, I'll close the issue.

closed 2022-07-10 by Mijzelf