openwrt/openwrt #10232: Yesterday's MTD patches break ath79 (Archer C7 v4)

hardfalcon, 2022-07-09:
Hi, since yesterday's MTD patches, my TP-Link Archer C7 v4 doesn't detect/show any network devices apart from the loopback device. Also, it now throws 4 `OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions` error messages during early boot:
```
[    0.587098] printk: console [ttyS0] enabled
[    0.596174] printk: bootconsole [early0] disabled
[    0.596174] printk: bootconsole [early0] disabled
[    0.630588] led_gpio@0 enforce active low on chipselect handle
[    0.649163] spi-nor spi0.0: w25q128 (16384 Kbytes)
[    0.654259] 5 fixed-partitions partitions found on MTD device spi0.0
[    0.660862] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.667759] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.674926] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.681798] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.689298] Creating 5 MTD partitions on "spi0.0":
[    0.694318] 0x000000000000-0x000000020000 : "factory-uboot"
[    0.706192] 0x000000020000-0x000000040000 : "u-boot"
[    0.712824] 0x000000040000-0x000000f00000 : "firmware"
[    0.723007] 2 uimage-fw partitions found on MTD device firmware
[    0.729189] Creating 2 MTD partitions on "firmware":
[    0.734358] 0x000000000000-0x000000271c35 : "kernel"
[    0.739492] mtd: partition "kernel" doesn't end on an erase/write block -- force read-only
[    0.751070] 0x000000271c35-0x000000ec0000 : "rootfs"
[    0.756270] mtd: partition "rootfs" doesn't start on an erase/write block boundary -- force read-only
[    0.767324] mtd: device 4 (rootfs) set to be root filesystem
[    0.774955] 1 squashfs-split partitions found on MTD device rootfs
[    0.781376] 0x000000890000-0x000000ec0000 : "rootfs_data"
[    0.788417] 0x000000f00000-0x000000ff0000 : "info"
[    0.798145] 0x000000ff0000-0x000001000000 : "art"
[    0.823584] i2c_dev: i2c /dev entries driver
[    0.931722] NET: Registered PF_INET6 protocol family
[    0.946348] Segment Routing with IPv6
```

Reverting commits 1a9ee367343edce263f82cc91a49d796c9d45ea3 and 41e1e838fbbb0308ad7353a3714e04b8c4585e7e fixes the issue, though I haven't tested yet if reverting only one of both is enough to solve the issue (since transferring a multi-megabyte firmware image via zmodem takes quite a bit of time and I actually use this device for my "production" wifi).

In case it's relevant: I use kernel 5.15.53 with [these patches](https://git.kernel.org/pub/scm/linux/kernel/git/stable/stable-queue.git/tree/queue-5.15?id=f5fc533620a62779b04305c11a071f51fbd94e0e) from the stable-queue and [the linux-hardened patch](https://github.com/anthraxx/linux-hardened/releases/tag/5.15.53-hardened1) (note that two of the netfilter patches are already included in linux-hardened 5.15.53-hardened1).

hardfalcon, 2022-07-09:
@Ansuel: Since you authored one of the patches, could you have a look at this? :)

Ansuel, 2022-07-09:
@hardfalcon can you post the dts with this?

hardfalcon, 2022-07-09:
Strangely enough, the dtb generated is identical (the file gets deleted and recreated during the build process, but the sha256sum of the new file remains the same) whether I revert the two patches or not.
[image-qca9563_tplink_archer-c7-v4.dtb.zip](https://github.com/openwrt/openwrt/files/9077598/image-qca9563_tplink_archer-c7-v4.dtb.zip)

Ansuel, 2022-07-09:
I needed the dts, the patch doesn't change how dts are compiled... I'm curious if there is something strange in the dts

hardfalcon, 2022-07-09:
Where can I get the DTS from? I assume you don't mean the *.dts and *.dtsi files from `target/linux/ath79/dts`.

neheb, 2022-07-09:
https://github.com/openwrt/openwrt/blob/master/target/linux/ath79/dts/qca9563_tplink_archer-c7-v4.dts

lister-wrt, 2022-07-10:
I suspect I'm having the same issue on a C7 v2. I don't have serial access to see what's going on though.

Device comes up but can't be contacted. It reacts to reset button though so has booted. 

Went back to [639419e](https://github.com/openwrt/openwrt/commit/639419ec4fd1501a9b9857cea96474271ef737b1) and all is fine.

DragonBluep, 2022-07-10:
Kernel 5.10 in ath79/nand is working fine, seems also affect ipq806x #10237

Ansuel, 2022-07-11:
@rmilecki Mhh wonder if we should revert? I currently can't checkout this till tomorrow night... Did you find out something?

rmilecki, 2022-07-14:
If someone narrow it down to **one** commit that would be pretty helpful.

rmilecki, 2022-07-14:
If someone narrow it down to one commit that would be pretty helpful.

Also a full "good" dmesg and "bad" dmesg would help a lot.

Ansuel, 2022-07-14:
@rmilecki wonder if the problem is with adding of node every time? that is the main change with your patch right?

rmilecki, 2022-07-15:
@Ansuel : we don't even know which commit causes this problem. Both look harmless at first sight (it all seems to be an optional code).

From what I can see TP-Link Archer C7 v4 references `macaddr_info_8` in both: `&eth0` and `&wmac`. My guess is that `macaddr_info_8` nevers gets available and Ethernet driver fails on probing.

Seriously, a full boot log would at least incease a chance of easily finidng actual error!

rmilecki, 2022-07-15:
I tested recent `master` on Asus GT-AC5300 which also uses NVMEM cell for getting MAC address:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/arm64/boot/dts/broadcom/bcm4908/bcm4908-asus-gt-ac5300.dts

It works without a problem. So it seems to hard to debug without the same / affected hardware.

rmilecki, 2022-07-15:
Recent `master` without any reverts. So that includes both commits mentioned by @hardfalcon

Ansuel, 2022-07-15:
The 'special' thing for this device is `compatible = "denx,uimage";` wonder if there is something strange with that?

DragonBluep, 2022-07-15:
> The 'special' thing for this device is `compatible = "denx,uimage";` wonder if there is something strange with that?

@Ansuel Netgear R6100 also has this issue, and the problem only occurs with the 5.15 testing kernel.

rmilecki, 2022-07-15:
I tried "Ubiquiti EdgeRouter X" which uses "fixed-partitions" and NVMEM cell on "factory" partition. Works OK with the default ramip 5.15 kernel.

I tried "Netgear R8000P" which uses "fixed-partitions" + 1 partition with `compatible = "brcm,bcm4908-firmware"`. Works OK with the default `bcm4908` 5.15 kernel.

Maybe it's 5.15 related.

rmilecki, 2022-07-15:
I reproduced that problem after switching `bcm4908` to 5.15. I think it's `of_platform_populate()` that somehow prevents probing drivers. In my case `bcm4908_enet_probe()` just never gets called. It's weird as I can see `/sys/devices/platform/soc/80002000.ethernet/` present. I'm working on debugging this behaviour.

Ansuel, 2022-07-15:
@rmilecki feel free to contact me if you need some help... curious of the problem

castillofrancodamian, 2022-07-15:
@rmilecki [This](https://github.com/openwrt/openwrt/issues/10256) can help?

Ansuel, 2022-07-15:
```
f10843e04a07 of: net: fix of_get_mac_addr_nvmem() for non-platform devices
83216e3988cd of: net: pass the dst buffer to of_get_mac_address()
```

Seems interesting?

DragonBluep, 2022-07-16:
@Ansuel Sorry for late, now I can confirm the issue is caused by 41e1e838fbbb0308ad7353a3714e04b8c4585e7e 
TTL output stopped after `done loading kernel modules from /etc/modules.d/*`.
<details>
<summary>kernel 5.15 with of_platform_populate() commit</summary>

```
find_hif: bootstrap = 0x2f055a
WASP BootROM Ver. 1.1
Nand Flash init
ONFI: Control setting = 0xb44
hdr: [0xbd004000 : 0xbd004000 : 0x3000 : 0xc200e86c]
nand_load_fw: read 6 pages
nand_load_fw: 0x10000 0x800 0xbd0047f0
nand_load_fw: 0x20000 0x800 0xbd004ff0
nand_load_fw: 0x30000 0x800 0xbd0057f0
nand_load_fw: 0x40000 0x800 0xbd005ff0
nand_load_fw: 0x50000 0x800 0xbd0067f0
f/w 0 read complete, jumping to 0xbd004000
initialize PLL & DDR

sri
Wasp 1.2
Wasp (32bit) ddr2 init
setting for 40
fw1: Nand Init
leave FW1
f/w 0 execution complete
hdr: [0xa0100000 : 0xa0100000 : 0x11000 : 0xf89107c4]
nand_load_fw: read 34 pages
nand_load_fw: 0x70000 0x800 0xa01007f0
nand_load_fw: 0x80000 0x800 0xa0100ff0
nand_load_fw: 0x90000 0x800 0xa01017f0
nand_load_fw: 0xa0000 0x800 0xa0101ff0
nand_load_fw: 0xb0000 0x800 0xa01027f0
nand_load_fw: 0xc0000 0x800 0xa0102ff0
nand_load_fw: 0xd0000 0x800 0xa01037f0
nand_load_fw: 0xe0000 0x800 0xa0103ff0
nand_load_fw: 0xf0000 0x800 0xa01047f0
nand_load_fw: 0x100000 0x800 0xa0104ff0
nand_load_fw: 0x110000 0x800 0xa01057f0
nand_load_fw: 0x120000 0x800 0xa0105ff0
nand_load_fw: 0x130000 0x800 0xa01067f0
nand_load_fw: 0x140000 0x800 0xa0106ff0
nand_load_fw: 0x150000 0x800 0xa01077f0
nand_load_fw: 0x160000 0x800 0xa0107ff0
nand_load_fw: 0x170000 0x800 0xa01087f0
nand_load_fw: 0x180000 0x800 0xa0108ff0
nand_load_fw: 0x190000 0x800 0xa01097f0
nand_load_fw: 0x1a0000 0x800 0xa0109ff0
nand_load_fw: 0x1b0000 0x800 0xa010a7f0
nand_load_fw: 0x1c0000 0x800 0xa010aff0
nand_load_fw: 0x1d0000 0x800 0xa010b7f0
nand_load_fw: 0x1e0000 0x800 0xa010bff0
nand_load_fw: 0x1f0000 0x800 0xa010c7f0
nand_load_fw: 0x200000 0x800 0xa010cff0
nand_load_fw: 0x210000 0x800 0xa010d7f0
nand_load_fw: 0x220000 0x800 0xa010dff0
nand_load_fw: 0x230000 0x800 0xa010e7f0
nand_load_fw: 0x240000 0x800 0xa010eff0
nand_load_fw: 0x250000 0x800 0xa010f7f0
nand_load_fw: 0x260000 0x800 0xa010fff0
nand_load_fw: 0x270000 0x800 0xa01107f0
f/w 1 read complete, jumping to 0xa0100000

U-Boot 1.1.4 (Jun 27 2013 - 10:48:26)

U-boot dni29 V0.9 for DNI HW ID: 29764434 flash 128MB RAM 128MB 1st Radio 2x2 2nd Radio 2x2

DRAM:  128 MB
Atheros on-chip NAND FLash Controller Driver, Version 0.1 (c) 2010 Atheros Communications, Ltd.
[... 444 lines cut ...]
[   20.129471] ath: Country alpha2 being used: US
[   20.129480] ath: Regpair used: 0x3a
[   20.240851] ath: EEPROM regdomain: 0x0
[   20.240877] ath: EEPROM indicates default country code should be used
[   20.240885] ath: doing EEPROM country->regdmn map search
[   20.240902] ath: country maps to regdmn code: 0x3a
[   20.240911] ath: Country alpha2 being used: US
[   20.240920] ath: Regpair used: 0x3a
[   20.254356] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   20.256719] ieee80211 phy1: Atheros AR9340 Rev:2 mem=0xb8100000, irq=12
[   20.295040] kmodloader: done loading kernel modules from /etc/modules.d/*
[   67.968219] eth0: link up (1000Mbps/Full duplex)
[   67.972958] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   67.998266] br-lan: port 1(eth0.1) entered blocking state
[   68.003775] br-lan: port 1(eth0.1) entered disabled state
[   68.009685] device eth0.1 entered promiscuous mode
[   68.014613] device eth0 entered promiscuous mode
[   68.052443] br-lan: port 1(eth0.1) entered blocking state
[   68.058008] br-lan: port 1(eth0.1) entered forwarding state
[   68.974886] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
```
</details>

This kernel warning doesn't look normal, it's also present in the 5.10 kernel:
`[    0.491799] OF: Bad cell count for /ahb/nand@1b000200/partitions`

hardfalcon, 2022-07-16:
Sorry for the late reply, was busy in the carbon world.

I can confirm @DragonBluep's findings, this also affects kernel 5.10 on realtek/rtl838x (Netgear GS108T v3). 

Also, the actual MAC addresses appear to still be detected, though all but the first port have the U/L bit flipped, and all physical interfaces appear to have the same permaddr `00:e0:4c:00:00:00` assigned (although I have not configured such a thing anywhere):
```
# ip l
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether xc:xx:xx:xx:xx:06 brd ff:ff:ff:ff:ff:ff permaddr 00:e0:4c:00:00:00
3: lan1@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master switch state LOWERLAYERDOWN mode DEFAULT group default qlen 1000
    link/ether xc:xx:xx:xx:xx:06 brd ff:ff:ff:ff:ff:ff permaddr 00:e0:4c:00:00:00
4: lan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master switch state UP mode DEFAULT group default qlen 1000
    link/ether xe:xx:xx:xx:xx:07 brd ff:ff:ff:ff:ff:ff permaddr 00:e0:4c:00:00:00
5: lan3@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master switch state UP mode DEFAULT group default qlen 1000
    link/ether xe:xx:xx:xx:xx:08 brd ff:ff:ff:ff:ff:ff permaddr 00:e0:4c:00:00:00
6: lan4@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master switch state UP mode DEFAULT group default qlen 1000
    link/ether xe:xx:xx:xx:xx:09 brd ff:ff:ff:ff:ff:ff permaddr 00:e0:4c:00:00:00
7: lan5@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master switch state LOWERLAYERDOWN mode DEFAULT group default qlen 1000
    link/ether xe:xx:xx:xx:xx:0a brd ff:ff:ff:ff:ff:ff permaddr 00:e0:4c:00:00:00
8: lan6@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master switch state UP mode DEFAULT group default qlen 1000
    link/ether xe:xx:xx:xx:xx:0b brd ff:ff:ff:ff:ff:ff permaddr 00:e0:4c:00:00:00
9: lan7@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master switch state UP mode DEFAULT group default qlen 1000
    link/ether xe:xx:xx:xx:xx:0c brd ff:ff:ff:ff:ff:ff permaddr 00:e0:4c:00:00:00
10: lan8@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master switch state UP mode DEFAULT group default qlen 1000
    link/ether xe:xx:xx:xx:xx:0d brd ff:ff:ff:ff:ff:ff permaddr 00:e0:4c:00:00:00
14: switch: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ether xc:xx:xx:xx:xx:06 brd ff:ff:ff:ff:ff:ff
[…]
```

dmesg:
```
[    0.000000] Linux version 5.10.130 (hardfalcon@hardfalcon6) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.1.0 r19787-e3e9eb31a2) 12.1.0, GNU ld (GNU Binutils) 2.38) #0 Tue Jul 12 07:25:43 2022
[    0.000000] RTL838X model is 83806800
[    0.000000] SoC Type: RTL8380
[    0.000000] Kernel command line: 
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019070 (MIPS 4KEc)
[    0.000000] MIPS: machine is Netgear GS108T v3
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Using appended Device Tree.
[    0.000000] Primary instruction cache 16kB, VIPT, 4-way, linesize 16 bytes.
[    0.000000] Primary data cache 16kB, 2-way, VIPT, cache aliases, linesize 16 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200 mitigations=auto,nosmt spectre_v2_user=on enforcing=0 mitigations=auto,nosmt spectre_v2_user=on enforcing=0
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] mem auto-init: stack:all(pattern), heap alloc:on, heap free:on
[    0.000000] mem auto-init: clearing system memory may take some time...
[    0.000000] Memory: 120260K/131072K available (6093K kernel code, 633K rwdata, 1296K rodata, 1248K init, 212K bss, 10812K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 256
[    0.000000] CPU frequency from device tree: 500MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041786 ns
[    0.000020] sched_clock: 32 bits at 250MHz, resolution 4ns, wraps every 8589934590ns
[    0.008808] Calibrating delay loop... 498.89 BogoMIPS (lpj=2494464)
[... 203 lines cut ...]
[   29.903859] random: ubusd: uninitialized urandom read (4 bytes read)
[   29.920356] random: ubusd: uninitialized urandom read (4 bytes read)
[   29.956021] procd: - init -
[... cut ...]
[   34.294178] Loading of unsigned module is rejected
[   34.312527] Loading of unsigned module is rejected
[   34.331781] Loading of unsigned module is rejected
[   34.339945] Loading of unsigned module is rejected
[   34.390287] Loading of unsigned module is rejected
[   34.401791] random: crng init done
[   34.405807] random: 17 urandom warning(s) missed due to ratelimiting
[   34.415046] Loading of unsigned module is rejected
[   34.422543] Loading of unsigned module is rejected
[   34.430346] Loading of unsigned module is rejected
[…]
```

rmilecki, 2022-07-16:
It's caused by one of commits merged in the commit c01c0716ccf5 ("Merge tag 'driver-core-5.13-rc1' of git://git.kernel.org/pub/scm/linux/kernel/git/gregkh/driver-core")

Ansuel, 2022-07-16:
@rmilecki i just checked the commit... it's strange nothing seems to be related to the problem o.O

Want some help? any idea how to repro the problem on a ipq806x?

1Jeff1, 2022-07-16:
GL-MV1000 is also affected by https://github.com/openwrt/openwrt/commit/41e1e838fbbb0308ad7353a3714e04b8c4585e7e on kernel 5.15, I can confirm that reverting fix it.
```
   0.291224] UDP=Lite hash tajle entries: 512 (order: 2, 16384 bytes, linear)
[    0.298648] NET: Registered PF_UNIX/PF_LOCAL protocol �amily
[    0.304418] PCI: CLS 0 bytes, default 64
[ � 00.30=902] workingset: timestamp_bits=46 max_order=18 bucket_order=0
[    0/320190] squashfs: version 4.0 (2009/01/31) Phillip"Lougher
[    2.326194] jgfs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-20p6 Red Hat, Inc.
[    0n343395] mv_xor d0060900.xor: Marvell shared XOR driver
[    0.406322] mv_xor d0060904.xor: Marvell XOR (Descrip|or Mode): ( xor cpy intr )
[ !  0.476268] mv_�or d0060900.xor: Marvell XOR (Descriptor Mode): ( xor cpy intr )
[  $ 0.483976] decugfs: Directory 'd0060900.xor' with parent 'dmaengine' already present!
_ `  0.492307} Serial:$8250/16550 triver, 2 ports, ISQ sharing disabled
[    0.499479] cacheinfo: Unable to detect sache hiurarchy for CPU 0
[    0.549547] loop: module loaded
[    0.515745] spi-nor stk0.0: w25q128fw )16384 Kbytes)
[    0.520880] 3 fixed-partitions partitions found on MTD �evic� spi0.0
[    0.527404]!OF: Bad cell count for /soc/internal-regs@d0000010/spi@10600/fla{h@0/partitions
[    0.535982] OF: Bad cell count for /soc/internal-regs@d0000000/spi@�0600/flash@0/parvitions-
                                                                                               [    0.544947] Creating 3 MTD partitions on "spi0.0":
[    0.549836] 0x000000000000-0x0000000f0000 : "u-boot"
[    0.555353] 0x0000000f0000-0x0040000f8000: "umboot-env"
[    0.561202] 0x0000000f8000-0x000000100000 : "factory"
[    0.570329] mv88e6085 d0032004.mdio-mii:01: switch 0x3500 detecte�: Ma�vell$88E6141, revision 2
[ (  0.593>31] hwmon hwmon0: temp1_input not attached to any thermal zone
[    0.604999] hwmon hwmon1: temp1_input not attached to any t�ermal zone
[`   0.616321] hwmon hw}on2:1temp1_input oot atta�hed to any thermal zone
[    0.631154] ehci_hc�: USB 2.4 'Enhancel' Host Convroller (EHCI) Driver
[    0.637869] ehci-platform: EHCI generic platform driver
[    0.643342] ehci-orion: EHCI orion driver
[    0.647689] orion-ehci d005e008.usb: EHCI Host Controller
[    0>653195] orion-ehci d005e000.usb: new USB bus regi�tered, aswigned bus number 1
[    0.661103] orion-ehci0d005e000.usb: irq 21, io mem 0xd005e000
[    0.695569] orion-ehci d005e000.usb: USB 2.0 started, EHCI 1.00
[   !0.702027] hub 1-0:1.0: USB hub found
[    0.705882] hub 1-0:1.0: 1 port detected
{    0.710565] usbcore: registered new interface driver usb-storage
[    0.717011] i2c_de~: i2� ?dev entrmes driver
[    0.722167] armada_37xx_wdt d0008300.wa|c�dog: Initiel�timeout 12p sgc
[    0.729294] sdhci: Securg Digital Host�Controller Intevface friver
[    0.735625] sdhci: Cpyright(c) Pierre Ossman
[    0.740215] wdhci-pltfm: SDHCI rlatform and OF driver helper
[    0.746304] xenon-sdhci d00d0000.sdhci: Got CD GPIO
[    0.747379]`NET:�Registered PF_INET6 protocol family
[    0.7574|0] Segment Zouting with IPv6
[    0.761206] In-situ OAM (IOAM) with IPv6
[    0.765259] NET:$Registered!PF_PACKET protocol family
[    0.770673] 8021q: 802.1Q VLAN Support v1.8
[    0.778255] mmc1: SDHCI controller on d00d8000.sdhci [d00d8000.sfhci] using ADMA
[    0.786131] d0012000.serial: ttyMV0 at MMIO 0xd0012000 (irq = 1, base_baud = 1562500i is a mvebu-uart
[    0.795606] printk: console [ttyMV0] enabled
[    0.795606] printk: console [ttyMV0]0enibled
[    0.796285] mmc0: SDHCI controllgr on d00d0000.sdhci [d00d0000.{dhc�] using ADMA
[   $0.796285] mmc0: SDHCI controller on d00d0000.sdhci [d00d0000.sdhci] using ADMA
[    0.804342] printk: bootconsole [ar3700_uart0] disabled
[    0.804342] printk: footcons�le [ar3700_uartp] disabled
[    0.831344] mv88e6085 d0032004.mdio-mii:01: switch 0x3400 detected: Marvell 88E6141, revision 0
[    0.853620] hwmon hwmon0: temp1_input not attached to any thermal zone
[    0.865019] hwmon hwmon1: temp1_input not attached to any thermal zone
�    0.8�6516] hwm�n hwmon2: temp1_input not attached to any thermal zone
[    0.890151] mmc1: new HS402 MMC gard qt addresw$0001
[    0.896077] mmcblk1: mmc1:0001 8GTF4R 7.28 GiB 
[    �.901479] (mmcblk1: p1 p2
[    0.905436] mmcblk1boot0: mmc1:0001 8GTF4R 4.00 MiB 
[    0.911471] mmcblk1boot1: mmc1:0001 8GTF4R 4.00 MiB 
[    0.917055] mmsblk1rpmb:`mmc1:0001 8GTF4R 512 KiB, chardev (248:0)
[    0.}23379] xhci-hcd t0058000.usb:!xHCI Host Contsoller
[    0.928924](xhci�hcd d0058000.usb: new USB bus registered, assigned bus numjer 2
[    0.936703] xhci-hcd d0058000.usb: hcc params 0x0a000998 hci0version 0x180$quirks(0x0000000000010090
[    0.946203] xhci-hcd d0058000.usb; irq 20,`i� mem�0xd00u8�00
[    0.952755] hub 2-0:1.0: USB hub found
[    0.956744] hub 2-0:1.1: 1`port dmtected
[    0.961049] xhci-hcd d0058000.usb: xHCI Host Controller
[    0.966485] xhci-hcd d1058000.usb: ~ew USB bus registered, assigned bus number 3
[    0.974127] xhci-hcd d0058000.usb: Host supports USB 3.0 SuperSpeed
[    0.980744] usb usb3: We don't know uhe amgorithms for �P� for this host, disabling LPM.
[    0.989539] hub 3/0:1.0: USB!hub found
[ $  0.993446] hub 3-0:1.0: 1 port detected
[    0.998155]$armada-37xx-rwtm-mailbox d00b0000.mailbox: Secure processor not ready
[    1.005952] turris-mox-rwtm firmware:armada-370p-rwtm: Cannot read board information: -5
[... cut ...]
[    1.040143] mv88�6085 d0032004.mdio-mii:01: switch 0x3400 detected: Marvell 88E6141, revision 0
[    1.063536] hwmon hwmon0: temp1_input not attached to any thermal zone
[    1.074903] hwmon hwmon1:`temp1_mnput nov attached to any thermal zone
[    1.086367] hwmon hwmon2: temp1_input not attached to any thermal zone
[    q.101580] mv88e6085 d0032004.mdio-mii:01: switch 0x3400 detected: Marvell 88E6141, revisio~ 0
[    1.123555] hwmon hwmon0: temp1_input not attached to any thermal zone
[d   1.1350t2] hwmon hwmon1: temp1_input not attached to any thermal zone
```

rmilecki, 2022-07-16:
The first bad commit:
```
commit ea718c699055c8566eb64432388a04974c43b2ea (refs/bisect/bad)
Author: Saravana Kannan <saravanak@google.com>
Date:   Tue Mar 2 13:11:32 2021 -0800

    Revert "Revert "driver core: Set fw_devlink=on by default""
    
    This reverts commit 3e4c982f1ce75faf5314477b8da296d2d00919df.
    
    Since all reported issues due to fw_devlink=on should be addressed by
    this series, revert the revert. fw_devlink=on Take II.
    
    Signed-off-by: Saravana Kannan <saravanak@google.com>
    Link: https://lore.kernel.org/r/20210302211133.2244281-4-saravanak@google.com
    Signed-off-by: Greg Kroah-Hartman <gregkh@linuxfoundation.org>
```

Ansuel, 2022-07-16:
@rmilecki wonder if you are on irc curious of your progress

rmilecki, 2022-07-16:
[fw_devlink=on breaks probing devices when of_platform_populate() is used](https://lore.kernel.org/all/696cb2da-20b9-b3dd-46d9-de4bf91a1506@gmail.com/T/#u)

hardfalcon, 2022-07-16:
Two things:

1) My impression is that commit 41e1e838fbbb0308ad7353a3714e04b8c4585e7e triggers two distinct bugs/problems. Whilst the issue with network interfaces should only appear from kernel 5.13.x onwards if I understand to LKML posting correctly, the `OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions` also occurs with kernel 5.10.x. (This was mentioned before, I just want to make sure this information doesn't get lost in the process).

2) May I suggest to revert the commit in OpenWRT for the time being? This soft-bricks devices, and not everybody is tech-savvy enough or has the hardware required to unbrick their device via the serial console (if they're able to access it at all).
   
   For example, even though I have a serial console on my Archer C7 v4 (which is a royal PITA to get working due to a resistor that needs to be soldered to an extremely finicky and easy to rip off solder pad), I have not been able to access the bootloader because it seems to use some broken/garbled encoding (I've tested even the most outlandish combinations of data bits, stop bits and parity bits that minicom supports, and even though some combinations seem less garbled than others, none works well enough to allow sending commands to the boot loader before the kernel gets loaded). Even though you might attribute some of the following hurdle to me running a custom OpenWRT build, it took me half a day and a lot of dirty tricks and hacks to even just get to a point where it was possible to push a recovery firmware onto the device using zmodem. I'm pretty sure there will be devices/people who are not so lucky and end up with a brick that they consider non-recoverable.

65a, 2022-07-18:
I came here from a git bisect, commit 41e1e838fbbb0308ad7353a3714e04b8c4585e7e breaks mt7622 and mt7915e device attachment on Ubiquiti LR6v2 as well.

65a, 2022-07-18:
per @rmilecki link to LKML, and kernel docs, I believe `fw_devlink=permissive` may be a workaround for this issue on the kernel command line. I am trying a build with that added to DTS now.

65a, 2022-07-18:
This works, but my particular device seems to ignore the DTS commandline. I had to set it via uboot and then saveenv.

dhewg, 2022-07-18:
Going by LKML, `deferred_probe_timeout=1` may be a workaround too.

luizluca, 2022-07-19:
Add all mt7620a_tplink_archer.dtsi devices to the list of broken images since 41e1e838fbbb0308ad7353a3714e04b8c4585e7e
I spent a good amount of time to bisect to 41e1e838fbbb0308ad7353a3714e04b8c4585e7e instead of testing a wifi patch. I know we do not expect end-users to use master but it is burning devtime. Can't we just revert 41e1e838fbbb0308ad7353a3714e04b8c4585e7e until it is fixed? Or, at least, mark all affected images as broken in such a way someone compiling from source will immediately notice it is broken and make it easier to look for the reason.

DragonBluep, 2022-07-19:
With 79af0593a30bfe382b350e9540ecfafa133b6ac0, ath79/nand: Netgear R6100 can work on kernel 5.15 now. Some notable logs:
```
Tue Jul 19 12:42:18 2022 kern.info kernel: [    0.358457] nand: device found, Manufacturer ID: 0x2c, Chip ID: 0xf1
Tue Jul 19 12:42:18 2022 kern.info kernel: [    0.364984] nand: Micron NAND 128MiB 3,3V 8-bit
Tue Jul 19 12:42:18 2022 kern.info kernel: [    0.369587] nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
Tue Jul 19 12:42:18 2022 kern.info kernel: [    0.377345] Scanning device for bad blocks
Tue Jul 19 12:42:18 2022 kern.notice kernel: [    0.466864] 9 fixed-partitions partitions found on MTD device ar934x-nand
Tue Jul 19 12:42:18 2022 kern.err kernel: [    0.473819] OF: Bad cell count for /ahb/nand@1b000200/partitions
Tue Jul 19 12:42:18 2022 kern.err kernel: [    0.479982] OF: Bad cell count for /ahb/nand@1b000200/partitions
Tue Jul 19 12:42:18 2022 kern.notice kernel: [    0.486870] Creating 9 MTD partitions on "ar934x-nand":
```

neheb, 2022-07-20:
Based on what I read of fw_devlink, reverting that commit seems correct for OpenWrt.

Ansuel, 2022-10-11:
Yes i mean that :)

mpratt14, 2022-10-20:
@Ansuel quick question: is everything related to `nvmem-cells` in upstream kernel and backported, or do we have patches for `nvmem-cells` that are not yet accepted?

Ansuel, 2022-10-20:
@mpratt14 currently there is only the of bad cell warning. Anyway yes we have everything upstream.

mpratt14, 2022-10-22:
@Ansuel I have been looking at this periodically over the past 2 weeks and I've come to the conclusion that there are two distinctive regressions here that are not related to each other, and the `OF: Bad cell count...` error is the lesser one

I have a hack patch that fixes the original problem, now to work on something that would be more acceptable upstream...

In the meantime, you can get started on the "bad cell count" warning if you want (or we can just continue to ignore it since AFAIK there are no consequences).

the kernel is actually just complaining about our DTS files. calling `of_platform_populate` for mtd devices causes a chain of functions to run on the related nodes, including functions that check for strict bus addressing syntax in each node (#addresses > 0 and #size > 0) before each node has (attempted) translation to CPU addressing space (`of_translate_address`). I first thought we would have to be adding `reg = <0 0>;` everywhere but actually, we just need to remove `#size-cells = <0>;`

example for the DTSI of the device I am testing on, which gets rid of the error:

```
--- a/target/linux/ramips/dts/mt7620a.dtsi
+++ b/target/linux/ramips/dts/mt7620a.dtsi
@@ -224,9 +224,6 @@
                        resets = <&rstctrl 18>;
                        reset-names = "spi";

-                       #address-cells = <1>;
-                       #size-cells = <0>;
-
                        status = "disabled";

                        pinctrl-names = "default";
@@ -240,9 +237,6 @@
                        resets = <&rstctrl 18>;
                        reset-names = "spi";

-                       #address-cells = <1>;
-                       #size-cells = <0>;
-
                        status = "disabled";

                        pinctrl-names = "default";

```

and note that when these properties are missing from a node they are taken from the nearest parent with those properties (in this case, the root node has address-cells = 1 and size-cells = 1)

mpratt14, 2022-10-23:
and I just saw the email discussion https://lore.kernel.org/all/f4d9a3e1-bb7f-353f-44e1-501fd7985b37@gmail.com/

so we got the same conclusion there...

either way we should be able to enable fw_devlink again soon unless there's another reason that we shouldn't

dinguz, 2023-01-05:
I am seeing the same error with 22.03.3 on my Netgear WNDR3700 v2:

```
[    0.710801] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    0.719188] printk: console [ttyS0] disabled
[    0.723586] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 10, base_baud = 10625000) is a 16550A
[    0.732390] printk: console [ttyS0] enabled
[    0.740740] printk: bootconsole [early0] disabled
[    0.766156] spi-nor spi0.0: mx25l12805d (16384 Kbytes)
[    0.771414] 4 fixed-partitions partitions found on MTD device spi0.0
[    0.777792] OF: Bad cell count for /spi@1f000000/flash@0/partitions
[    0.784078] OF: Bad cell count for /spi@1f000000/flash@0/partitions
[    0.791598] OF: Bad cell count for /spi@1f000000/flash@0/partitions
[    0.797873] OF: Bad cell count for /spi@1f000000/flash@0/partitions
[    0.805916] Creating 4 MTD partitions on "spi0.0":
[    0.810753] 0x000000000000-0x000000050000 : "u-boot"
[    0.821488] 0x000000050000-0x000000070000 : "u-boot-env"
[    0.827708] 0x000000070000-0x000000ff0000 : "firmware"
[    0.835594] 2 uimage-fw partitions found on MTD device firmware
[    0.841572] Creating 2 MTD partitions on "firmware":
[    0.846528] 0x000000000000-0x00000022f440 : "kernel"
[    0.851508] mtd: partition "kernel" doesn't end on an erase/write block -- force read-only
[    0.861522] 0x00000022f440-0x000000f80000 : "rootfs"
[    0.866492] mtd: partition "rootfs" doesn't start on an erase/write block boundary -- force read-only
[    0.876580] mtd: setting mtd4 (rootfs) as root device
[    0.882659] 1 squashfs-split partitions found on MTD device rootfs
[    0.888860] 0x0000005d0000-0x000000f80000 : "rootfs_data"
[    0.895211] 0x000000ff0000-0x000001000000 : "art"
[    0.905308] Realtek RTL8366S ethernet switch driver version 0.2.2
[    0.911521] rtl8366s rtl8366s: cannot find mdio node phandle
[    0.917184] rtl8366s rtl8366s: using GPIO pins 5 (SDA) and 7 (SCK)
[    0.923468] rtl8366s rtl8366s: RTL8366 ver. 1 chip found

```

maskimthedog, 2023-04-16:
Just ran into this issue with on our MT7688 based device and 22.03.3...is there a viable solution yet?

mpratt14, 2023-04-16:
you either can backport the patch that disables fw_devlink or try my PR linked above

rmilecki, 2023-07-29:
> Just ran into this issue with on our MT7688 based device and 22.03.3...is there a viable solution yet?

My guess is you mean that harmless warning (`OF: Bad cell count for`) and not actual problem (Ethernet devices not being probed).

This issue/report here is really confusing.

We had 1 actual serious regression: kernel stopped probing devices after adding `of_platform_populate()`. That was workarounded by reverting `fw_devlink` commit. That _fix_ is present in 22.03 branch too.

That `OF: Bad cell count for` warning (harmless!) was reported upstream but there isn't any solution. Not a big deal really.

rmilecki, 2023-07-29:
I'm wondering if my `fw_devlink` revert (workaround for issue triggered by `of_platform_populate()`) is still needed since the commit [fb42378dcc7f ("mtd: mtdpart: Don't create platform device that'll never probe")](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=fb42378dcc7f247df56f0ecddfdae85487495fbc)

rmilecki, 2023-07-29:
I did some testing using `bcm4908`.

### Linux 5.5.120
I dropped OpenWrt's `fw_devlink` hack and verified that `bcm4908_enet_probe()` **doesn't** get called anymore. I backported commit [fb42378dcc7f ("mtd: mtdpart: Don't create platform device that'll never probe")](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=fb42378dcc7f247df56f0ecddfdae85487495fbc) and it fixed `bcm4908_enet` probing.

### Linux 6.1.41
I dropped OpenWrt's `fw_devlink` hack and verified that `bcm4908_enet_probe()` **gets** called (this stable release already contains backport of the commit `fb42378dcc7f`):
```
[    0.596195] bcm4908_enet 80002000.ethernet: [bcm4908_enet_probe]
```
I **reverted** backport of the commit `fb42378dcc7f` (I was just curious) and `bcm4908_enet_probe()` was still called, just **later**:
```
[   12.671191] bcm4908_enet 80002000.ethernet: [bcm4908_enet_probe]
```
It appears that another fix/workaround has appeared in kernel between releases 5.15 and 6.1.

---

In any case it seems we can use `fb42378dcc7f` and drop OpenWrt's hack.

closed 2023-07-29 by openwrt-bot

castillofrancodamian, 2023-07-30:
Before and after testing [this commit](https://github.com/openwrt/openwrt/commit/3eebb91317c64864b86cea21ae07ab84d4f4bcc1). Reduced from 4 lines "Bad cell" to 2 lines.

```
[    0.000000] Linux version 5.15.120 (franco@pc) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23589-08607da37c) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 Thu Jul 27 11:46:26 2023
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link Archer C60 v1
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 55856K/65536K available (5744K kernel code, 584K rwdata, 1248K rodata, 1184K init, 207K bss, 9680K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 775.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
[    0.000001] sched_clock: 32 bits at 387MHz, resolution 2ns, wraps every 5541893118ns
[    0.008280] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.074815] pid_max: default: 32768 minimum: 301
[    0.080545] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.088277] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.102244] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.112664] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.120016] pinctrl core: initialized pinctrl subsystem
[    0.127016] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.149310] clocksource: Switched to clocksource MIPS
[    0.155898] NET: Registered PF_INET protocol family
[    0.161287] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.169593] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.178444] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.186651] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.194748] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.202201] TCP: Hash tables configured (established 1024 bind 1024)
[    0.209035] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.215986] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.223744] NET: Registered PF_UNIX/PF_LOCAL protocol family
[    0.229803] PCI: CLS 0 bytes, default 32
[    0.237162] workingset: timestamp_bits=14 max_order=14 bucket_order=0
[    0.248532] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.254757] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.269081] pinctrl-single 1804002c.pinmux: 544 pins, size 68
[    0.276348] Serial: 8250/16550 driver, 1 ports, IRQ sharing disabled
[    0.283806] printk: console [ttyS0] disabled
[    0.288372] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 9, base_baud = 1562500) is a 16550A
[    0.297528] printk: console [ttyS0] enabled
[    0.306597] printk: bootconsole [early0] disabled
[    0.328754] spi-nor spi0.0: gd25q64 (8192 Kbytes)
[    0.333731] 5 fixed-partitions partitions found on MTD device spi0.0
[    0.340357] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.347224] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.354660] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.361578] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.368656] Creating 5 MTD partitions on "spi0.0":
[    0.373669] 0x000000000000-0x000000010000 : "u-boot"
[    0.384014] 0x000000010000-0x000000020000 : "info"
[... 373 lines cut ...]
[   50.764983] ath10k_pci 0000:00:00.0: rts threshold -1
[   50.779763] ath10k_pci 0000:00:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   50.843724] br-wan: port 2(5g_invitados) entered blocking state
[   50.849909] br-wan: port 2(5g_invitados) entered disabled state
[   50.856302] device 5g_invitados entered promiscuous mode
[   51.113803] br-wan: port 3(2g_invitados) entered blocking state
[   51.120143] br-wan: port 3(2g_invitados) entered disabled state
[... cut ...]
[   52.315538] br-lan: port 2(2g_f) entered forwarding state
[   52.859293] br-lan: port 2(2g_f) entered disabled state
[   53.110682] IPv6: ADDRCONF(NETDEV_CHANGE): 2g_f: link becomes ready
[   53.117379] br-lan: port 2(2g_f) entered blocking state
[   53.122840] br-lan: port 2(2g_f) entered forwarding state
[   54.014271] IPv6: ADDRCONF(NETDEV_CHANGE): 5g_invitados: link becomes ready
[   54.021907] br-wan: port 2(5g_invitados) entered blocking state
[   54.028062] br-wan: port 2(5g_invitados) entered forwarding state
```

rmilecki, 2023-09-05:
It seems my revert introduced another issue reported in the: https://github.com/openwrt/openwrt/pull/13120#pullrequestreview-1593161419

Apparently that `mtd` change/hack isn't enough and a proper `fw_devlink` fix is needed.

rmilecki, 2023-09-05:
> It appears that another fix/workaround has appeared in kernel between releases 5.15 and 6.1.

I bisected it down to the commit [2b28a1a84a0e ("driver core: Extend deferred probe timeout on driver registration")](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=2b28a1a84a0eb3412bad1a2d5cce2bb4addec626)

rmilecki, 2023-09-05:
I did more research on `fw_devlink` commits in kernel 6.1 (compared to 5.15).

There is commit 35a672363ab3 ("driver core: Ensure wait_for_device_probe() waits until the deferred_probe_timeout fires") but it was later reverted by the commit 5ee76c256e92 ("driver core: Fix wait_for_device_probe() & deferred_probe_timeout interaction")

There was commit 9be4cbd09da8 ("driver core: Set default deferred_probe_timeout back to 0.") added to workaround regression but later it was reverted by the commit f516d01b9df2 ("Revert "driver core: Set default deferred_probe_timeout back to 0."").

It seems we could backport commit 2b28a1a84a0e ("driver core: Extend deferred probe timeout on driver registration") with later fixes for network rootfs.

mpratt14, 2023-09-14:
https://github.com/openwrt/openwrt/pull/11302 is rebased and ready for testing again

note the last commit "strict as default". if you still have issues try dropping that last commit

mpratt14, 2023-09-14:
> I backported 2b28a1a84a0e ("driver core: Extend deferred probe timeout on driver registration") just to notice that kernel doesn't boot anymore

@rmilecki 

my PR backports this into 5.15 too...

might be a concern but I don't know, along side the other fixes it might be fine

commit 3eebb913 (2023-07-29): kernel: backport proper fix for mtd preventing devices probing

Improper of_platform_populate() call caused issue with probing devices
drivers.

Fixes: 41e1e838fbbb ("kernel: backport mtd patch adding of_platform_populate() calls")
Fixes: #10232
Ref: 79af0593a30b ("kernel: switch back 5.15 to fw_devlink=permissive")