openwrt/openwrt #10237: Current Master Breaks IPQ806x, R7800, TP C2600, Kernel 5.15

klojcnik, 2022-07-10:
System boots ( LED blinks than stable ), but network wont come up. Also tried fix IP, not working. Seems if there is something wrong with the network Service.

Klojsch

tapper82, 2022-07-19:
Hi I all so have a r7800 that bootloops after flashing a snapshot using Kernel 5.15. If I flash it with a snapshot with kernel 5.10 it will work.

tapper82, 2022-07-19:
Hi this is fixt: 
TappersRouter
Model
Netgear Nighthawk X4S R7800
Architecture
ARMv7 Processor rev 0 (v7l)
Target Platform
ipq806x/generic
Firmware Version
OpenWrt SNAPSHOT r20146-bc43ad88ed / LuCI Master git-22.189.49133-e193259
Kernel Version
5.15.53

I think it was fixt by this:
https://github.com/openwrt/openwrt/commit/79af0593a30bfe382b350e9540ecfafa133b6ac0