openwrt/openwrt #10243: MediaTek MT7610E 5GHz radio not functional in 22.03.0-rc5

lespinasse, 2022-07-11:
I am observing this issue on a tl-wr902ac device. When running 22.03.0-rc5, the device is unable to connect (as a wifi client) to my main AP over the 5GHz band. The 2GHz radio works fine, and the 5GHz radio functionality is restored if I go back to the 22.03.0-rc4 release.

I believe the issue is with the MT7610E support rather than the individual device - I have seem similar reports (https://forum.openwrt.org/t/openwrt-22-03-0-rc5-fifth-release-candidate/131652/9 and https://github.com/openwrt/openwrt/issues/10190) for other devices that use the same radio. In all 3 reports (counting this one), only the 5GHz radio is broken and the break happens when moving from 22.03.0-rc4 to 22.03.0-rc5.

Since the tl-wr902ac is a small-memory device (64M), I must mention that I am running very little on it, and see about 27MB free memory in normal usage under 22.03.0-rc4.

lespinasse, 2022-08-02:
Still broken on tl-wr902ac in the 22.03.0-rc6 release.

jaccc, 2022-08-02:
still broken on AC51U

jaccc, 2022-08-03:
AC51U Log with Android phone, 2.4G disabled, free memory: 22 MB 

Wed Aug  3 13:54:35 2022 daemon.info hostapd: wlan0: STA xx IEEE 802.11: authenticated
Wed Aug  3 13:54:35 2022 daemon.info hostapd: wlan0: STA xx IEEE 802.11: associated (aid 1)
Wed Aug  3 13:54:35 2022 daemon.notice hostapd: wlan0: AP-STA-CONNECTED xx
Wed Aug  3 13:54:35 2022 daemon.info hostapd: wlan0: STA xx WPA: pairwise key handshake completed (RSN)
Wed Aug  3 13:54:35 2022 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED xx
Wed Aug  3 13:54:39 2022 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) xx
Wed Aug  3 13:54:39 2022 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.143 xx
Wed Aug  3 13:54:39 2022 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) xx
Wed Aug  3 13:54:39 2022 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.143 xx
Wed Aug  3 13:54:39 2022 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.143 xx
Wed Aug  3 13:54:39 2022 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.143 xx Android
Wed Aug  3 13:54:43 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED xx

Ntemis, 2022-08-05:
MQmaker Witi MT7602EN & MT7612EN both radio same issue radios not functional(unable to connect 22.03.0-rc6 release.

lespinasse, 2022-09-04:
Still broken in the final 22.03.0 release (well, at least as much as the current tagged but not officially announced images count as a release).

DragonBluep, 2022-09-04:
@lespinasse revert this commit can fix problem. It's a bug on mt76.
https://github.com/openwrt/mt76/commit/d6ae3505ac6c0e34ca2a209d4ac9ffe9ad87575d

lespinasse, 2022-09-05:
I've never done binary openwrt builds (only image builds from packages). What would be the target to build after reverting the mt76 commit - are we talking a full kernel build or just rebuilding the mt76 package ?

DragonBluep, 2022-09-05:
> I've never done binary openwrt builds (only image builds from packages). What would be the target to build after reverting the mt76 commit - are we talking a full kernel build or just rebuilding the mt76 package ?

These kernel modules need to be compiled with the kernel at the same time. If this process is more challenging for you, you can open a new issue in the mt76 repository. Maybe the maintainers will soon notice.

lespinasse, 2022-09-05:
I think the maintainers may not want to revert https://github.com/openwrt/mt76/commit/d6ae3505ac6c0e34ca2a209d4ac9ffe9ad87575d because that would undo whatever this commit was supposed to achieve on mt7615, mt7915 and mt7921 hardware.

I wonder if this change may be more acceptable to them, preserving the behavior on that hardware while keeping the prior behavior on other hardware:

```
--- drivers/net/wireless/mediatek/mt76/mac80211.c
+++ drivers/net/wireless/mediatek/mt76/mac80211.c
@@ -267,7 +267,8 @@ static void mt76_init_stream_cap(struct mt76_phy *phy,
        }
        vht_cap->vht_mcs.rx_mcs_map = cpu_to_le16(mcs_map);
        vht_cap->vht_mcs.tx_mcs_map = cpu_to_le16(mcs_map);
-       vht_cap->vht_mcs.tx_highest |=
+       if (ieee80211_hw_check(phy->hw, SUPPORTS_VHT_EXT_NSS_BW))
+               vht_cap->vht_mcs.tx_highest |=
                                cpu_to_le16(IEEE80211_VHT_EXT_NSS_BW_CAPABLE);
 }
 

```

I have not tested this - I did build it on x86, but can't test it there. I could probably figure out the openwrt build sometime... but if you could test if this fixes the issue on mt7610e, and if someone could test on mt7615, mt7915 or mt7921, this may be enough to try submitting the change upstream ?

DragonBluep, 2022-09-05:
@lespinasse Your changes are not much different from reverting the entire commit. They need to cooperate with each other to work. I mean let the maintainers fix this, while keeping the feature. Since we are not familiar with this driver framework, reverting commits is only a temporary workaround.

jaccc, 2022-09-06:
build scripts can probably revert commit for affected routers

jaccc, 2022-09-06:
Release Notes:
Known issues
None so far.
It seems this breakage is intentional.

lespinasse, 2022-09-06:
> @lespinasse Your changes are not much different from reverting the entire commit. They need to cooperate with each other to work. I mean let the maintainers fix this, while keeping the feature. Since we are not familiar with this driver framework, reverting commits is only a temporary workaround.

The intention is to effectively revert the https://github.com/openwrt/mt76/commit/d6ae3505ac6c0e34ca2a209d4ac9ffe9ad87575d commit on anything but rt7615, rt7915 and rt7921 radios. My reading of the change is that it tried to fix a throughput issue on these radios, and that the mac80211.c part of the change was intended for these radios only. I will admit that I have no greater understanding of wifi code beyond that, but in my experience (not wifi related) it's not uncommon for upstream to have significant blind spots, and sometimes a tentative patch (rather than just reporting the issue) can help get things started. Yes it's very reasonable to try to let upstream maintainers fix this first... but as it's been 3+ months that the mt7610e radios have been broken, I think we can reasonably assume that we're dealing with one of those upstream blind spots right now.

nbd168, 2022-09-06:
Sorry for the delay. Fix pushed to master and 22.03 branch

lespinasse, 2022-09-06:
Thanks Felix! And for reference here, the commited fix (master) is https://github.com/openwrt/mt76/commit/d70546462b7b51ebc2bcdd5c534fdf3465be62a4

Shine-, 2022-09-07:
For the record: I just tested 22.03.0 (ie. without the fix) on MiR3G (MT7612E) and Archer C2v1 (MT7610E) and I could connect to 5G just fine with both of them. So it seems that not all MT7610/MT7612 devices are affected, after all.
Either way, thanks for the fix!

lespinasse, 2022-09-07:
Interesting. I am not sure why I'm seeing different behavior than Shine on tl-wr902ac. Maybe relevant that I'm using the 5GHz radio as STA not AP ?
Either way, I can confirm 22.03 now works for me (on tl-wr902ac) after picking the 22.03-SNAPSHOT kmod-mt76* updated packages.

Shine-, 2022-09-07:
> Interesting. I am not sure why I'm seeing different behavior than Shine on tl-wr902ac. Maybe relevant that I'm using the 5GHz radio as STA not AP ?

Good point, retried as STA, connecting to a device with MT7915 (also on 22.03.0) - working fine as well.

jaccc, 2022-10-04:
Will usable firmware be released?

DragonBluep, 2022-10-04:
> Will usable firmware be released?

Just wait for v22.03.1, or you can download SNAPSHOT version https://firmware-selector.openwrt.org/ , need to add extra package "luci".

sailinggalaxians, 2022-10-09:
Problem is fixed in the latest snapshot for RT-AC51U.

tanoem, 2022-10-14:
I have a Cudy WR2100 (ramips/mt7621) and still experiencing problems with the 5GHz band after upgrading to the 22.03.1 release. Things worked fine when I was previously on the 21.02.3 release.

tanoem, 2022-10-24:
22.03.2 is also still having problems with the 5GHz band on my Cudy WR2100. The radio is listed as a MT7615E if that helps at all.

DragonBluep, 2022-10-24:
> 22.03.2 is also still having problems with the 5GHz band on my Cudy WR2100. The radio is listed as a MT7615E if that helps at all.

@tanoem We only talk about MT7610E here. Please search https://github.com/openwrt/mt76/issues to find if anyone meet the same problem as you.

@lespinasse Maybe we can close this issue. @sailinggalaxians reported problem is fixed.

tanoem, 2022-10-24:
Apologies, looks like this is likely the culprit for my problem: https://github.com/openwrt/mt76/issues/690