openwrt/openwrt #10251: My wireless card doesn't work    intel wifi  201

creaplus, 2022-07-12:
### Hardware description:
https://ark.intel.com/content/www/cn/zh/ark/products/130293/intel-wifi-6-ax201-gig.html

### executive command:
lspci -nnk

### Information: 
00:10.2 Network controller [0280]: Intel Corporation Alder Lake-S PCH CNVi WiFi [8086:7af0] (rev 11)
	Subsystem: Intel Corporation Device [8086:0074]

I installed Kmod-iwlwifi but that didn't work either

walterav1984, 2022-07-12:
Firmware???
https://openwrt.org/packages/index/firmware

```
opkg install kmod-iwlwifi #needs internet connection setup on WAN port
opkg install iwlwifi-firmware-xxx #see firmware link
ls /lib/firmware #check if firmware file is installed
dmesg #check if iwlwifu driver loads firmware or specifies missing file
```

Some debugging info different model:
https://forum.openwrt.org/t/openwrt-21-02-2-drivers/123491/35

jow-, 2022-07-13:
You need to install the appropriate firmware and kmod packages.

closed 2022-07-13 by jow-