openwrt/openwrt #10257: The password set by passwd cannot log in to luci

shenghuang147, 2022-07-12:
**Source Version:** `OpenWrt-21.02.3`

I use `passwd root` to set the password in the terminal, but this password  (even though the password set with `passwd root` is the same as the one set in the `luci` console)  cannot log in to `luci`

Also `Openssh` cannot log in with this password.

On the contrary using the `luci` interface to set the password everything works fine including `login root` , `ssh`.

Is there any solution, I haven't encountered this problem before

jow-, 2022-07-13:
Did you include shadow-passwd in your build? It might use a shadow password hash which is not supported by musl's crypt function.

shenghuang147, 2022-07-13:
> 您是否在构建中包含了 shadow-passwd？ 它可能使用 musl 的 crypt 函数不支持的影子密码哈希。

Thanks a lot, I chose `shadow-utils` when building
It has build `shadow-passwd` checked by default
I canceled building it and I this glitch disappeared

closed 2022-07-13 by shenghuang147