openwrt/openwrt #10267: miniupnpd litters the log with numerous different errors

attila-lendvai, 2022-07-13:
it's `OpenWrt 22.03.0-rc5` on a `TP-Link TL-WDR4300 v1`.

the router is a simple home router with a few devices connected. one of them is an old laptop used as a server, running two Bee nodes (quite a few connection and p2p traffic; more details at https://www.ethswarm.org/ and https://github.com/ethersphere/bee/)

```
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: nft_send_rule(0x77cf0950, 6, 0) send_batch failed -4
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: send_batch: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: nft_send_rule(0x77cef600, 6, 0) send_batch failed -4
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: send_batch: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: nft_send_rule(0x77d01950, 6, 0) send_batch failed -4
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: send_batch: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: nft_send_rule(0x77cf0b40, 6, 0) send_batch failed -4
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: send_batch: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: nft_send_rule(0x77cef7e0, 6, 0) send_batch failed -4
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: send_batch: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: nft_send_rule(0x77d13b40, 6, 0) send_batch failed -4
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: send_batch: mnl_cb_run returned -1
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: nft_send_rule(0x77d35600, 6, 0) send_batch failed -4
Wed Jul 13 22:03:25 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:33 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:34 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:34 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: send_batch: mnl_cb_run returned -1
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: nft_send_rule(0x77cf0580, 6, 0) send_batch failed -4
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: send_batch: mnl_cb_run returned -1
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: nft_send_rule(0x77cef020, 6, 2) send_batch failed -4
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: send_batch: mnl_cb_run returned -1
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: nft_send_rule(0x77cf0960, 6, 2) send_batch failed -4
Wed Jul 13 22:03:44 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:45 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:45 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:45 2022 daemon.err miniupnpd[3013]: send_batch: mnl_cb_run returned -1
Wed Jul 13 22:03:45 2022 daemon.err miniupnpd[3013]: nft_send_rule(0x77cef220, 6, 2) send_batch failed -4
Wed Jul 13 22:03:45 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:45 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:45 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:45 2022 daemon.err miniupnpd[3013]: send_batch: mnl_cb_run returned -1
Wed Jul 13 22:03:45 2022 daemon.err miniupnpd[3013]: nft_send_rule(0x77cf0b50, 6, 2) send_batch failed -4
Wed Jul 13 22:03:45 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:45 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
Wed Jul 13 22:03:45 2022 daemon.err miniupnpd[3013]: refresh_nft_cache: mnl_cb_run returned -1
[... 84 lines cut ...]

# hw details

```
# cat /proc/cpuinfo 
system type		: Atheros AR9344 rev 2
machine			: TP-Link TL-WDR4300 v1
processor		: 0
cpu model		: MIPS 74Kc V4.12
BogoMIPS		: 278.93
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp dsp2
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available
```

attila-lendvai, 2022-07-13:
this seems to be a config issue? see https://github.com/miniupnp/miniupnp/issues/593#issuecomment-1171277513

attila-lendvai, 2022-07-13:
not sure it's relevant, but:

```
# /etc/init.d/miniupnpd restart
Section @include[0] is not marked as compatible with fw4, ignoring section
Section @include[0] requires 'option fw4_compatible 1' to be considered compatible
Section miniupnpd option 'family' is not supported by fw4
Section miniupnpd option 'reload' is not supported by fw4
Section miniupnpd specifies unreachable path '/usr/share/miniupnpd/firewall.include', ignoring section
```

tiagogaspar8, 2022-07-14:
Hi @attila-lendvai this is a known issue and is being worked with upstream, also, this package doesn't belong here, so these kind of issues should be opened in the packages repo

attila-lendvai, 2022-07-14:
@tiagogaspar8 thanks for the info! i'd appreciate if you could link the relevant issue here.

other than that, feel free to close this (or let me know if you think this should be closed but you lack the priviledges).

tiagogaspar8, 2022-07-14:
It is being discussed in issue packages#17871 and partially in a pull request in packages#18133

I do not have the permition to close it, I am just a regular user.

attila-lendvai, 2022-07-14:
FTR, it's discussed in https://github.com/openwrt/packages/issues/17871 and https://github.com/openwrt/packages/pull/18133.

closing this one.

closed 2022-07-14 by attila-lendvai