openwrt/openwrt #10272: generic: track and understand wrong formatted patch in hack/pending

Ansuel, 2022-07-14:
Hi,

This is not a real issue with Openwrt itself but more of a problem we have on our repo.

We have too many patches that are wrongly formatted, with no author/sob, with no reason they are actually there and sometimes (probably) even problematic with recent changes to kernel version.

We have to fix this for good or at least have a place to track them and with times try to fix them.

**So for every maintainer, feel free to edit this and add additional info about patch status...**

(this is a wip issue so i'm adding stuff as i find more info)

THE LIST
---
TO BE COMPLETED

Ansuel, 2022-07-14:
Ok... I added all the comments for the patch that i think can be improved...

blocktrron, 2022-07-15:
> 761-dt-bindings-net-add-RTL8152-binding-documentation.patch

There is no need for the Documentation without the OF part. We might as well just drop this patch altogether, as it isn't going upstream.

Ansuel, 2022-07-15:
> There is no need for the Documentation without the OF part. We might as well just drop this patch altogether, as it isn't going upstream.

The idea would be to push a simple Documentation and then if we are lucky adding the leds relevant part when it will be ready.

blocktrron, 2022-07-15:
There is no need for documentation without the LED binding. I have no interest in investing time to push a stub upstream.

hauke, 2022-07-18:
I think this patch is not needed any more:
https://github.com/openwrt/openwrt/blob/master/target/linux/generic/hack-5.15/773-bgmac-add-srab-switch.patch

The b53 swconfig srab is not  used any more:
````
$ git grep SWCONFIG_B53_SRAB_DRIVER
target/linux/bcm47xx/config-5.10:# CONFIG_SWCONFIG_B53_SRAB_DRIVER is not set
target/linux/generic/config-5.10:# CONFIG_SWCONFIG_B53_SRAB_DRIVER is not set
target/linux/generic/config-5.15:# CONFIG_SWCONFIG_B53_SRAB_DRIVER is not set
target/linux/generic/files/drivers/net/phy/b53/Kconfig:config SWCONFIG_B53_SRAB_DRIVER
target/linux/generic/files/drivers/net/phy/b53/Makefile:obj-$(CONFIG_SWCONFIG_B53_SRAB_DRIVER)  += b53_srab.o
target/linux/mpc85xx/p1020/config-default:# CONFIG_SWCONFIG_B53_SRAB_DRIVER is not set
````

neheb, 2022-07-23:
r4k cache blast patch comes from an old QCA SDK.