openwrt/openwrt #10290: libustream-wolfssl selected by default

lucize, 2022-07-17:
trying to build on x86 without wolfssl seems to fail.
I see in the config that libustream-wolfssl is setup as a default package and will fail if the package  is not actually selected (any of libustream variant will not do)

```
/home/build/staging/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/build/staging/tmp/cc3GCQan.ltrans11.ltrans.o: in function `wolfSSL_BN_generate_prime_ex':
<artificial>:(.text+0x4275): undefined reference to `sp_rand_prime'
/home/build/staging/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: src/.libs/libwolfssl.so.5.4.0.d242c60a: hidden symbol `sp_rand_prime' isn't defined
/home/build/staging/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[6]: *** [Makefile:4454: src/libwolfssl.la] Error 1
```
guess we don't actually have to enforce this ?

Vladdrako, 2022-07-18:
Error caused by wolfssl 5.4.0 update. Reverting fixed the issue. But yeah, I don't understand what wolfssl enforces even though it is not used anywhere. I edited include/target.mk to use libustream-openssl but it didn't resolve the problem.

jwmullally, 2022-07-19:
Looks related? Building with the default profile options on x86_64 gives this error:

```
$ curl -LO https://downloads.openwrt.org/releases/22.03.0-rc5/targets/x86/64/openwrt-imagebuilder-22.03.0-rc5-x86-64.Linux-x86_64.tar.xz
$ tar -xvf openwrt-imagebuilder-22.03.0-rc5-x86-64.Linux-x86_64.tar.xz
$ cd openwrt-imagebuilder-22.03.0-rc5-x86-64.Linux-x86_64
```

```
$ make info
Current Target: "x86/64"
Current Architecture: "x86_64"
Current Revision: "r19523-bfd070e7fa"
Default Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-wolfssl logd mtd netifd opkg uci uclient-fetch urandom-seed urngd busybox procd procd-ujail procd-seccomp partx-utils mkf2fs e2fsprogs kmod-button-hotplug dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe
Available Profiles:

generic:
    Generic x86/64
    Packages: kmod-amazon-ena kmod-amd-xgbe kmod-bnx2 kmod-e1000e kmod-e1000 kmod-forcedeth kmod-fs-vfat kmod-igb kmod-igc kmod-ixgbe kmod-r8169 kmod-tg3
    hasImageMetadata: 0
```

```
$ make image PROFILE="generic"
Unknown package 'libustream-wolfssl'.
...
 * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for libustream-wolfssl20201210
 * pkg_hash_fetch_best_installation_candidate: Packages for libustream-wolfssl found, but incompatible with the architectures configured
  * opkg_install_cmd: Cannot install package libustream-wolfssl.
make[2]: *** [Makefile:169: package_install] Error 255
```

Borromini, 2022-07-21:
I just ran into this issue as well with what used to be just working before (22.03 HEAD). Always fun updates, those WolfSSL bumps...

`OpenWrt-libtool: link: mipsel-openwrt-linux-musl-gcc -shared  -fPIC -DPIC  wolfcrypt/src/.libs/src_libwolfssl_la-hmac.o wolfcrypt/src/.libs/src_libwolfssl_la-hash.o wolfcrypt/src/.libs/src_libwolfssl_la-cpuid.o wolfcrypt/src/.libs/src_libwolfssl_la-kdf.o wolfcrypt/src/.libs/src_libwolfssl_la-random.o wolfcrypt/src/.libs/src_libwolfssl_la-sha256.o wolfcrypt/src/.libs/src_libwolfssl_la-rsa.o wolfcrypt/src/.libs/src_libwolfssl_la-sp_int.o wolfcrypt/src/.libs/src_libwolfssl_la-aes.o wolfcrypt/src/.libs/src_libwolfssl_la-des3.o wolfcrypt/src/.libs/src_libwolfssl_la-sha.o wolfcrypt/src/.libs/src_libwolfssl_la-sha512.o wolfcrypt/src/.libs/src_libwolfssl_la-logging.o wolfcrypt/src/.libs/src_libwolfssl_la-wc_port.o wolfcrypt/src/.libs/src_libwolfssl_la-error.o wolfcrypt/src/.libs/src_libwolfssl_la-wc_encrypt.o wolfcrypt/src/.libs/src_libwolfssl_la-signature.o wolfcrypt/src/.libs/src_libwolfssl_la-wolfmath.o wolfcrypt/src/.libs/src_libwolfssl_la-memory.o wolfcrypt/src/.libs/src_libwolfssl_la-asn.o wolfcrypt/src/.libs/src_libwolfssl_la-coding.o wolfcrypt/src/.libs/src_libwolfssl_la-md4.o wolfcrypt/src/.libs/src_libwolfssl_la-md5.o wolfcrypt/src/.libs/src_libwolfssl_la-pwdbased.o wolfcrypt/src/.libs/src_libwolfssl_la-pkcs12.o wolfcrypt/src/.libs/src_libwolfssl_la-ecc.o src/.libs/src_libwolfssl_la-internal.o src/.libs/src_libwolfssl_la-wolfio.o src/.libs/src_libwolfssl_la-keys.o src/.libs/src_libwolfssl_la-ssl.o src/.libs/src_libwolfssl_la-tls.o src/.libs/src_libwolfssl_la-ocsp.o src/.libs/src_libwolfssl_la-crl.o   -L/var/media/data/code/lede/22.03/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/usr/lib -L/var/media/data/code/lede/22.03/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/lib -lpthread  -pthread -O2 -Os -mno-branch-likely -mips32r2 -mtune=24kc -msoft-float -fstack-protector -Wl,-z -Wl,now -Wl,-z -Wl,relro -flto -flto   -pthread -Wl,-soname -Wl,libwolfssl.so.5.4.0.fce64776 -o src/.libs/libwolfssl.so.5.4.0.fce64776
/var/media/data/code/lede/22.03/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/lib/gcc/mipsel-openwrt-linux-musl/11.2.0/../../../../mipsel-openwrt-linux-musl/bin/ld: /var/media/data/code/lede/22.03/tmp/ccqBeaTy.ltrans11.ltrans.o: in function `wolfSSL_BN_generate_prime_ex':
<artificial>:(.text+0x66ec): undefined reference to `sp_rand_prime'
/var/media/data/code/lede/22.03/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/lib/gcc/mipsel-openwrt-linux-musl/11.2.0/../../../../mipsel-openwrt-linux-musl/bin/ld: <artificial>:(.text+0x66f8): undefined reference to `sp_rand_prime'
/var/media/data/code/lede/22.03/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/lib/gcc/mipsel-openwrt-linux-musl/11.2.0/../../../../mipsel-openwrt-linux-musl/bin/ld: src/.libs/libwolfssl.so.5.4.0.fce64776: hidden symbol `sp_rand_prime' isn't defined
/var/media/data/code/lede/22.03/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/lib/gcc/mipsel-openwrt-linux-musl/11.2.0/../../../../mipsel-openwrt-linux-musl/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:4454: src/libwolfssl.la] Error 1
make[5]: Leaving directory '/var/media/data/code/lede/22.03/build_dir/target-mipsel_24kc_musl/wolfssl-5.4.0-stable'
make[4]: *** [Makefile:6558: all-recursive] Error 1
make[4]: Leaving directory '/var/media/data/code/lede/22.03/build_dir/target-mipsel_24kc_musl/wolfssl-5.4.0-stable'
make[3]: *** [Makefile:3794: all] Error 2
make[3]: Leaving directory '/var/media/data/code/lede/22.03/build_dir/target-mipsel_24kc_musl/wolfssl-5.4.0-stable'
make[2]: *** [Makefile:172: /var/media/data/code/lede/22.03/build_dir/target-mipsel_24kc_musl/wolfssl-5.4.0-stable/.built] Error 2
make[2]: Leaving directory '/var/media/data/code/lede/22.03/package/libs/wolfssl'
time: package/libs/wolfssl/compile#26.16#1.87#28.02
    ERROR: package/libs/wolfssl failed to build.
make[1]: *** [package/Makefile:116: package/libs/wolfssl/compile] Error 1`

borkra, 2022-07-22:
See here for fixes: https://github.com/openwrt/openwrt/issues/10311

lucize, 2022-07-22:
#10322

closed 2022-07-22 by lucize