openwrt/openwrt #10293: nanopi r2s usb Cannot enable. Maybe the USB cable is bad?

DRTEK, 2022-07-17:
Hello ,
on some nanopi r2s devices the second ethernet (lan usb side) is not recognized. when i compile the firmware with jayanta fork or friendlywrt  it works but with openwrt repos it won't .
while some nanopi devices same openwrt firmware just works ,their firmware seems to be working for all devices.
can you tell me what feature openwrt branches are missing for some nanopi devices
i am attaching original openwrt kernel messages and  @jayanta525  's version for your reference ,
thank you in advance

OPENWRT master branch firmware

[ 0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[ 0.000000] Linux version 5.15.53 (node@devel) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r20116-9710fe70a6) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT Sat Jul 16 20:22:40 2022
[ 0.000000] Machine model: FriendlyElec NanoPi R2S
[ 0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
[ 0.000000] printk: bootconsole [uart8250] enabled
[ 0.000000] Zone ranges:
[ 0.000000] DMA [mem 0x0000000000200000-0x000000003fffffff]
[ 0.000000] DMA32 empty
[ 0.000000] Normal empty
[ 0.000000] Movable zone start for each node
[ 0.000000] Early memory node ranges
[ 0.000000] node 0: [mem 0x0000000000200000-0x000000003fffffff]
[ 0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000003fffffff]
[ 0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[ 0.000000] cma: Reserved 8 MiB at 0x000000003e400000
[ 0.000000] psci: probing for conduit method from DT.
[ 0.000000] psci: PSCIv1.1 detected in firmware.
[ 0.000000] psci: Using standard PSCI v0.2 function IDs
[ 0.000000] psci: MIGRATE_INFO_TYPE not supported.
[ 0.000000] psci: SMC Calling Convention v1.2
[ 0.000000] percpu: Embedded 17 pages/cpu s32216 r8192 d29224 u69632
[ 0.000000] pcpu-alloc: s32216 r8192 d29224 u69632 alloc=174096
[ 0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[ 0.000000] Detected VIPT I-cache on CPU0
[ 0.000000] CPU features: detected: ARM erratum 845719
[ 0.000000] Built 1 zonelists, mobility grouping on. Total pages: 257544
[ 0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
[ 0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[ 0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[ 0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[ 0.000000] Memory: 1001948K/1046528K available (8512K kernel code, 2366K rwdata, 2416K rodata, 1664K init, 720K bss, 36388K reserved, 8192K cma-reserved)
[ 0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[ 0.000000] rcu: Preemptible hierarchical RCU implementation.
[ 0.000000] rcu: RCU event tracing is enabled.
[ 0.000000] rcu: RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[ 0.000000] Trampoline variant of Tasks RCU enabled.
[ 0.000000] Tracing variant of Tasks RCU enabled.
[ 0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[ 0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[ 0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[ 0.000000] Root IRQ handler: gic_handle_irq
[ 0.000000] GIC: Using split EOI/Deactivate mode
[ 0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[ 0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[ 0.000002] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[ 0.001337] Console: colour dummy device 80x25
[ 0.001811] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[ 0.002757] pid_max: default: 32768 minimum: 301
[ 0.003478] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[ 0.004163] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[ 0.007415] rcu: Hierarchical SRCU implementation.
[ 0.008228] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[ 0.009906] smp: Bringing up secondary CPUs ...
[ 0.011110] Detected VIPT I-cache on CPU1
[ 0.011221] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[ 0.012086] Detected VIPT I-cache on CPU2
[ 0.012166] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[ 0.012991] Detected VIPT I-cache on CPU3
[ 0.013073] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[ 0.013201] smp: Brought up 1 node, 4 CPUs
[ 0.016425] SMP: Total of 4 processors activated.
[... 513 lines cut ...]
[ 15.133388] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[ 20.585071] br-lan: port 1(eth0) entered disabled state
[ 20.590106] device eth0 left promiscuous mode
[ 20.590619] br-lan: port 1(eth0) entered disabled state
[ 20.615757] rk_gmac-dwmac ff540000.ethernet eth0: Link is Down
[ 20.711795] br-023d7ad553a1: port 1(eth1) entered blocking state
[ 20.712392] br-023d7ad553a1: port 1(eth1) entered disabled state
[ 20.713322] device eth1 entered promiscuous mode
[ 20.715184] br-023d7ad553a1: port 1(eth1) entered blocking state
[ 20.715751] br-023d7ad553a1: port 1(eth1) entered forwarding state
[ 20.740650] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:00] driver [RTL8211E Gigabit Ethernet]
[ 20.750984] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[ 20.751732] rk_gmac-dwmac ff540000.ethernet eth0: RX IPC Checksum Offload disabled
[ 20.752428] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[ 20.753049] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[ 20.754779] br-lan: port 1(eth0) entered blocking state
[ 20.756082] br-lan: port 1(eth0) entered disabled state
[ 20.756934] device eth0 entered promiscuous mode
[ 20.758493] br-lan: port 1(eth0) entered blocking state
[ 20.759050] br-lan: port 1(eth0) entered forwarding state
[ 21.595101] br-lan: port 1(eth0) entered disabled state
[ 21.596553] br-023d7ad553a1: port 1(eth1) entered disabled state
[ 25.883545] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[ 25.884453] br-lan: port 1(eth0) entered blocking state
[ 25.884949] br-lan: port 1(eth0) entered forwarding stateHello ,

AntonyFl, 2022-07-21:
I was having same problems on my NanoPi R2S with the testing kernel 5.15, I fix it with this patch https://github.com/openwrt/openwrt/pull/10123
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.53 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r20151+3-a0b7fef0ff) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT Wed Jul 20 23:50:59 2022
[    0.000000] Machine model: FriendlyElec NanoPi R2S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 8 MiB at 0x000000003e400000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 17 pages/cpu s32728 r8192 d28712 u69632
[    0.000000] pcpu-alloc: s32728 r8192 d28712 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1001372K/1046528K available (8896K kernel code, 2388K rwdata, 2520K rodata, 1728K init, 737K bss, 36964K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000002] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001359] Console: colour dummy device 80x25
[    0.001848] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.002794] pid_max: default: 32768 minimum: 301
[    0.003513] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004201] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.007434] rcu: Hierarchical SRCU implementation.
[    0.008250] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.009948] smp: Bringing up secondary CPUs ...
[    0.011138] Detected VIPT I-cache on CPU1
[    0.011248] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.012101] Detected VIPT I-cache on CPU2
[    0.012180] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.012993] Detected VIPT I-cache on CPU3
[    0.013070] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.013209] smp: Brought up 1 node, 4 CPUs
[    0.016434] SMP: Total of 4 processors activated.
[    0.016867] CPU features: detected: 32-bit EL0 Support
[    0.017336] CPU features: detected: CRC32 instructions
[    0.018119] CPU: All CPU(s) started at EL2
[    0.018608] alternatives: patching kernel code
[    0.029479] KASLR disabled due to lack of seed
[    0.030108] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.031016] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[... 199 lines cut ...]
[    5.759473] kmodloader: done loading kernel modules from /etc/modules.d/*
[    5.905565] usb 4-1: reset SuperSpeed USB device number 2 using xhci-hcd
[    5.964315] r8152 4-1:1.0 (unnamed net_device) (uninitialized): Invalid ether addr 00:00:00:00:00:00
[    5.965186] r8152 4-1:1.0 (unnamed net_device) (uninitialized): Random ether addr ba:c8:93:71:68:2a
[    6.071802] r8152 4-1:1.0: load rtl8153b-2 v1 10/23/19 successfully
[    6.127274] r8152 4-1:1.0 eth1: v1.12.12
[   17.547269] br-lan: port 1(eth1) entered blocking state
[   17.547801] br-lan: port 1(eth1) entered disabled state
[   17.549091] device eth1 entered promiscuous mode
[... cut ...]
[   18.565836] br-lan: port 1(eth1) entered disabled state
[   20.806525] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   20.807411] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   21.189822] r8152 4-1:1.0 eth1: Promiscuous mode enabled
[   21.190642] r8152 4-1:1.0 eth1: carrier on
[   21.191440] br-lan: port 1(eth1) entered blocking state
[   21.191913] br-lan: port 1(eth1) entered forwarding state
[   21.192769] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
```

DRTEK, 2022-07-21:
Hello @AntonyFl 
i tried the patch but nothing changed on my side 
would you share your bin file to check it again ?
here is my kernel dmesg output.
thanks 

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.53 (node@devel) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r20151-a0b7fef0ff) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT Wed Jul 20 19:52:06 2022
[    0.000000] Machine model: FriendlyElec NanoPi R2S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 8 MiB at 0x000000003e400000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 17 pages/cpu s32216 r8192 d29224 u69632
[    0.000000] pcpu-alloc: s32216 r8192 d29224 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1001884K/1046528K available (8576K kernel code, 2368K rwdata, 2432K rodata, 1664K init, 722K bss, 36452K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001341] Console: colour dummy device 80x25
[    0.001817] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.002764] pid_max: default: 32768 minimum: 301
[    0.003474] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004162] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.007390] rcu: Hierarchical SRCU implementation.
[    0.008122] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.009899] smp: Bringing up secondary CPUs ...
[    0.011095] Detected VIPT I-cache on CPU1
[    0.011207] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.012066] Detected VIPT I-cache on CPU2
[    0.012144] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.012976] Detected VIPT I-cache on CPU3
[    0.013055] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.013184] smp: Brought up 1 node, 4 CPUs
[    0.016406] SMP: Total of 4 processors activated.
[    0.016839] CPU features: detected: 32-bit EL0 Support
[    0.017308] CPU features: detected: CRC32 instructions
[    0.018090] CPU: All CPU(s) started at EL2
[... 191 lines cut ...]
[    5.539745] procd: - init -
[    5.834227] urngd: v1.0.2 started.
[    5.858140] random: crng init done
[    5.858483] random: 22 urandom warning(s) missed due to ratelimiting
[    5.894611] kmodloader: loading kernel modules from /etc/modules.d/*
[    5.928033] GACT probability on
[    5.929642] Mirror/redirect action on
[    5.939120] u32 classifier
[    5.939395]     input device check on
[    5.939718]     Actions configured
[    5.951384] Simple TC action Loaded
[    6.000265] usbcore: registered new interface driver r8152
[    6.014369] xt_time: kernel timezone is -0000
[    6.016402] usbcore: registered new interface driver cdc_ether
[    6.018604] usbcore: registered new interface driver cdc_ncm
[    6.068006] PPP generic driver version 2.4.2
[    6.070611] NET: Registered PF_PPPOX protocol family
[    6.089415] kmodloader: done loading kernel modules from /etc/modules.d/*
[... cut ...]
[   17.289887] rk_gmac-dwmac ff540000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   17.291790] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[   17.292509] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[   17.293791] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[   21.446235] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   21.447087] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

AntonyFl, 2022-07-21:
Hello @DRTEK this is my [bin ](https://github.com/openwrt/openwrt/files/9161479/openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-squashfs-sysupgrade.img.gz) file, maybe something is messed up in your buildroot , did you try with a fresh clone and a new config?

DRTEK, 2022-07-22:
Hello @AntonyFl 
unfortunately your firmware did not make any changes .
thank you 

[    1.000210] init: Console is alive
[    1.166515] kmodloader: loading kernel modules from /etc/modules-boot.d/*
[    1.179007] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
[    1.183781] init: - preinit -
[    1.485719] usb usb3-port1: Cannot enable. Maybe the USB cable is bad?
[    1.657692] random: jshn: uninitialized urandom read (4 bytes read)
[    1.716775] random: jshn: uninitialized urandom read (4 bytes read)
[    1.770021] random: jshn: uninitialized urandom read (4 bytes read)
Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
[    2.775655] usb usb3-port1: Cannot enable. Maybe the USB cable is bad?
[    2.776608] usb usb3-port1: attempt power cycle
[    3.915828] usb usb3-port1: Cannot enable. Maybe the USB cable is bad?

AntonyFl, 2022-07-23:
@DRTEK  Problem seems to be the SD card loads very slow and unable to load the _xhci-hcd_ driver.
```
[ 0.541064] dwmmc_rockchip ff500000.mmc: DW MMC controller at irq 33,32 bit host data width,256 deep fifo
[ 0.543014] dw-apb-uart ff130000.serial: forbid DMA for kernel console
[ 0.561354] mmc_host mmc0: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[ 0.575721] Waiting for root device PARTUUID=5452574f-02...
[ 0.614368] mmc_host mmc0: Bus speed (slot 0) = 50000000Hz (slot req 50000000Hz, actual 50000000HZ div = 0)
[ 0.615527] mmc0: new high speed SDHC card at address 0001
[ 0.617719] mmcblk0: mmc0:0001 SD 3.70 GiB
[ 0.620607] mmcblk0: p1 p2
[ 0.654568] VFS: Mounted root (squashfs filesystem) readonly on device 179:2
```
on my R2S
```
[    0.563201] dwmmc_rockchip ff500000.mmc: Version ID is 270a
[    0.564284] dwmmc_rockchip ff500000.mmc: DW MMC controller at irq 33,32 bit host data width,256 deep fifo
[    0.583616] mmc_host mmc0: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[    0.596508] Waiting for root device PARTUUID=5452574f-02...
[    0.655874] usb 4-1: new SuperSpeed USB device number 2 using xhci-hcd
[    0.665407] mmc_host mmc0: Bus speed (slot 0) = 150000000Hz (slot req 150000000Hz, actual 150000000HZ div = 0)
[    1.093943] dwmmc_rockchip ff500000.mmc: Successfully tuned phase to 12
[    1.094601] mmc0: new ultra high speed SDR104 SDHC card at address b368
[    1.096954] mmcblk0: mmc0:b368 LX32G 29.5 GiB 
[    1.099613]  mmcblk0: p1 p2
[    1.113182] VFS: Mounted root (squashfs filesystem) readonly on device 179:2.
```

DRTEK, 2022-07-28:
Hello @AntonyFl 
i changed the sd card unfortunately same output :(