openwrt/openwrt #10302: OpenWrt x86_64 22.03/ Details for package libcurl4

karafen, 2022-07-19:
Hi guys! How to solve the problem :
Unknown package 'libcurl4'.
Errors
Collected errors:
  * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for libcurl4
  * pkg_hash_fetch_best_installation_candidate: Packages for libcurl4 found, but incompatible with the architectures configured
  * opkg_install_cmd: Cannot install package libcurl4.?
OpenWrt x86_64 22.03/

matfra, 2022-07-19:
I'm having the same issue on aarch64
the hash is seems to be wrong either on libcurl4 dependency or libwolfssl

```
root@router:/etc# cat banner
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 22.03.0-rc4, r19426-2b1941e47d
 -----------------------------------------------------
root@router:/etc# opkg list-installed |grep libwolf
libwolfssl5.3.0.ee39414e - 5.3.0-stable-5
libwolfssl5.4.0.e624513f - 5.4.0-stable-1
root@router:/etc# opkg install curl
Package curl (7.83.1-1) installed in root is up to date.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for libcurl4
 * pkg_hash_fetch_best_installation_candidate: Packages for libcurl4 found, but incompatible with the architectures configured
root@router:/etc#
```

Lochnair, 2022-07-20:
Seeing the same behavior with ImageBuilder for a mt7621 target with rc5:

```
make manifest PROFILE="zyxel_nr7101"

...

Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for libustream-wolfssl20201210
 * pkg_hash_fetch_best_installation_candidate: Packages for libustream-wolfssl found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package libustream-wolfssl.
 * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for wpad-basic-wolfssl
 * pkg_hash_fetch_best_installation_candidate: Packages for wpad-basic-wolfssl found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package wpad-basic-wolfssl.
```

f4nff, 2022-07-21:
Seeing the same behavior with ImageBuilder for a x86 target with rc5:

```
root@OpenWrt:~# opkg install curl
Installing curl (7.83.1-2) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/packages/x86_64/packages/curl_7.83.1-2_x86_64.ipk
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for libcurl4
 * pkg_hash_fetch_best_installation_candidate: Packages for libcurl4 found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for curl:
 *      libwolfssl5.4.0.ee39414e
 * opkg_install_cmd: Cannot install package curl.
root@OpenWrt:~# opkg list-installed |grep libwolf
libwolfssl5.3.0.ee39414e - 5.3.0-stable-6
```

TDL3, 2022-07-21:
same here with the profile "friendlyarm_nanopi-r2s"

```
collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for libcurl4
 * pkg_hash_fetch_best_installation_candidate: Packages for libcurl4 found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for curl:
 * 	libwolfssl5.4.0.ee39414e
 * opkg_install_cmd: Cannot install package curl.
```

335maria335, 2022-07-22:
ramips/mt7621
22.03.0-rc5
[image]

ouinouin, 2022-07-25:
same issue on ath79 , target gl-mifi

Neustradamus, 2022-07-30:
curl specified PR:
- For master: https://github.com/openwrt/packages/pull/19008
- For 22.03: https://github.com/openwrt/packages/pull/19058

-------------------

Linked to:

Issues:

OpenWrt core:
- https://github.com/openwrt/openwrt/issues/10290
- https://github.com/openwrt/openwrt/issues/10311
- https://github.com/openwrt/openwrt/issues/10324

OpenWrt packages:
- https://github.com/openwrt/packages/issues/19019

wolfSSL:
- https://github.com/wolfSSL/wolfssl/issues/5386

----------------

PRs:

OpenWrt core:
- https://github.com/openwrt/openwrt/pull/10317 -> https://github.com/openwrt/openwrt/commit/c2aa816f28e0fe2f6f77d0c6da4eba19ea8db4ea
- https://github.com/openwrt/openwrt/pull/10322 -> https://github.com/openwrt/openwrt/commit/00718b9d7a73587152f9330347995bc24ab1d027 + https://github.com/openwrt/openwrt/commit/d94c94d79542558d91c5a5de8d5f594dbafcf3c8

OpenWrt packages:
- https://github.com/openwrt/packages/pull/19021

wolfSSL:
- https://github.com/wolfSSL/wolfssl/pull/5390

ArrEssJay, 2022-08-01:
Same problem installing hostapd:

```
Installing hostapd (2022-01-16-cff80b4f-9) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/packages/arm_cortex-a9/base/hostapd_2022-01-16-cff80b4f-9_arm_cortex-a9.ipk
Installing hostapd-common (2022-01-16-cff80b4f-9) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/packages/arm_cortex-a9/base/hostapd-common_2022-01-16-cff80b4f-9_arm_cortex-a9.ipk
Configuring hostapd-common.
Configuring hostapd.

Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for hostapd-basic-wolfssl
 * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for hostapd-wolfssl
 * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for wpad-basic-wolfssl
 * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for wpad-mesh-wolfssl
 * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for wpad-wolfssl
 ```

aparcar, 2022-08-07:
Fixed via rc6

closed 2022-08-07 by aparcar

schuettecarsten, 2022-08-18:
Same problem in snapshot...?