openwrt/openwrt #10303: hostapd does not support mbedtls

neheb, 2022-07-19:
Which necessitates the use of buggy wolfSSL by default

@gstrauss has a WIP branch: https://github.com/gstrauss/hostap/tree/mbedtls

I think this should be researched.

gstrauss, 2022-07-19:
My test branch may be far enough along for a *test package* to be made for hostapd-mini, wpad-mini, and wpa_supplicant-mini variants, but please note that my test branch has not yet been reviewed by others.  I would like to get some feedback from the hostap mailing list when I am a little further along.

Aside: wolfSSL is a solid, open-source library, though like all other software, can have bugs.  (Hopefully) friendly competition in open source can help improve the quality of the open source offerings.

My personal preference for a base TLS package is for mbed TLS when given a choice between mbed TLS and wolfSSL, though keep in mind that wolfSSL has TLSv1.3 and OCSP stapling support, while mbed TLS currently does not (TLSv1.3 support is currently being developed and is experimental in mbed TLS).

gstrauss, 2022-07-19:
Question: https://github.com/openwrt/openwrt/blob/master/package/libs/mbedtls/Makefile#L91
`TARGET_CFLAGS := $(filter-out -O%,$(TARGET_CFLAGS))`
appears to filter out optimization flags.  Is the library built with any optimization?  Can it be at least `-Os`?

gstrauss, 2022-07-21:
My *development* branch https://github.com/gstrauss/hostap/tree/mbedtls now contains a *mostly* functional hostapd and wpa_suppplicant using mbedtls, including SAE and DPP.  Almost all tests in the tests/hwsim test suite pass on a Debian 11 Bullseye VM in which I manually upgraded to mbedtls 2.28.1.  (I have not yet tested with mbedtls 3.2.1, but expect any needed adjustments to be minor since I have written the code to be aware of both mbedtls 2.x and mbedtls 3.x APIs.)

**It should now be feasible to build *test* packages of hostapd-mini, hostapd-basic, and hostapd-full using mbedtls.**

I have created branch https://github.com/gstrauss/hostap/tree/mbedtls.0 as a checkpoint, since I plan to continue developing (and force-pushing any bug fixes to) https://github.com/gstrauss/hostap/tree/mbedtls

Besides code review/audit by others and more testing, there is plenty more polishing to be done, as well as additional configuration options which might need to be added before this is ready for production.  That said, the current state of my development branch proves mbedtls is a viable TLS option for hostapd and wpa_supplicant.

I'll be at a conference and away for the next few days, but plan to post to the hostapd mailing list early next week, if not sooner.

In the meantime, may I request some assistance from others in identifying feature gaps that should be implemented for OpenWRT use of mbedtls with hostapd and wpa_supplicant?  Thank you!

stokito, 2022-07-23:
Maybe it worth to send a letter in answer to the [[PATCH] Add support for mbedtls crypto library for STA mode](https://www.spinics.net/lists/hostap/msg09799.html) so that other interested developers can join

gstrauss, 2022-07-23:
@stokito  did you read what I posted above?
> I'll be at a conference and away for the next few days, but plan to post to the hostapd mailing list early next week, if not sooner.

Neustradamus, 2022-07-30:
To follow this ticket

gstrauss, 2022-07-30:
@Neustradamus in the future, please click the "Subscribe" button in the right sidebar rather than adding useless noise in your comments.

gstrauss, 2022-07-30:
@Neustradamus in the future, please click the "Subscribe" button in the right sidebar to avoid adding useless noise in a comment.

gstrauss, 2022-07-31:
status: further testing shows more work is needed to get EAP-PWD, EAP-TEAP, SAE, DPP and others working.

I previously wrote:
>  I would like to get some feedback from the hostap mailing list when I am a little further along.

and
> I'll be at a conference and away for the next few days, but plan to post to the hostapd mailing list early next week, if not sooner.

I am going to defer posting to the hostap mailing list until next week, as I try to get more things working in tests/hwsim test suite.

gstrauss, 2022-09-14:
Posted to hostap mailing list: http://lists.infradead.org/pipermail/hostap/2022-September/040794.html

gstrauss, 2022-09-14:
Almost 2 months ago (above https://github.com/openwrt/openwrt/issues/10303#issuecomment-1191054221) I posted the following request **but I have received zero feedback**:

> In the meantime, may I request some assistance from others in identifying feature gaps that should be implemented for OpenWRT use of mbedtls with hostapd and wpa_supplicant? Thank you!

neheb, 2022-09-14:
@gstrauss unfortunate. I think most of the core developers talk on the openwrt-devel mailing list.

gstrauss, 2022-09-14:
@neheb since you raised this issue "hostapd does not support mbedtls" (at my request), would you please post to the openwrt-devel mailing list to raise this issue?

mbedtls already is a core piece of openwrt as mbedtls provides TLS for numerous applications.  mbedtls in openwrt should be better maintained, at least similar to openssl, where patches on the stable branch are incorporated on a periodic basis, where there period to pick up patch releases is measured in weeks, not years.

My development patches for hostap to be able to use mbedtls currently require at least mbedtls 2.27.0 (released 6 Jul 2021)

Latest release on the mbedtls 2.x series is mbedtls 2.28.1 (released 11 Jul 2022)

https://github.com/Mbed-TLS/mbedtls/releases
> Mbed TLS 2.28 is a long-time support branch. It will be supported with bug-fixes and security fixes until end of 2024.

ynezz, 2022-09-15:
> may I request some assistance from others in identifying feature gaps that should be implemented for OpenWRT use of mbedtls with hostapd and wpa_supplicant

IMO the best way to move forward is to actually create something actionable for folks interested in development, so this would be pull request or patch series, which would incorporate needed hostapd/mbedtls changes, so anyone interested with helping/testing can easily join the effort. This process should very well expose the missing bits as well.

gstrauss, 2022-09-15:
@ynezz: I have patches at https://github.com/gstrauss/hostap/tree/mbedtls.  Are you suggesting creating a patch set and submitting patches to openwrt?  That is trivially done with some git commands, but then what?

hostap is created in multiple flavors for openwrt.  If you are asking for something trivial for others to more easily test, then I ask for more detailed guidance from openwrt packages about the best way to go about this.

You see, I have spent hundreds of hours of professional development time and have provided patches in a canonical form: a git repository.  For anyone interested in development who has the skills for development, a git repository is sufficient.

I think you are aiming at something different, which why I am asking for further assistance in packaging to make it easier for the target audience you have in mind.  Thanks.

neheb, 2022-09-15:
I think what he means is, people look at PRs, pull, and then test. So far, there is no such PR. I can create such a thing if desired.

gstrauss, 2022-09-15:
If I can get some assistance with openwrt-specific packaging to create an mbedtls "flavor" of hostap for openwrt, I'll be happy to take that starting point and add hostap patches from my development branch.

ynezz, 2022-09-15:
> So far, there is no such PR. I can create such a thing if desired.

Awesome, that would be a proper way for moving forward, thank you!

So I'm closing this this as this is not a bug (this is a bug tracker) and I hope, that it's now clear what needs to be done. So lets move any further discussion about this topic into appropriate channels, like that future PR, mailing list or forum.

ynezz, 2022-09-15:
> If I can get some assistance with openwrt-specific packaging to create an mbedtls "flavor" of hostap for openwrt, I'll be happy to take that starting point and add hostap patches from my development branch.

I think, that @neheb has a lot of experience in this area, so he is likely the best candidate for that,

closed 2022-09-15 by ynezz

ynezz, 2022-09-15:
> You see, I have spent hundreds of hours of professional development time and have provided patches in a canonical form: a git repository.

Few years ago we've estimated the development time needed for 6-12 months, so thats why we went with wolfSSL which was already available.

> For anyone interested in development who has the skills for development, a git repository is sufficient.

As in any FOSS project, it's not about lack of interest, but rather lack of time.

gstrauss, 2022-09-15:
> As in any FOSS project, it's not about lack of interest, but rather lack of time.

...and good project management and communication, which has now been dead-ended by you closing this issue when there is no currently-established PR for reference for the next steps.  Therefore, I'll continue posting here.

I have rebased my patches onto a branch: https://github.com/gstrauss/hostap/tree/openwrt-hostap-vers
for which I intended to follow the hostap PKG_SOURCE_VERSION used in https://github.com/openwrt/openwrt/tree/master/package/network/services/hostapd/Makefile

That should make it easier to produce patche files for openwrt using a command like `PKG_SOURCE_VERSION=4383528e01955d995d3b3db201e4c0f9840e8236; git format-patch $PKG_SOURCE_VERSION..HEAD`

neheb, 2022-09-15:
Current WIP: https://github.com/neheb/openwrt/commit/459aa77c2160ec519dc02e41f861898465950356

Doesn't quite work:

```
echo ` 	CFLAGS="-I/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/libnl-tiny -I/home/mangix/devstuff/openwrt/build_dir/target-mipsel_24kc_musl/hostapd-wpad-full-mbedtls/hostapd-2022-06-02-4383528e/src/crypto -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/usr/include -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/include/fortify -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/include -DCONFIG_LIBNL20 -D_GNU_SOURCE -DCONFIG_MSG_MIN_PRIORITY=3 -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/mangix/devstuff/openwrt/build_dir/target-mipsel_24kc_musl/hostapd-wpad-full-mbedtls/hostapd-2022-06-02-4383528e=hostapd-2022-06-02-4383528e -Wformat -Werror=format-security -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -flto" make  -C /home/mangix/devstuff/openwrt/build_dir/target-mipsel_24kc_musl/hostapd-wpad-full-mbedtls/hostapd-2022-06-02-4383528e/hostapd AR="mipsel-openwrt-linux-musl-gcc-ar" AS="ccache_cc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/mangix/devstuff/openwrt/build_dir/target-mipsel_24kc_musl/hostapd-wpad-full-mbedtls/hostapd-2022-06-02-4383528e=hostapd-2022-06-02-4383528e -Wformat -Werror=format-security -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -flto" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CONFIG_ACS=y CONFIG_DRIVER_NL80211=y CONFIG_IEEE80211N=y CONFIG_IEEE80211AC=y CONFIG_IEEE80211AX=y CONFIG_DRIVER_WEXT= CONFIG_MBO=y LIBS="-L/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/usr/lib -L/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/lib -DPIC -fpic -specs=/home/mangix/devstuff/openwrt/include/hardened-ld-pie.specs -znow -zrelro -Wl,--gc-sections -flto=jobserver -fuse-linker-plugin -lubox -lubus -lm -lnl-tiny" LIBS_c="" AR="mipsel-openwrt-linux-musl-gcc-ar" BCHECK= V=1 -s MULTICALL=1 dump_cflags; 	CFLAGS="-I/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/libnl-tiny -I/home/mangix/devstuff/openwrt/build_dir/target-mipsel_24kc_musl/hostapd-wpad-full-mbedtls/hostapd-2022-06-02-4383528e/src/crypto -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/usr/include -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/include/fortify -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/include -DCONFIG_LIBNL20 -D_GNU_SOURCE -DCONFIG_MSG_MIN_PRIORITY=3 -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/mangix/devstuff/openwrt/build_dir/target-mipsel_24kc_musl/hostapd-wpad-full-mbedtls/hostapd-2022-06-02-4383528e=hostapd-2022-06-02-4383528e -Wformat -Werror=format-security -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -flto" make  -C /home/mangix/devstuff/openwrt/build_dir/target-mipsel_24kc_musl/hostapd-wpad-full-mbedtls/hostapd-2022-06-02-4383528e/wpa_supplicant AR="mipsel-openwrt-linux-musl-gcc-ar" AS="ccache_cc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/mangix/devstuff/openwrt/build_dir/target-mipsel_24kc_musl/hostapd-wpad-full-mbedtls/hostapd-2022-06-02-4383528e=hostapd-2022-06-02-4383528e -Wformat -Werror=format-security -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -flto" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CONFIG_ACS=y CONFIG_DRIVER_NL80211=y CONFIG_IEEE80211N=y CONFIG_IEEE80211AC=y CONFIG_IEEE80211AX=y CONFIG_DRIVER_WEXT= CONFIG_MBO=y LIBS="-L/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/usr/lib -L/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/lib -DPIC -fpic -specs=/home/mangix/devstuff/openwrt/include/hardened-ld-pie.specs -znow -zrelro -Wl,--gc-sections -flto=jobserver -fuse-linker-plugin -lubox -lubus -lm -lnl-tiny" LIBS_c="" AR="mipsel-openwrt-linux-musl-gcc-ar" BCHECK= V=1 -s MULTICALL=1 dump_cflags | sed -e 's,-n ,,g' -e 's^-Os -pipe -mno-branch-likely -mips32r2  [... cut ...] l-mbedtls/hostapd-2022-06-02-4383528e/.cflags)" MULTICALL=1 hostapd_cli hostapd_multi.a 
make[3]: Entering directory '/home/mangix/devstuff/openwrt/build_dir/target-mipsel_24kc_musl/hostapd-wpad-full-mbedtls/hostapd-2022-06-02-4383528e/hostapd'
make[3]: *** No rule to make target '.config', needed by '/home/mangix/devstuff/openwrt/build_dir/target-mipsel_24kc_musl/hostapd-wpad-full-mbedtls/hostapd-2022-06-02-4383528e/build/hostapd/hostapd_cli.o'.  Stop.
```

Missing something obvious probably...

gstrauss, 2022-09-15:
The Makefile contains references such as: `./files/hostapd-$(CONFIG_VARIANT).config` which suggest the need to create similar ones from mbedtls, perhaps containing CONFIG_TLS=mbedtls

gstrauss, 2022-09-15:
@neheb your Config.in has a `config WPA_WOLFSSL` section but not one for mbedtls.  Is one needed for mbedtls?

neheb, 2022-09-15:
figured it out. Next issue I see is:

```
../src/crypto/tls_wolfssl.c:18:10: fatal error: wolfssl/options.h: No such file or directory
   18 | #include <wolfssl/options.h>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.

```

I see CONFIG_TLS_WOLFSSL. No idea where it's getting set.

config WPA_WOLFSSL is unused. Seems like a leftover.

gstrauss, 2022-09-15:
CONFIG_DES=y might need to be set in the Makefile.

neheb, 2022-09-15:
Didn't help. CONFIG_DES is definitely being passed.

gstrauss, 2022-09-15:
Try `NEED_DES=y`, which should result in -DCONFIG_DES on the gcc command line.  crypto_mbedtls.c checks for `#if defined(CONFIG_DES)` before defining `des_encrypt()`

neheb, 2022-09-15:
here's the full log. https://gist.github.com/neheb/d8c4bf0ca9bff4ad4ca327acc9bda6da

I see it there.

gstrauss, 2022-09-15:
https://github.com/openwrt/openwrt/blob/master/package/libs/mbedtls/patches/200-config.patch#L175

gstrauss, 2022-09-15:
I think you might be able to work around it by setting `CONFIG_INTERNAL_DES=y`, though I would prefer mbedtls to provide it rather than hostap to include its own.

neheb, 2022-09-15:
Added PR: https://github.com/openwrt/openwrt/pull/10727

Let's see how this goes...

gstrauss, 2022-09-15:
FYI: https://github.com/openwrt/openwrt/blob/master/package/libs/mbedtls/patches/200-config.patch#L23 disables quite a few curves in mbedtls.  Some of these might end up being needed for wider support in hostap.  I am not sure at first glance if any are required.

neheb, 2022-09-15:
I assume that's an exercise in size optimization. I don't think there's any hurry.

gstrauss, 2022-09-15:
@neheb: Thank you for putting together and submitting the PR.  It would probably have taken me much longer to work through the openwrt packaging and build issues for the complex hostap package.

neheb, 2022-09-15:
Yeah that's the first time I touch it. It needs serious refactoring. If I had to guess, there was only openssl and internal originally. Then wolfssl was added without refactoring the Makefile. The PR doesn't really fix anything.