openwrt/openwrt #10311: Build error on wrt3200acm since "wolfssl: bump to 5.4.0"

johnjacob10, 2022-07-20:
I build OpenWRT for my wrt3200acm from snapshot and I've been hitting this build error since the commit mentioned in the title.
/var/lib/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.1.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/12.1.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /var/lib/openwrt/tmp/ccTXipBn.ltrans14.ltrans.o: in function `wolfSSL_BN_generate_prime_ex':
<artificial>:(.text+0x404c): undefined reference to `sp_rand_prime'
/var/lib/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.1.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/12.1.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: src/.libs/libwolfssl.so.5.4.0.d242c60a: hidden symbol `sp_rand_prime' isn't defined
/var/lib/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.1.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/12.1.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status

The function sp_rand_prime is defined in wolfcrypt/src/sp_int.c under the following conditionals:
#if defined(WOLFSSL_KEY_GEN) && (!defined(NO_DH) || !defined(NO_DSA)) && \
    !defined(WC_NO_RNG)
/* Generate a random prime for RSA only.
 *
 * @param  [out]  r     SP integer to hold result.
 * @param  [in]   len   Number of bytes in prime.
 * @param  [in]   rng   Random number generator.
 * @param  [in]   heap  Heap hint. Unused.
 *
 * @return  MP_OKAY on success
 * @return  MP_VAL when r or rng is NULL, length is not supported or random
 *          number generator fails.
 */
int sp_rand_prime(sp_int* r, int len, WC_RNG* rng, void* heap)
...

However, we compile sp_int.c with WOLFSSL_KEY_GEN, NO_DH and NO_DSA all defined, so we skip compiling the sp_rand_prime function, leading to the above error.

I've attached the compile log for WolfSSL and my OpenWRT config file.

Please let me know if any other information is required and I'll provide it.

Thanks in advance for looking at this.

Best regards,
- John
[compile.txt](https://github.com/openwrt/openwrt/files/9150474/compile.txt)
[config.txt](https://github.com/openwrt/openwrt/files/9150477/config.txt)

dhewg, 2022-07-20:
Same here on ipq40xx.

graysky2, 2022-07-20:
I did not see this task when I commented on the commit https://github.com/openwrt/openwrt/commit/9710fe70a68e0a004b1906db192d7a6c8f810ac5#comments ... seems to be affecting multiple flavors and is fixed by reverting that commit.

borkra, 2022-07-21:
The problem seems to be related to this wolfssl commit: https://github.com/wolfSSL/wolfssl/commit/6d2a41b9fdfb310602814c523fefd4f7bb8a01d8

Adding --enable-fastmath wolfssl config option fixes the issue.

graysky2, 2022-07-21:
@borkra - I created the PR referenced above based on your feedback.

hauke, 2022-07-21:
Did someone report this to upstream wolfssl?

cotequeiroz, 2022-07-21:
Even before we can get to the wolfSSL build failure, I need to understand why is wolfssl getting built if it is being disabled at the `.config`:
```
# CONFIG_PACKAGE_libwolfssl is not set
```
Something is causing the package to be built without config, so there are no wolfssl config options set in `.config`.  The package should build without any options set, so I'm going to check on that, but we should remove this bogus dependency somewhere, if able.

cotequeiroz, 2022-07-21:
So another way of working around this bug should be to set `CONFIG_PACKAGE_libwolfssl=m`, so that the package defaults are used.

borkra, 2022-07-21:
@cotequeiroz  That is general fault of openwrt build system, on my system build at least 10 extra packages are built and not included into the final image, wolfssl is one of them. The problem been there forever, but lately, became really bad. My guess is if package have variants, dependencies for all variants are built.

borkra, 2022-07-21:
Well, I do not want wolfssl in my image, so enabling fastmath and then not using it is better.

cotequeiroz, 2022-07-21:
I suspected something like that.  It is solvable, most of the times, by having the dependency being selected only if the package is selected.  Sounds redundant, but it works.
I would like to see if there are any side-effects to adding it.
Selecting it as a module will not add it to the image, will it?

borkra, 2022-07-21:
I am not sure, the only thing I know there are lot's of packages affected.

mpratt14, 2022-07-21:
@borkra if you can please try the option `--enable-heapmath` instead of `--enable-fastmath` and let us know if that also fixes the issue

borkra, 2022-07-21:
@mpratt14 Yes, --enable-heapmath fixes this too

borkra, 2022-07-22:
@cotequeiroz Attached patch fixes wolfssl and openssl dependency issue
[wolfssl_unneeded_compile.patch.txt](https://github.com/openwrt/openwrt/files/9163463/wolfssl_unneeded_compile.patch.txt)

mpratt14, 2022-07-22:
> Did someone report this to upstream wolfssl?

done 👍🏼

cotequeiroz, 2022-07-22:
> @cotequeiroz Attached patch fixes wolfssl and openssl dependency issue
> 
> [wolfssl_unneeded_compile.patch.txt](https://github.com/openwrt/openwrt/files/9163967/wolfssl_unneeded_compile.patch.txt)

This is exactly what I meant.  Then you don't need to compile it every time you build your own.  Can you open a PR with your patch?  I'll look at some other packages needing this.   I see right away that `libustream-ssl-mbedtls` is forcing mbedtls to be built as well.

Meanwhile, we still have to take care of the wolfssl problem itself.... 😞

borkra, 2022-07-22:
https://github.com/openwrt/openwrt/pull/10322

borkra, 2022-07-22:
Similar issues in packages: https://github.com/openwrt/packages/pull/19021

kareem-wolfssl, 2022-07-22:
Please try building with https://github.com/wolfSSL/wolfssl/pull/5390 applied and let me know if it helps.

graysky2, 2022-07-22:
> done 👍🏼
For reference: https://github.com/wolfSSL/wolfssl/issues/5386

graysky2, 2022-07-22:
Upstream has an unmerged PR that fixed the build for me without modifying the `Makefile`, https://github.com/wolfSSL/wolfssl/issues/5386

I am happy to amend this PR to include that patch unless upstream tags a new stable release including it.

closed 2022-08-01 by johnjacob10