openwrt/openwrt #10313: [BMIPS]Internal switch cannot coexist with external switch

rikka0w0, 2022-07-20:
I'm porting OpenWRT to Sagemcom F@st 3864OP, because [this patch](https://github.com/openwrt/openwrt/blob/9b4169df2d26ddcb8cea41de47c159d421389130/target/linux/bmips/patches-5.15/501-net-dsa-b53-mmap-fix-enabled_ports.patch) has not yet been included in any tag, my port is based on the latest commit (bc43ad88ed18722c0621fd6dfef0ff68268f4e14) of the master branch as of 2022 07 21.

The image boots if I only enable the internal switch. With this setup, the WAN port is functional, but none of the LAN ports works. With the help of the `mdio` tool, I can read the PHY status of both switches. However, once I enable the external switch, the system hangs during boot:

```
[    2.086792] bcm6368-mdio-mux 107000b0.mdio: Broadcom BCM6368 MDIO mux bus
[    2.095453] bcm53xx 0.1:1e: found switch: BCM53125, rev 4
[    2.101076] bcm53xx 0.1:1e: failed to register switch: -517
[    2.107601] b53-switch 10700000.switch: found switch: BCM63xx, rev 0
[    2.118387] bcm6368-enetsw 1000d800.ethernet: mtd mac d8:d7:75:13:03:72
[    2.248248] bcm7038-wdt 1000009c.watchdog: Registered BCM7038 Watchdog
[    2.261817] NET: Registered PF_INET6 protocol family
[    2.298322] Segment Routing with IPv6
[    2.302295] In-situ OAM (IOAM) with IPv6
[    2.306643] NET: Registered PF_PACKET protocol family
[    2.312635] 8021q: 802.1Q VLAN Support v1.8
[    2.349723] bcm53xx 0.1:1e: found switch: BCM53125, rev 4
[    2.355488] bcm53xx 0.1:1e: failed to register switch: -517
[    2.361827] b53-switch 10700000.switch: found switch: BCM63xx, rev 0
^ Stucked at this line
```

If I disable the internal switch, then the external switch can be successfully recognized. Both switches use the b53 driver, but they cannot co-exist at this moment. [My DTS](https://github.com/rikka0w0/openwrt-fast3864op/blob/7aeec3d1c1fd413c2d526c6f736a8ab25d72cce1/target/linux/bmips/dts/bcm63168-sagem-fast-3864op.dts)

rikka0w0, 2022-11-27:
>   - The BCM53134S external switch is managed via MDIO which isn't
    supported in this target. Therefore it will behave as a dumb switch.

https://github.com/openwrt/openwrt/commit/ff2c96333f418c0c06ebf0bd947b449f579dda7c

The `bcm63xx ` does not support external MDIO. Hopefully, I can make the `bmips` target work.

Noltari, 2023-03-07:
Hi @rikka0w0,

After doing some tests on my Netgear DGND3700v2 I was able to successfully boot my device without hanging, but the switch isn't working yet...
I believe that the problem is that the ethernet and switch drivers are deferred and the kernel tries to probe them several times, and at some point the MMAP switch is accessed without proper initialization.
Therefore, we should probably register the MDIO driver from the b53-mmap switch, as a child of the switch, instead of as its own separate driver.
However, in order to test that it fixes the problem, we can delay the registration of the MDIO driver with the following patch and the device won't hang:
```
--- a/drivers/net/mdio/mdio-mux-bcm6368.c
+++ b/drivers/net/mdio/mdio-mux-bcm6368.c
@@ -177,7 +177,15 @@ static struct platform_driver bcm6368_md
 	.probe	= bcm6368_mdiomux_probe,
 	.remove	= bcm6368_mdiomux_remove,
 };
-module_platform_driver(bcm6368_mdiomux_driver);
+
+int __init bcm6368_mdiomux_init(void)
+{
+	int ret = platform_driver_register(&bcm6368_mdiomux_driver);
+	if (ret)
+		pr_err("bcm6368_mdiomux: Error registering platform driver!\n");
+	return ret;
+}
+late_initcall(bcm6368_mdiomux_init);
 
 MODULE_AUTHOR("Álvaro Fernández Rojas <noltari@gmail.com>");
 MODULE_DESCRIPTION("BCM6368 mdiomux bus controller driver");
```
See:
https://github.com/Noltari/openwrt/commits/bmips-dgnd3700v2-switch
https://github.com/Noltari/openwrt/commit/156f17def3352194badfb67b2849d535a3a8183e

I don't know why I can't get any traffic, but at least I can see that both tags (brcm and legacy) are being added and consumed.
Maybe @ffainelli or @KanjiMonster can help here...

Noltari, 2023-03-09:
> Hi @rikka0w0,
> 
> After doing some tests on my Netgear DGND3700v2 I was able to successfully boot my device without hanging, but the switch isn't working yet... I believe that the problem is that the ethernet and switch drivers are deferred and the kernel tries to probe them several times, and at some point the MMAP switch is accessed without proper initialization. Therefore, we should probably register the MDIO driver from the b53-mmap switch, as a child of the switch, instead of as its own separate driver. However, in order to test that it fixes the problem, we can delay the registration of the MDIO driver with the following patch and the device won't hang:
> 
> ```
> --- a/drivers/net/mdio/mdio-mux-bcm6368.c
> +++ b/drivers/net/mdio/mdio-mux-bcm6368.c
> @@ -177,7 +177,15 @@ static struct platform_driver bcm6368_md
>  	.probe	= bcm6368_mdiomux_probe,
>  	.remove	= bcm6368_mdiomux_remove,
>  };
> -module_platform_driver(bcm6368_mdiomux_driver);
> +
> +int __init bcm6368_mdiomux_init(void)
> +{
> +	int ret = platform_driver_register(&bcm6368_mdiomux_driver);
> +	if (ret)
> +		pr_err("bcm6368_mdiomux: Error registering platform driver!\n");
> +	return ret;
> +}
> +late_initcall(bcm6368_mdiomux_init);
>  
>  MODULE_AUTHOR("Álvaro Fernández Rojas <noltari@gmail.com>");
>  MODULE_DESCRIPTION("BCM6368 mdiomux bus controller driver");
> ```
> 
> See: https://github.com/Noltari/openwrt/commits/bmips-dgnd3700v2-switch [Noltari@156f17d](https://github.com/Noltari/openwrt/commit/156f17def3352194badfb67b2849d535a3a8183e)
> 
> I don't know why I can't get any traffic, but at least I can see that both tags (brcm and legacy) are being added and consumed. Maybe @ffainelli or @KanjiMonster can help here.

After performing more tests with this branch I could find out that the packets going out are fine: both tags (brcm and brcm-legacy) are added. However, when a package is received, only the brcm-legacy tag is present in the network packet, and I couldn't find a way to receive both tags...
I guess that there should be some way to force the BRCM53125 external switch into sending the packet to the BCM63xx internal switch via RGMII with the brcm tag...
Any suggestions @ffainelli @KanjiMonster?

Thanks!

ffainelli, 2023-03-09:
Your set-up is such that the BCM63xx internal switches uses the 6-bytes legacy Broadcom tag, and the external BCM53125 switch uses the 4-bytes Broadcom tag, right?

Broadcom switches using the 4-bytes Broadcom tag format do not support nested Broadcom tags, which is why we have this set of changes to the DSA framework to indicate when such a set-up is found, and to disable the outer most switch tagging:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=4d776482ecc689bdd68627985ac4cb5a6f325953
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=8fab459e69abfd04a66d76423d18ba853fced4ab

What you observe seems to extend to the older switches as well whereby they are discarding the 4-bytes tag. I do not believe there is any way to make this work.

KanjiMonster, 2023-03-10:
I guess the solution would be to add some more heuristics there to in case the master switch has only one port disable it there instead of the stacked switch.

Might also be done in the enetsw driver, deciding to not register to dsa/switchdev if there is only one configured port, and then stay in dumb forwarding mode (basically emulate a standard ethernet mac).

Noltari, 2023-03-10:
So in that case, if I understood it correctly, for those devices with ports on both switches (internal and external), then only 1 of the two switches can be configured in managed mode with DSA.
And therefore we have to choose between VLANs on either the internal or the external switch, right?

KanjiMonster, 2023-03-10:
> So in that case, if I understood it correctly, for those devices with ports on both switches (internal and external), then only 1 of the two switches can be configured in managed mode with DSA.
> And therefore we have to choose between VLANs on either the internal or the external switch, right?

Are there devices that have both? I only remember devices with either an external switch, or using the integrated phys, but never both at the same time.

Maybe there is some way, I remember the Broadcom code for handling the tagging very large and convoluted. Have you verified that the `B53_BRCM_HDR` register is properly setup for the BCM53125 (and it being put into management mode)?

Noltari, 2023-03-10:
> Are there devices that have both? I only remember devices with either an external switch, or using the integrated phys, but never both at the same time.

Yes, see for example the [Sercomm H-500s](https://openwrt.org/toh/sercomm/h500-s#switch)
[image]

> 
> 
> 
> Maybe there is some way, I remember the Broadcom code for handling the tagging very large and convoluted. Have you verified that the `B53_BRCM_HDR` register is properly setup for the BCM53125 (and it being put into management mode)?
No, I haven't verified that, but I assumed that since both tags were added to the packets the switch had been configured properly...
I'll try to check that and report back. However, according to @ffainelli this shouldn't be possible...

Noltari, 2023-03-11:
> Are there devices that have both? I only remember devices with either an external switch, or using the integrated phys, but never both at the same time.
>
> Maybe there is some way, I remember the [...]

@KanjiMonster this is what I get:
```
[    1.615075] b53-switch 10e00000.switch: found switch: BCM63xx, rev 0
[    1.750218] b53-switch 10e00000.switch: b53_vlan_port_needs_forced_tagged: port=4 tag_prot=22 is_cpu=0
[    1.759796] b53-switch 10e00000.switch: b53_vlan_port_needs_forced_tagged: port=8 tag_prot=22 is_cpu=1
[    1.790537] b53-switch 10e00000.switch: b53_enable_cpu_port: B53_CTRL_PAGE.B53_PORT_CTRL(8)=0x1c
[    1.799556] b53-switch 10e00000.switch: b53_brcm_hdr_setup: port=8 tag_en=1
[    1.806775] b53-switch 10e00000.switch: b53_brcm_hdr_setup: B53_CTRL_PAGE.B53_SWITCH_MODE=0x07
[    1.815615] b53-switch 10e00000.switch: b53_brcm_hdr_setup: B53_MGMT_PAGE.B53_GLOBAL_CONFIG=0x80
[    1.824664] b53-switch 10e00000.switch: b53_brcm_hdr_setup: B53_MGMT_PAGE.B53_BRCM_HDR=0x01
[    1.871707] b53-switch 10e00000.switch: Using legacy PHYLIB callbacks. Please migrate to PHYLINK!
[    1.917639] b53-switch 10e00000.switch: Configured port 8 for internal
[    1.927695] DSA: tree 0 setup
[    2.161964] bcm53xx 0.1:1e: found switch: BCM53125, rev 4
[    2.303395] bcm53xx 0.1:1e: b53_vlan_port_needs_forced_tagged: port=0 tag_prot=1 is_cpu=0
[    2.312052] bcm53xx 0.1:1e: b53_vlan_port_needs_forced_tagged: port=1 tag_prot=1 is_cpu=0
[    2.320602] bcm53xx 0.1:1e: b53_vlan_port_needs_forced_tagged: port=2 tag_prot=1 is_cpu=0
[    2.329128] bcm53xx 0.1:1e: b53_vlan_port_needs_forced_tagged: port=3 tag_prot=1 is_cpu=0
[    2.337679] bcm53xx 0.1:1e: b53_vlan_port_needs_forced_tagged: port=4 tag_prot=1 is_cpu=0
[    2.346224] bcm53xx 0.1:1e: b53_vlan_port_needs_forced_tagged: port=5 tag_prot=1 is_cpu=0
[    2.354771] bcm53xx 0.1:1e: b53_vlan_port_needs_forced_tagged: port=6 tag_prot=1 is_cpu=0
[    2.363330] bcm53xx 0.1:1e: b53_vlan_port_needs_forced_tagged: port=7 tag_prot=1 is_cpu=0
[    2.371880] bcm53xx 0.1:1e: b53_vlan_port_needs_forced_tagged: port=8 tag_prot=1 is_cpu=1
[    2.414373] bcm53xx 0.1:1e: b53_enable_cpu_port: B53_CTRL_PAGE.B53_PORT_CTRL(8)=0x1c
[    2.422359] bcm53xx 0.1:1e: b53_brcm_hdr_setup: port=8 tag_en=1
[    2.428714] bcm53xx 0.1:1e: b53_brcm_hdr_setup: B53_CTRL_PAGE.B53_SWITCH_MODE=0x03
[    2.436841] bcm53xx 0.1:1e: b53_brcm_hdr_setup: B53_MGMT_PAGE.B53_GLOBAL_CONFIG=0x80
[    2.445091] bcm53xx 0.1:1e: b53_brcm_hdr_setup: B53_MGMT_PAGE.B53_BRCM_HDR=0x01
[    2.639924] bcm53xx 0.1:1e: Using legacy PHYLIB callbacks. Please migrate to PHYLINK!
[    2.686165] bcm53xx 0.1:1e: Configured port 8 for rgmii
[    2.692540] bcm53xx 0.1:1e lan1 (uninitialized): PHY [dsa-1.0:00] driver [Generic PHY] (irq=POLL)
[    2.706379] bcm53xx 0.1:1e lan2 (uninitialized): PHY [dsa-1.0:01] driver [Generic PHY] (irq=POLL)
[    2.720246] bcm53xx 0.1:1e lan3 (uninitialized): PHY [dsa-1.0:02] driver [Generic PHY] (irq=POLL)
[    2.734011] bcm53xx 0.1:1e lan4 (uninitialized): PHY [dsa-1.0:03] driver [Generic PHY] (irq=POLL)
[    2.747737] bcm53xx 0.1:1e wan (uninitialized): PHY [dsa-1.0:04] driver [Generic PHY] (irq=POLL)
[    2.762737] DSA: tree 1 setup
[    2.766097] b53-switch 10e00000.switch: MDIO mux bus init
```
It seems that the `B53_BRCM_HDR` register is properly configured on both switches...

KanjiMonster, 2023-03-11:
> Yes, see for example the [Sercomm H-500s](https://openwrt.org/toh/sercomm/h500-s#switch) [image]
>
>
> No, I haven't verified that, but I assumed that since both tags were added to the packets the [...]

The tx path is purely software, so seeing both tags is expected, since it is before any switch got its hands on it. You said the packets are properly going out, does that mean you see the expected packets (e.g. ARP requests) going out of the BCM53134's ports?

What you could try is mirroring the IMP port to one of the switch ports/wan ports in the hope this will mirror the full packets including any tags (https://github.com/torvalds/linux/blob/master/drivers/net/dsa/b53/b53_regs.h#L219).

And if this works, you could try a stupid thing and see if it helps if you setup the internal switch to mirror the enabled port's traffic to the IMP port by setting `B53_MIR_CAP_CTL` to `MIRROR_EN | 8` and `B53_IG_MIR_CTL` to `BIT(4) | BIT(3)`. Maybe also set `BLK_NOT_MIR`, to avoid packet duplication.

Noltari, 2023-03-11:
> The tx path is purely software, so seeing both tags is expected, since it is before any switch got its hands on it. You said the packets are properly going out, does that mean you see the expected [...]

Yeah, exactly that, I can see the ARP requests going out of the BCM53125 ports (I'm testing with the DGND3700v2).

> What you could try is mirroring the IMP port to one of the switch ports/wan ports in the hope this will mirror the full packets including any tags [...]

Ok, I will try this and report back.

Noltari, 2023-03-11:
After doing some more tests, I think I'm wrong and that both tags are being received. It's just that the broadcom tag seems to be at a different offset (+4 bytes) or the tag is different :confused:

This is a ping (ICMP packet) sent to the device:
```
0000   4c 60 de 86 52 12 00 50 b6 1a f8 2e 08 00 45 00
0010   00 54 38 65 40 00 40 01 7c f0 c0 a8 02 01 c0 a8
0020   02 02 08 00 3e 2f d1 5f 00 04 a3 e5 0c 64 00 00
0030   00 00 6c 50 0d 00 00 00 00 00 10 11 12 13 14 15
0040   16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25
0050   26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35
0060   36 37
```

And this is what I could see in the logs:
1. brcm legacy tag
```
[  321.171155] brcm_leg_tag_rcv: [88 74 00 37 00 04]
[  321.175980] skb len=98 headroom=142 headlen=98 tailroom=16
[  321.175980] mac=(128,14) net=(142,0) trans=142
[  321.175980] shinfo(txflags=0 nr_frags=0 gso(size=0 type=0 segs=0))
[  321.175980] csum(0x0 ip_summed=0 complete_sw=0 valid=0 level=0)
[  321.175980] hash(0x0 sw=0 l4=0) proto=0x00f8 pkttype=0 iif=2
[  321.204572] dev name=eth0 feat=0x0000000000004000
[  321.209378] skb headroom: 00000000: 00 02 18 10 00 00 00 00 00 00 00 00 00 02 1c 88
[  321.217315] skb headroom: 00000010: 00 02 18 f8 00 02 1d 60 00 02 19 70 00 02 18 80
[  321.225202] skb headroom: 00000020: 00 02 19 ec 00 02 1e 24 00 02 1a a0 00 02 1b 18
[  321.233107] skb headroom: 00000030: 00 02 1a d8 00 02 1b 70 00 00 00 00 00 00 00 08
[  321.241011] skb headroom: 00000040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  321.248849] skb headroom: 00000050: 00 02 2b 34 00 00 00 00 00 00 00 00 00 02 2c 38
[  321.256758] skb headroom: 00000060: 00 02 1f 90 00 02 27 18 00 02 21 40 00 02 23 00
[  321.264661] skb headroom: 00000070: 00 02 21 b0 00 02 35 40 00 02 1e f0 00 02 1f d4
[  321.272534] skb headroom: 00000080: 4c 60 de 86 52 12 00 50 b6 1a f8 2e 88 74
[  321.279854] skb linear:   00000000: 00 37 00 04 81 00 00 00 00 00 04 01 08 00 45 00
[  321.287788] skb linear:   00000010: 00 54 38 65 40 00 40 01 7c f0 c0 a8 02 01 c0 a8
[  321.295670] skb linear:   00000020: 02 02 08 00 3e 2f d1 5f 00 04 a3 e5 0c 64 00 00
[  321.303545] skb linear:   00000030: 00 00 6c 50 0d 00 00 00 00 00 10 11 12 13 14 15
[  321.311455] skb linear:   00000040: 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25
[  321.319292] skb linear:   00000050: 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35
[  321.327202] skb linear:   00000060: 36 37
[  321.331343] skb tailroom: 00000000: 00 00 00 5c 00 00 00 00 00 00 00 00 00 00 00 00
```

2. brcm tag (modified to 8 bytes)
```
[  321.339401] brcm_tag_rcv_ll: [81 00 00 00 00 00 04 01]
[  321.344752] skb len=92 headroom=148 headlen=92 tailroom=16
[  321.344752] mac=(134,14) net=(148,-6) trans=142
[  321.344752] shinfo(txflags=0 nr_frags=0 gso(size=0 type=0 segs=0))
[  321.344752] csum(0x0 ip_summed=0 complete_sw=0 valid=0 level=0)
[  321.344752] hash(0x0 sw=0 l4=0) proto=0x00f8 pkttype=0 iif=3
[  321.373469] dev name=extsw feat=0x0002000000005020
[  321.378359] skb headroom: 00000000: 00 02 18 10 00 00 00 00 00 00 00 00 00 02 1c 88
[  321.386293] skb headroom: 00000010: 00 02 18 f8 00 02 1d 60 00 02 19 70 00 02 18 80
[  321.394224] skb headroom: 00000020: 00 02 19 ec 00 02 1e 24 00 02 1a a0 00 02 1b 18
[  321.402133] skb headroom: 00000030: 00 02 1a d8 00 02 1b 70 00 00 00 00 00 00 00 08
[  321.409978] skb headroom: 00000040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  321.417937] skb headroom: 00000050: 00 02 2b 34 00 00 00 00 00 00 00 00 00 02 2c 38
[  321.425837] skb headroom: 00000060: 00 02 1f 90 00 02 27 18 00 02 21 40 00 02 23 00
[  321.433729] skb headroom: 00000070: 00 02 21 b0 00 02 35 40 00 02 1e f0 00 02 1f d4
[  321.441652] skb headroom: 00000080: 4c 60 de 86 52 12 4c 60 de 86 52 12 00 50 b6 1a
[  321.449497] skb headroom: 00000090: f8 2e 81 00
[  321.454207] skb linear:   00000000: 00 00 00 00 04 01 08 00 45 00 00 54 38 65 40 00
[  321.462125] skb linear:   00000010: 40 01 7c f0 c0 a8 02 01 c0 a8 02 02 08 00 3e 2f
[  321.469962] skb linear:   00000020: d1 5f 00 04 a3 e5 0c 64 00 00 00 00 6c 50 0d 00
[  321.477895] skb linear:   00000030: 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b
[  321.485797] skb linear:   00000040: 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b
[  321.493688] skb linear:   00000050: 2c 2d 2e 2f 30 31 32 33 34 35 36 37
[  321.500538] skb tailroom: 00000000: 00 00 00 5c 00 00 00 00 00 00 00 00 00 00 00 00
```
If I switch to a different port, I can see that the last byte matches exactly the port that I'm using, and I can also see that the previous byte is either `0x04` or `0x20`.
[... 22 lines cut ...]
		return NULL;

	/* We should never see a reserved reason code without knowing how to
	 * handle it
	 */
	if (unlikely(brcm_tag[2] & BRCM_EG_RC_RSVD))
		return NULL;
#endif

	/* Locate which port this is coming from */
	source_port = brcm_tag[7] & BRCM_EG_PID_MASK;

	skb->dev = dsa_master_find_slave(dev, 0, source_port);
	if (!skb->dev)
		return NULL;

	/* Remove Broadcom tag and update checksum */
	skb_pull_rcsum(skb, BRCM_TAG_LEN + BRCM_TAG_LEN);

	dsa_default_offload_fwd_mark(skb);

	return skb;
}
```
I guess that the tag should be the last 4 bytes, but I can't figure out the first 4 bytes (81 00 00 00) which aren't coming from the packet...

Noltari, 2023-03-11:
Okay, so I ended up modifying to:
```
static struct sk_buff *brcm_tag_rcv(struct sk_buff *skb, struct net_device *dev)
{
	int source_port;
	u8 *brcm_tag;

	if (unlikely(!pskb_may_pull(skb, BRCM_LEG_PORT_ID)))
		return NULL;

	brcm_tag = dsa_etype_header_pos_rx(skb);

	source_port = brcm_tag[7] & BRCM_LEG_PORT_ID;

	skb->dev = dsa_master_find_slave(dev, 0, source_port);
	if (!skb->dev)
		return NULL;

	/* Remove Broadcom tag and update checksum */
	skb_pull_rcsum(skb, 8);

	dsa_default_offload_fwd_mark(skb);

	dsa_strip_etype_header(skb, 8);

	return skb;
}
```
And now it's working :)
But, what the hell is going on here?
Why is the tag 8 bytes instead of the standard 4 bytes?
@ffainelli @KanjiMonster

ffainelli, 2023-03-11:
The format is documented in net/dsa/tag_brcm.c, and tcpdump also know how to decode Broadcom tags...

ffainelli, 2023-03-11:
It seems like what you are receiving is actually perfectly valid, in that you have the following:

- 6 bytes legacy Broadcom tag
- 4 bytes VLAN tag with VLAN ID 0 since the switch must be forcing tagging of ingress packets
- 4 bytes Broadcom tag

VLAN ID 0 is added by the internal switch which is why the VLAN tag is *after* the 6-bytes tag. I am not sure what kernel version you are working, but at some point we required that the bridge layer automatically pop the VLAN tag:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=1c5ad5a940387ce03bc6b6ce497e3523963292b

KanjiMonster, 2023-03-11:
> * 4 bytes VLAN tag with VLAN ID 0 since the switch must be forcing tagging of ingress packets

Ah yes it does, I remember the bcm63xx embedded switches enforce tagged on the management port and ignore the untag bit from the VLAN table.

And due to the 4-byte Broadcom tag, it does not identify the packet as VLAN tagged, so it adds one based on its PVID (which is likely 0).

After (or rather before) that then it inserts the 6-byte legacy Broadcom tag before passing it on to linux.

Thinking about this, daisy-chaining switches also means that forwarding between the attached switch and the wan port will not work in hardware, because the wan port will then receive Broadcom-tagged packets (which the other side will not understand). So any forwarding will need to happen in software by the bridge layer.

Not sure how to express this though in switchdev/DSA terms.

On a slightly (un)related topic, didn't D0+ revisions of BCM63268 family SoCs switching the gigabit port (ge3) to a separate ethernet mac? I vaguely remember that there was some pinmux stuff for that.

Noltari, 2023-03-12:
> It seems like what you are receiving is actually perfectly valid, in that you have the following:
>
> * 6 bytes legacy Broadcom tag
> * 4 bytes VLAN tag with VLAN ID 0 since the switch must be forcing [...]

I'm using OpenWrt's 5.15 kernel.

> 
> https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=1c5ad5a940387ce03bc6b6ce497e3523963292b

Yes, but it was later changed: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=2c32a3d3c233b855943677609fe388f82b1f0975
I tried enabling `ds->untag_bridge_pvid` for `is63xx()` but it didn't remove the VLAN tag...
So i'm not really sure on how to do this properly...

Noltari, 2023-03-21:
@rikka0w0 the issue should be fixed with https://github.com/openwrt/openwrt/commit/82dc16fde0337b573c543f023f3fe1d3657e17e4

BTW, you can see examples on how to configure your DTS file on the following commits:
- https://github.com/openwrt/openwrt/commit/61f3c3b1eebd598b62f03cb2c63dde965624e0f1
- https://github.com/openwrt/openwrt/commit/80a3ecc894897f7fa42ab0b98c61ae7533ea7703

Enjoy :)

closed 2023-03-21 by Noltari