openwrt/openwrt #10315: I/O error  22.03.0.rc5

f4nff, 2022-07-21:
```
root@OpenWrt:~# ping 9929fralg.idc.best
PING 9929fralg.idc.best (82.115.30.10): 56 data bytes
64 bytes from 82.115.30.10: seq=0 ttl=50 time=165.446 ms
^C
--- 9929fralg.idc.best ping statistics ---
2 packets transmitted, 1 packets received, 50% packet loss
round-trip min/avg/max = 165.446/165.446/165.446 ms
root@OpenWrt:~# mtr 9929fralg.idc.best
Error loading shared library libncursesw.so.6: I/O error (needed by /usr/sbin/mtr)
Error relocating /usr/sbin/mtr: start_color: symbol not found
Error relocating /usr/sbin/mtr: werase: symbol not found
Error relocating /usr/sbin/mtr: raw: symbol not found
Error relocating /usr/sbin/mtr: init_pair: symbol not found
Error relocating /usr/sbin/mtr: wattr_off: symbol not found
Error relocating /usr/sbin/mtr: wmove: symbol not found
Error relocating /usr/sbin/mtr: wrefresh: symbol not found
Error relocating /usr/sbin/mtr: printw: symbol not found
Error relocating /usr/sbin/mtr: mvprintw: symbol not found
Error relocating /usr/sbin/mtr: initscr: symbol not found
Error relocating /usr/sbin/mtr: use_default_colors: symbol not found
Error relocating /usr/sbin/mtr: noecho: symbol not found
Error relocating /usr/sbin/mtr: wgetch: symbol not found
Error relocating /usr/sbin/mtr: wattr_on: symbol not found
Error relocating /usr/sbin/mtr: wattrset: symbol not found
Error relocating /usr/sbin/mtr: endwin: symbol not found
Error relocating /usr/sbin/mtr: stdscr: symbol not found
root@OpenWrt:~# ls
-ash: ls: I/O error
root@OpenWrt:~# ls
-ash: ls: I/O error
root@OpenWrt:~# ls
-ash: ls: I/O error
root@OpenWrt:~# ls
-ash: ls: I/O error
root@OpenWrt:~# cd //
root@OpenWrt:/# ls
-ash: ls: I/O error
root@OpenWrt:/# ping www.qq.com
-ash: ping: I/O error
root@OpenWrt:/# top
-ash: top: I/O error
root@OpenWrt:/#
```

After this happens, you can only restart the device.

21.02.3

22.03.0-rc5

`https://downloads.openwrt.org/releases/22.03.0-rc5/targets/x86/64/openwrt-22.03.0-rc5-x86-64-generic-ext4-combined-efi.img.gz`

device name：r86s

f4nff, 2022-07-21:
[https://github.com/openwrt/openwrt/issues/9837](https://github.com/openwrt/openwrt/issues/9837)

same error.

jow-, 2022-08-02:
Duplicate of #9837

closed 2022-08-02 by jow-