openwrt/openwrt #10316: umdns gen wrap /wrong json data

ptpt52, 2022-07-21:
ubus call umdns browse
output:
```
{
	"_dawn._tcp": {
		"MESH-AGENT_E4671E37EC68": {
			"ipv4": "192.168.15.106",
			"ipv6": "fdd5:d636:7052::1",
			"ipv6": "fe80::e667:1eff:fe37:ec68",
			"port": 1026
		},
		"MESH-AGENT_E4671E37EC54": {
			"ipv4": "192.168.15.236",
			"ipv6": "fd0e:6d62:cc45::1",
			"ipv6": "fe80::e667:1eff:fe37:ec54",
			"port": 1026
		},
		"MESH-AGENT_E4671E37EC95": {
			"ipv4": "192.168.15.151",
			"ipv6": "fd2b:8175:aa9b::1",
			"ipv6": "fe80::e667:1eff:fe37:ec95",
			"port": 1026
		},
		"MESH-AGENT_E4671E37EC8B": {
			"ipv4": "192.168.16.1",
			"ipv6": "fda8:42e8:8129::1",
			"ipv6": "fe80::e667:1eff:fe37:ec8b",
			"ipv4": "192.168.15.141",
			"port": 1026
		}
	},
	"_ssh._tcp": {
		"MESH-AGENT_E4671E37EC68": {
			"ipv4": "192.168.15.106",
			"ipv6": "fdd5:d636:7052::1",
			"ipv6": "fe80::e667:1eff:fe37:ec68",
			"port": 22,
			"txt": "daemon=dropbear"
		},
		"MESH-AGENT_E4671E37EC54": {
			"ipv4": "192.168.15.236",
			"ipv6": "fd0e:6d62:cc45::1",
			"ipv6": "fe80::e667:1eff:fe37:ec54",
			"port": 22,
			"txt": "daemon=dropbear"
		},
		"MESH-AGENT_E4671E37EC95": {
			"ipv4": "192.168.15.151",
			"ipv6": "fd2b:8175:aa9b::1",
			"ipv6": "fe80::e667:1eff:fe37:ec95",
			"port": 22,
			"txt": "daemon=dropbear"
		},
		"MESH-AGENT_E4671E37EC8B": {
			"ipv4": "192.168.16.1",
			"ipv6": "fda8:42e8:8129::1",
			"ipv6": "fe80::e667:1eff:fe37:ec8b",
			"ipv4": "192.168.15.141",
			"port": 22,
			"txt": "daemon=dropbear"
		}
	}
}
```

this json data is wrong, there is Duplicate key 'ipv6' 'ipv4'

@stintel

stintel, 2022-07-21:
Please don't mention random people. I don't know the umdns code.

neheb, 2022-07-21:
I believe @rmilecki works on umdnsd.

t-8ch, 2022-07-23:
According to to the [documentation examples](https://openwrt.org/docs/guide-developer/mdns) this looks to be expected.
Duplicate keys in JSON objects are allowed, albeit unusual.
Maybe you need to configure your JSON parser to handle this properly.

ptpt52, 2022-07-30:
> According to to the [documentation examples](https://openwrt.org/docs/guide-developer/mdns) this looks to be expected. Duplicate keys in JSON objects are allowed, albeit unusual. Maybe you need to configure your JSON parser to handle this properly.

but most implementations of JSON libraries do not accept duplicate keys
what lib/tools should I use to parse such json?

t-8ch, 2022-07-31:
> what lib/tools should I use to parse such json?

This depends strongly on your environment.

For Python you can use [the `object_pairs_hook`parameter for `json.load`](https://docs.python.org/3/library/json.html#json.load).
For json-c you can (probably) use `json_object_object_foreach()`.

In general APIs that iterate over object contents are more likely to expose this feature than APIs that treat objects as dictionaries.