openwrt/openwrt #10324: wolfssl 5.4 bug

openips, 2022-07-23:
when i build a rom in openwrt x64 master tree with curl and wolfssl 5.4 ,then i find curl can not work with https. i build a now one with openssl or wolfssl 5.3.0-stable , all are work well .
check and fix bugs, thks

openips, 2022-07-23:
@[cotequeiroz](https://github.com/cotequeiroz)

BKPepe, 2022-07-23:
Already reported here: https://github.com/openwrt/openwrt/issues/10302

Fixed by https://github.com/openwrt/packages/pull/19008

openips, 2022-07-24:
the main bug is wolfssl 5.4 . only change default SSL to mbedTLS in curl can not fixed the wolfssl 5.4  bug

openips, 2022-07-24:
I change default SSL to openssl in curl can also let curl work well .

openips, 2022-07-24:
the wolfssl is a very importent lib in openwrt base system
the wolfssl 5.4 bug effect is very bad

Borromini, 2022-07-24:
This is a dupe of https://github.com/openwrt/openwrt/issues/10311.

BKPepe, 2022-07-24:
Not exactly, the issue, which you referenced is different than this.

Borromini, 2022-07-24:
You're right. The overly generic issue title lead me down the wrong path.

basha318i, 2022-07-25:
Having the same issue, had to switch to snapshot to get libcurl4 installing properly. I hope this dependency gets resolved soon.

nberlee, 2022-07-25:
`wpad-wolfssl` also depends on a non existing wolfssl 5.4 library

borkra, 2022-07-28:
https://github.com/openwrt/openwrt/pull/10322

Neustradamus, 2022-07-30:
To follow this ticket.