openwrt/openwrt #10325: python distutils check is broken

neheb, 2022-07-24:
even though python3 'import distutils' succeeds, it is not enough to check whether distutils is installed or not. ping @dhewg 

Without distutils: 
```
/usr/lib/python3.6/distutils
/usr/lib/python3.6/distutils/__init__.py
/usr/lib/python3.6/distutils/__pycache__
/usr/lib/python3.6/distutils/__pycache__/__init__.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/version.cpython-36.pyc
/usr/lib/python3.6/distutils/version.py
```

With:
```
/usr/lib/python3.6/distutils
/usr/lib/python3.6/distutils/__init__.py
/usr/lib/python3.6/distutils/__pycache__
/usr/lib/python3.6/distutils/__pycache__/__init__.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/version.cpython-36.pyc
/usr/lib/python3.6/distutils/version.py
mangix@DESKTOP-VIRSTQA ~/d/openwrt> sudo apt install python3-distutils
[sudo] password for mangix:
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following package was automatically installed and is no longer required:
  libdumbnet1
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  python3-lib2to3
The following NEW packages will be installed:
  python3-distutils python3-lib2to3
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 222 kB of archives.
After this operation, 3143 kB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 python3-lib2to3 all 3.6.9-1~18.04 [77.4 kB]
Get:2 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 python3-distutils all 3.6.9-1~18.04 [144 kB]
Fetched 222 kB in 1s (167 kB/s)
Selecting previously unselected package python3-lib2to3.
(Reading database ... 35949 files and directories currently installed.)
Preparing to unpack .../python3-lib2to3_3.6.9-1~18.04_all.deb ...
Unpacking python3-lib2to3 (3.6.9-1~18.04) ...
Selecting previously unselected package python3-distutils.
Preparing to unpack .../python3-distutils_3.6.9-1~18.04_all.deb ...
Unpacking python3-distutils (3.6.9-1~18.04) ...
Setting up python3-lib2to3 (3.6.9-1~18.04) ...
Setting up python3-distutils (3.6.9-1~18.04) ...
mangix@DESKTOP-VIRSTQA ~/d/openwrt> find /usr/ | grep distutils
/usr/lib/python3.6/distutils
/usr/lib/python3.6/distutils/README
/usr/lib/python3.6/distutils/__init__.py
/usr/lib/python3.6/distutils/__pycache__
/usr/lib/python3.6/distutils/__pycache__/__init__.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/_msvccompiler.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/archive_util.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/bcppcompiler.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/ccompiler.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/cmd.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/config.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/core.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/cygwinccompiler.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/debug.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/dep_util.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/dir_util.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/dist.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/errors.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/extension.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/fancy_getopt.cpython-36.pyc
/usr/lib/python3.6/distutils/__pycache__/file_util.cpython-36.pyc
[... 174 lines cut ...]
/usr/lib/python3.8/distutils/cygwinccompiler.py
/usr/lib/python3.8/distutils/debug.py
/usr/lib/python3.8/distutils/dep_util.py
/usr/lib/python3.8/distutils/dir_util.py
/usr/lib/python3.8/distutils/dist.py
/usr/lib/python3.8/distutils/errors.py
/usr/lib/python3.8/distutils/extension.py
/usr/lib/python3.8/distutils/fancy_getopt.py
/usr/lib/python3.8/distutils/file_util.py
/usr/lib/python3.8/distutils/filelist.py
/usr/lib/python3.8/distutils/log.py
/usr/lib/python3.8/distutils/msvc9compiler.py
/usr/lib/python3.8/distutils/msvccompiler.py
/usr/lib/python3.8/distutils/spawn.py
/usr/lib/python3.8/distutils/sysconfig.py
/usr/lib/python3.8/distutils/text_file.py
/usr/lib/python3.8/distutils/unixccompiler.py
/usr/lib/python3.8/distutils/util.py
/usr/lib/python3.8/distutils/versionpredicate.py
/usr/share/doc/python3-distutils
/usr/share/doc/python3-distutils/README.Debian
/usr/share/doc/python3-distutils/changelog.Debian.gz
/usr/share/doc/python3-distutils/copyright
/usr/share/lintian/overrides/python3-distutils
```

aparcar, 2022-08-07:
Looks like 3.12 will drop distutil btw

neheb, 2022-09-09:
sure. but meson will fail to compile several packages when distutils is missing for IIRC python < 3.10

dhewg, 2022-09-09:
While I never had just that minimal python installed myself, checking for the distutils module fixed multiple reoccurring annoying issues. Searching for distutils on the package tracker doesn't yield anything though, are users really hitting this?

dhewg, 2022-09-09:
There's #7901, but that looks ancient

neheb, 2022-09-09:
Sounds still applicable.

dhewg, 2022-09-09:
Most packages seem to contain `from distutils import sysconfig`, so maybe switch to that?

neheb, 2022-09-09:
Seems right.

```
mangix@DESKTOP-VIRSTQA ~/d/openwrt> python3.8 -c "from distutils import sysconfig"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: cannot import name 'sysconfig' from 'distutils' (/usr/lib/python3.8/distutils/__init__.py)
```

closed 2022-09-11 by jow-