openwrt/openwrt #10332: hostapd: SIGSEGV when trying to send ubus messages for BSS Transition Management responses

stevenj, 2022-07-24:
When a BSS Transition Management response is received by hostapd, that is NOT the WNM_BSS_TM_ACCEPT status, the bss target variable remains uninitialized and causes the ubus message marshaling to access invalid memory, which results in a segfault.  This is reported by the kernel like so:

```
do_page_fault(): sending SIGSEGV to hostapd for invalid read access from 00000005
epc = 555d4775 in wpad[555cd000+103000]
ra = 555d4775 in wpad[555cd000+103000]
```

`hostapd` terminates, and brings the WiFi down which is highly disruptive to all devices connected on WiFi.

This is only seen when a band steering service is running, because it is a response to their operation.  I have seen it with `dawn` and I believe `usteer` will also trigger this segfault if it uses these messages.

stevenj, 2022-07-24:
This is fixed by: http://lists.openwrt.org/pipermail/openwrt-devel/2022-July/039097.html

Ramon-0011, 2022-07-25:
I am suffering from this bug in 21.02, so please cherry pick for 21.02 an 22.03 branches as well!

Ramon-0011, 2022-07-25:
Note I am currently using usteer, so yes usteer triggers this as well