openwrt/openwrt #10338: /etc/hotplug.d/iface/20-firewall - fw reload triggered because of changes in ISP DHCP lease time

Pitterling, 2022-07-26:
udhcpc -p /var/run/udhcpc-eth1.pid -s /lib/netifd/dhcp.script
--> triggering hotplug event in case of  DHCP changes

--> /etc/hotplug.d/iface/20-firewall
https://github.com/openwrt/openwrt/blob/e37ba80633c30ff179df92e8826ba52ff00b2a66/package/network/config/firewall/files/firewall.hotplug

```
logger -t firewall "Reloading firewall due to $ACTION of $INTERFACE ($DEVICE)"
fw3 -q reload
```

I verified by using the [troubleshooting guide](https://openwrt.org/docs/guide-user/network/protocol.dhcp#dhcp_client_scripts), that in my case only the lease time has changed (actually in my case the lease is renewed every time) No other IFUPDATE_DATA has changed

the last commit from 2014 has added this line
`[ "$ACTION" = ifupdate -a -z "$IFUPDATE_ADDRESSES" -a -z "$IFUPDATE_DATA" ] && exit 0`
because of IFUPDATE_DATA=1 exit 0 is not reached.

My workaround for the moment - remove the check of IFUPDATE_DATA and check only for IFUPDATE_ADDRESSES changes
`[ "$ACTION" = ifupdate -a -z "$IFUPDATE_ADDRESSES" ] && exit 0`

jow-, 2022-08-05:
When you say the lease time changes, do you mean you get a different lease duration assigned on renew?

Pitterling, 2022-08-06:
@jow- yes, almost on every renew this is alternating bw 7200, 3600, but sometimes also 3599/3601

```
Sat Jul 23 19:58:53 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 3599
Sat Jul 23 19:58:53 2022 user.notice root: dhcp.script> timezone: lease:3599 message:
Sat Jul 23 20:28:52 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 7200
Sat Jul 23 20:28:52 2022 user.notice root: dhcp.script> timezone: lease:7200 message:
Sat Jul 23 21:28:52 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 7200
Sat Jul 23 21:28:52 2022 user.notice root: dhcp.script> timezone: lease:7200 message:
Sat Jul 23 22:28:53 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 7200
Sat Jul 23 22:28:53 2022 user.notice root: dhcp.script> timezone: lease:7200 message:
Sat Jul 23 23:28:53 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 7200
Sat Jul 23 23:28:53 2022 user.notice root: dhcp.script> timezone: lease:7200 message:
Sun Jul 24 00:28:53 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 3600
Sun Jul 24 00:28:53 2022 user.notice root: dhcp.script> timezone: lease:3600 message:
Sun Jul 24 00:58:53 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 7200
Sun Jul 24 00:58:53 2022 user.notice root: dhcp.script> timezone: lease:7200 message:
Sun Jul 24 01:58:53 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 7200
Sun Jul 24 01:58:53 2022 user.notice root: dhcp.script> timezone: lease:7200 message:
Sun Jul 24 02:58:53 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 3600
Sun Jul 24 02:58:53 2022 user.notice root: dhcp.script> timezone: lease:3600 message:
Sun Jul 24 03:28:53 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 7200
Sun Jul 24 03:28:53 2022 user.notice root: dhcp.script> timezone: lease:7200 message:
Sun Jul 24 04:28:54 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 3599
Sun Jul 24 04:28:54 2022 user.notice root: dhcp.script> timezone: lease:3599 message:
Sun Jul 24 04:58:53 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 7200
Sun Jul 24 04:58:53 2022 user.notice root: dhcp.script> timezone: lease:7200 message:
Sun Jul 24 05:58:53 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 3599
Sun Jul 24 05:58:53 2022 user.notice root: dhcp.script> timezone: lease:3599 message:
Sun Jul 24 06:28:52 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 7200
Sun Jul 24 06:28:52 2022 user.notice root: dhcp.script> timezone: lease:7200 message:
Sun Jul 24 07:28:52 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 3601
Sun Jul 24 07:28:52 2022 user.notice root: dhcp.script> timezone: lease:3601 message:
Sun Jul 24 07:58:52 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 7200
Sun Jul 24 07:58:52 2022 user.notice root: dhcp.script> timezone: lease:7200 message:
Sun Jul 24 08:58:53 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 3600
Sun Jul 24 08:58:53 2022 user.notice root: dhcp.script> timezone: lease:3600 message:
Sun Jul 24 09:28:53 2022 daemon.notice netifd: wan (3082): udhcpc: lease of 85.233.xx.yy obtained, lease time 7200
Sun Jul 24 09:28:53 2022 user.notice root: dhcp.script> timezone: lease:7200 message: 
```

alanswanson, 2022-10-23:
Looks like a duplicate of #5723 which needs netifd updated to remove lifetime notifications previously required by now long dead Homenet. Patch posted to openwrt-devel list in August.

Pitterling, 2023-03-28:
@opty77 i don't get your point. I just use udhcp to get an IP from ISP. I'm not aware of any configuration option.

opty77, 2023-03-29:
Let me show you an example:

```Tue Mar 28 20:03:34 2023 daemon.notice netifd: wan (2365): udhcpc: sending renew to server 172.25.8.4
Tue Mar 28 20:03:34 2023 daemon.notice netifd: wan (2365): udhcpc: lease of 1.2.3.4 obtained from 172.25.8.4, lease time 14400
Tue Mar 28 20:03:35 2023 user.notice firewall: "data":{"dhcpserver":"172.25.8.4","leasetime":14400}
Tue Mar 28 20:03:35 2023 user.notice firewall: Reloading firewall due to ifupdate of wan (eth0.2)
Tue Mar 28 22:03:34 2023 daemon.notice netifd: wan (2365): udhcpc: sending renew to server 172.25.8.4
Tue Mar 28 22:03:34 2023 daemon.notice netifd: wan (2365): udhcpc: lease of 1.2.3.4 obtained from 172.25.8.4, lease time 14400
Tue Mar 28 22:03:35 2023 user.notice firewall: "data":{"dhcpserver":"172.25.8.250","leasetime":14400}
Tue Mar 28 22:03:35 2023 user.notice firewall: Reloading firewall due to ifupdate of wan (eth0.2)
Wed Mar 29 00:03:33 2023 daemon.notice netifd: wan (2365): udhcpc: sending renew to server 172.25.8.4
Wed Mar 29 00:03:33 2023 daemon.notice netifd: wan (2365): udhcpc: lease of 1.2.3.4 obtained from 172.25.8.4, lease time 14400
Wed Mar 29 02:03:34 2023 daemon.notice netifd: wan (2365): udhcpc: sending renew to server 172.25.8.4
Wed Mar 29 02:03:34 2023 daemon.notice netifd: wan (2365): udhcpc: lease of 1.2.3.4 obtained from 172.25.8.4, lease time 14400
Wed Mar 29 04:03:34 2023 daemon.notice netifd: wan (2365): udhcpc: sending renew to server 172.25.8.4
Wed Mar 29 04:03:34 2023 daemon.notice netifd: wan (2365): udhcpc: lease of 1.2.3.4 obtained from 172.25.8.4, lease time 14400
Wed Mar 29 06:03:34 2023 daemon.notice netifd: wan (2365): udhcpc: sending renew to server 172.25.8.4
Wed Mar 29 06:03:34 2023 daemon.notice netifd: wan (2365): udhcpc: lease of 1.2.3.4 obtained from 172.25.8.4, lease time 14400
Wed Mar 29 08:03:35 2023 daemon.notice netifd: wan (2365): udhcpc: sending renew to server 172.25.8.4
Wed Mar 29 08:03:35 2023 daemon.notice netifd: wan (2365): udhcpc: lease of 1.2.3.4 obtained from 172.25.8.4, lease time 14400
Wed Mar 29 08:03:35 2023 user.notice firewall: "data":{"dhcpserver":"172.25.8.4","leasetime":14400}
Wed Mar 29 08:03:35 2023 user.notice firewall: Reloading firewall due to ifupdate of wan (eth0.2)
```

Pitterling, 2023-03-29:
Yes, looks like the same issue, just in your case with DHCP and not with lease time. You could use my workaround from above.

Pitterling, 2023-03-29:
> Looks like a duplicate of #5723 which needs netifd updated to remove lifetime notifications previously required by now long dead Homenet. Patch posted to openwrt-devel list in August.

@alanswanson Not so sure if this is a duplicate, as I'm not using IPv6.