openwrt/openwrt #10342: @robimarko

cantona, 2022-07-27:
@robimarko 

With using IPQ40XX DSA ports with Network switch that has loop protection or STP turned on that will be detected as a looping.

ie. 

> brctl addbr br-lan
> brctl addif br-lan lan1
> brctl addif br-lan lan2
> ifconfig br-lan up
> ifconfig lan1 up
> ifconfig lan2 up

lan1 connect to end point client - ie PC
lan2 connect to a Switch w/ STP turned on

When both link up,  the Switch will detected a loop, hence goes to port discarding mode. Further investigating the issues, turned out it is because STP packets from Switch will duplicate in receiver side, ie lan1 interface. then send back the same packet to Switch, so it think it is a loop which is false alarm and causes a port discarding

_Originally posted by @cantona in https://github.com/openwrt/openwrt/issues/4721#issuecomment-1192147851_

robimarko, 2022-07-27:
@Ansuel Have you seen something like this with the standalone QCA8337?

robimarko, 2022-07-27:
And this, does not belong as OpenWrt issue as the PR is not merged.

cantona, 2022-08-10:
> And this, does not belong as OpenWrt issue as the PR is not merged.

Understand. Anything need me to try?

robimarko, 2022-08-10:
That's the thing, I dont see how this could happen.
Maybe @Ansuel knows something as he did a lot of work on the standalone switch.

cantona, 2022-08-10:
I can tell QCA8337 standalone switch has no such issue. Even with DSA enabled qca8k.

What I refer to is IPQ4019 SoC switch.  I am welcome to test anything if you have any idea or suggestion regarding this issue.

robimarko, 2022-08-10:
I understand, but functionally these switches are the same, its quite literally the same switch just embedded into the SoC

closed 2022-08-10 by ynezz