openwrt/openwrt #10348: Mikrotik cap ac - poe out only 10v

DSchndr, 2022-07-28:
Hi there,
it seems like that the Stock RouterOS firmware for the cap ap changes more than one GPIO when POE is enabled, since on stock I get the full POE voltage of my (802.3af) switch on the output port when its forced on.

On Openwrt its spot on 10v - not even a second ap will boot with that.

f00b4r0, 2022-08-17:
the commit message from 70eedac9b30a06d0d1445bfa31743440ba535e51 does mention that PoE out is unsupported though. Not a bug?

DSchndr, 2022-08-17:
For some users it seems like enabling poe with uci enables poe properly, the config for the gpio is there and the led turns on (with poe detection voltage in my case)
Would be something worth investigating since some users have the same issue and pass through is one of the (important) features of the ap

It is there, its just broken in some way

What i didn't test is if i force enable poe and then flash openwrt if it then works properly, would be a workaround

f00b4r0, 2022-08-17:
This comment https://github.com/openwrt/openwrt/pull/4055#issuecomment-815120363 from @alaraun is pretty clear that PoE out is not supported, so it not working properly isn't exactly a bug. It seems the toggle switch is only for the LEDs, not for the actual power output.

closed 2022-08-18 by DSchndr