openwrt/openwrt #10354: Routing/NAT hardware flow offloading only working temporarily with ramips/mt7621.

xervox, 2022-07-29:
Hi,

NAT offloading is causing one core max. CPU load after boot up resulting in ~200 Mbit/s max. speed via NAT forward instead 1000 Mbit/s over that WAN interface. When changing any settings like "Enable SYN-flood protection" or "Drop invalid packets" and push "save&apply", the offload is working temporarily again. After some time it will disable again (cause unknown yet).
```
 PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
   10     2 root     RW       0   0%  25% [ksoftirqd/0]
```
System: OpenWrt 22.03.0-rc5 r19523-bfd070e7fa / LuCI openwrt-22.03 branch git-22.167.28394-8a4486a
Target: ramips/mt7621

Software based offloading for routing/NAT: ON
Hardware flow offloading: ON
Packet Steering: ON
One VLAN Bridge with DSA VLAN filtering exists and two cusom VLAN IDs (802.1q) are added. 

I hope you could look into this. The bug also exists at stable release before and was appearing since the DSA Networking was added.

Thx a lot.

ningzichun, 2022-08-01:
Facing the same problem on OpenWrt 22.03-SNAPSHOT r19590-042d558536, ramips/mt7621

When Hardware flow offloading is enabled, the maximum traffic is ~200Mbps, even less than Software based offloading only, which is ~300Mbps. Also, enable Hardware flow offloading might result in random reboot on my device.

ziseon, 2022-08-01:
Facing the same problem on OpenWrt 22.03-SNAPSHOT r19590-042d558536, ramips/mt7621

When Hardware flow offloading is enabled, the maximum traffic is ~200Mbps, even less than Software based offloading only, which is ~300Mbps. Also, enable Hardware flow offloading might result in random reboot on my device.

neheb, 2022-08-10:
does https://github.com/openwrt/openwrt/pull/10238 help?

nazar554, 2022-08-10:
I'm having a similar issue when running `iperf3` on the Xiaomi Router 3G V1.
I have my `wan` port tagged and `lan1` and `lan2` untagged as VLAN 200 and 300 with bridge filtering enabled.

Switching works fine and I get gigabit speeds (~890 mbit/s) on devices plugged to `lan1`/`lan2`. 
But running iperf3 on the router caps at around 300 mbit/s with high cpu usage on `ksoftirqd`.
Is it because of no flow offload for `br.200` bridge local VLAN interface or some other conntrack overhead?

I'm running `OpenWrt 22.03.0-rc6 r19590-042d558536`

quarkysg, 2022-09-26:
@nazar554 AFAIK flow-offloads only works for non-local traffic.  I.e. if the traffic originates from the router (i.e. local), it goes thru the entire netfilter chain and will not be processed by the PPE, and thus no off-load.

szero, 2022-10-19:
@nazar554 yeah if you run speed tests directly on router the traffic enters wan interface but never leaves it, the offload only works if the traffic is passed between wan and lan interface so you have to test the speeds on some pc connected to the lan port