openwrt/openwrt #10367: Meraki MR12 firmware 21.02.1(3) Low speed Upload WIFI

lexmark3200, 2022-08-01:
Hi
I've installed a new firmware to Meraki MR12 (21.02.3), and the problem is when I test a wi-fi speed I get about 130-140 Mb/s DW, but only 2-3 Mb/s UP.
So I take a UART and flashed a 21.02.1, but the problem is the same.
Then I flashed 19.07.8 and everything is working fine.
The DW/UP speed are mostly the same 180/180 Mb/s.

lexmark3200, 2022-08-08:
I've upgradded to newest version OpenWrt 22.03.0-rc6 r19590-042d558536 the problem is stiil present. DW speed 150-160 Mb/s UP max 3 Mb/s.

Hurricos, 2022-08-23:
Please show `/sys/kernel/debug/ieee80211/phy0/netdev:*/stations/*/rc_stats`. Perhaps it's a calibration issue.

lexmark3200, 2022-08-23:
Hi Martin.
Did You want that ?
`root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy0/netdev:*/stations/*/rc_stats

              best    ____________rate__________    ____statistics___    _____last____    ______sum-of________
mode guard #  rate   [name   idx airtime  max_tp]  [avg(tp) avg(prob)]  [retry|suc|att]  [#success | #attempts]
HT20  LGI  1       S  MCS0     0    1477     5.6       0.0       0.0       1     0 0             0   0
HT20  LGI  1          MCS1     1     738    10.6       0.0       0.0       0     0 0             0   0
HT20  LGI  1          MCS2     2     492    14.9       0.0       0.0       0     0 0             0   0
HT20  LGI  1          MCS3     3     369    18.8       0.0       0.0       0     0 0             0   0
HT20  LGI  1          MCS4     4     246    25.4       0.0       0.0       0     0 0             0   0
HT20  LGI  1          MCS5     5     185    30.7       0.0       0.0       0     0 0             0   0
HT20  LGI  1          MCS6     6     164    33.0       0.0       0.0       0     0 0             0   0
HT20  LGI  1          MCS7     7     148    35.1       0.0       0.0       0     0 0             0   0
HT20  LGI  2          MCS8    16     738    10.6      10.6     100.0       4     0 0             1   1
HT20  LGI  2          MCS9    17     369    18.8      18.8     100.0       0     0 0             0   0
HT20  LGI  2          MCS10   18     246    25.4      25.4     100.0       0     0 0             0   0
HT20  LGI  2          MCS11   19     185    30.7      30.7     100.0       0     0 0             1   1
HT20  LGI  2          MCS12   20     123    38.9      38.9     100.0       0     0 0             0   0
HT20  LGI  2          MCS13   21      92    44.8      44.8     100.0       6     0 0             1   1
HT20  LGI  2          MCS14   22      82    47.3      47.3     100.0       6     0 0             1   1
HT20  LGI  2          MCS15   23      74    49.4      49.4     100.0       0     0 0             1   1
HT40  LGI  1          MCS0   128     711    10.9      10.9     100.0       4     0 0             1   1
HT40  LGI  1          MCS1   129     356    19.4      19.4     100.0       5     0 0             0   0
HT40  LGI  1          MCS2   130     237    26.0      26.0     100.0       5     0 0             2   2
HT40  LGI  1          MCS3   131     178    31.4       0.0       0.0       0     0 0             0   0
HT40  LGI  1          MCS4   132     119    39.6       0.0       0.0       0     0 0             0   0
HT40  LGI  1          MCS5   133      89    45.6       0.0       0.0       0     0 0             0   0
HT40  LGI  1          MCS6   134      79    48.0       0.0       0.0       0     0 0             0   0
HT40  LGI  1          MCS7   135      71    50.2       0.0       0.0       0     0 0             0   0
HT40  LGI  2          MCS8   144     356    19.4      19.4     100.0       0     0 0             0   0
HT40  LGI  2          MCS9   145     178    31.4      31.4     100.0       5     0 0             0   0
HT40  LGI  2          MCS10  146     119    39.6      39.6     100.0       6     0 0             1   1
HT40  LGI  2          MCS11  147      89    45.6      45.6     100.0       6     0 0             4   4
HT40  LGI  2          MCS12  148      59    53.7      53.7     100.0       6     0 0             2   2
HT40  LGI  2      P   MCS13  149      44    59.0      59.0     100.0       6     0 0             3   3
HT40  LGI  2     D    MCS14  150      40    60.9      60.9     100.0       6     0 0             2   2
HT40  LGI  2   B      MCS15  151      36    62.6      62.6     100.0       6     0 0             1   1
HT40  SGI  1          MCS0   192     640    12.0      12.0     100.0       0     0 0             0   0
HT40  SGI  1          MCS1   193     320    21.0      21.0     100.0       0     0 0             0   0
HT40  SGI  1          MCS2   194     214    27.9      27.9     100.0       0     0 0             0   0
HT40  SGI  1          MCS3   195     160    33.5      33.5     100.0       5     0 0             1   1
HT40  SGI  1          MCS4   196     107    41.8      41.8     100.0       6     0 0             1   1
HT40  SGI  1          MCS5   197      80    47.8       0.0       0.0       0     0 0             0   0
HT40  SGI  1          MCS6   198      71    50.1       0.0       0.0       0     0 0             0   0
HT40  SGI  1          MCS7   199      64    52.2       0.0       0.0       0     0 0             0   0
HT40  SGI  2          MCS8   208     320    21.0      21.0     100.0       0     0 0             0   0
HT40  SGI  2          MCS9   209     160    33.5      33.5     100.0       0     0 0             0   0
HT40  SGI  2          MCS10  210     107    41.8      41.8     100.0       6     0 0             1   1
HT40  SGI  2          MCS11  211      80    47.8      47.8     100.0       6     0 0             6   6
HT40  SGI  2          MCS12  212      54    55.6      55.6     100.0       6     0 0             2   2
HT40  SGI  2          MCS13  213      40    60.7      60.7     100.0       6     0 0             4   4
HT40  SGI  2    C     MCS14  214      36    62.5      62.5     100.0       6     0 0             4   4
HT40  SGI  2  A       MCS15  215      32    64.1      64.1     100.0       6     8 8           896   896

Total packet count::    ideal 941      lookaround 25
Average # of aggregated frames per A-MPDU: 1.0
root@OpenWrt:~#`

Hurricos, 2022-08-23:
That's all good. I realized after I posted that your problem is Client -> MR12, not MR12 -> client. The rate control table I asked for shows MR12 -> client.

Can you run `iperf3` from host A, through MR12 as host B, to a WiFi client host C? And do the same in reverse? I'd like to know whether the MR12 is being slowed down by CPU load.

Hurricos, 2022-08-23:
This is oddly similar to https://forum.openwrt.org/t/limited-speed-at-openwrt-access-point/118750/23. Though the two devices use different chips (AR928X vs AR922X).

lexmark3200, 2022-08-23:
> 
Can you run iperf3 from host A, through MR12 as host B, to a WiFi client host C? And do the same in reverse? I'd like to know whether the MR12 is being slowed down by CPU load.

All of the test upper post are done exacly what you want (if I understand incorrect) On my server HP t620 I have a docker with LibreSpeed, the MR12 is my wifi AP, my laptop is downloading from t620 with 147MB/s but uploading - 0.43Mb. This problem is only with the firmware 2x.x.x, but on 19 the speed are equall 180Mb/s.

Hurricos, 2022-08-26:
I'm really interested in this bit:

> I'd like to know whether the MR12 is being slowed down by CPU load.

Honestly, the fastest way to get a response would be to provide a configuration file backup of an MR12 which has this behavior. Feel free to manually scrub the configuration before you do so. I struggle to believe there's an actual regression here; I do know the MR12 was ported to ath79 from ar71xx, so in my view the default conf difference would be a switch of the ports.

lexmark3200, 2022-08-27:
Hi, Martin

I can send to You the backup of configuration. After flashed the new firmwarem I've change the IP adres, and config of Wi-fi (Mode N, Width 40 MHz, Force 40MHz, password : WPA2-PSK, Force AES). On Startup I've disabled Firewall, dnsmasq, odhcpd (Meraki working as Dumb AP)

I think thats all.

The backup is bellow.
[backup-OpenWrt-2022-08-27.tar.gz](https://github.com/openwrt/openwrt/files/9437956/backup-OpenWrt-2022-08-27.tar.gz)

Hurricos, 2022-08-28:
/etc/config/wireless:
- You do not need `hwmode` as of 21.02.3.

I was going to restore this to an MR12, but I just took the one I used to port out of its box and discovered it is dead.

lexmark3200, 2022-08-28:
Hmm
I''ve always setup my AP via LUCI. But strange if I understand correct hwmode set my AP to correct version b,g,n etc. So if in /etc/config/wireless I can see hwmode '11g', that means AP is setting to "g" standard?? I'm asking because in LUCI I've setting up this to N.
The print screen bellow.
[image]

raphielscape, 2022-09-05:
Hopping in, this issue is reproducible with Powerbeam M5 400 too, both in AP or CPE mode, this seems like a regression in ath9k itself in OpenWRT 2x.x as upload speed is OK with OpenWRT 19.x

lexmark3200, 2022-09-07:
A few minutes ago, I've upgraded to the new version 22.03.0, after quick tests, the problem is still present.

lexmark3200, 2022-10-02:
Another problem is with the Samsung a53 5G. I discovered it yesterday. I've bought a new one (from the shop).
When I connected it to the wifi network. The power of wifi is weak. Sometimes is full but a few swcond is very low. So the speed is very slowly.
I decided to test Samsung with the second device Meraki MR16 - the same issue. So i flashed the old version 21.02.03 to the Meraki MR16 - the same issue.

But on TL-WR1043ND with OpenWRT 22.03 everything is fine.
The speed connection in Samsung is 72Mbps, on Meraki's 65 Mbps, The Network Cell Info shows the -88dBm on Merakis - the distance between AP and the a53 is about 2-3m. On Motorola g7 Power is showing -34dBm - the same distance.
Anybody any idee??

lexmark3200, 2023-02-17:
Hi.
I reflashed MR12 with the newest 23.03.3 ver.
I maked some test with iperf.
Fisrt I checked the wiress conection between two PC (one PC one laptop) both with GB LAN ports. One port of Merakis is only FE, so I run server on PC and the client on laptop, what I discovered is strange:
```
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   640 KBytes  5.23 Mbits/sec
[  4]   1.00-2.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   2.00-3.01   sec  0.00 Bytes  0.00 bits/sec
[  4]   3.01-4.01   sec  0.00 Bytes  0.00 bits/sec
[  4]   4.01-5.01   sec  0.00 Bytes  0.00 bits/sec
[  4]   5.01-6.00   sec  0.00 Bytes  0.00 bits/sec
[  4]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec
[  4]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec
[  4]   8.00-9.01   sec   256 KBytes  2.09 Mbits/sec
[  4]   9.01-10.01  sec   256 KBytes  2.09 Mbits/sec
[  4]  10.01-11.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  11.00-12.00  sec   512 KBytes  4.19 Mbits/sec
[  4]  12.00-13.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  13.01-14.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  14.01-15.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  15.01-16.02  sec  0.00 Bytes  0.00 bits/sec
[  4]  16.02-17.00  sec   128 KBytes  1.06 Mbits/sec
[  4]  17.00-18.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  18.00-19.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  19.01-19.88  sec  0.00 Bytes  0.00 bits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-19.88  sec  1.88 MBytes   791 Kbits/sec                  sender
[  4]   0.00-19.88  sec  0.00 Bytes  0.00 bits/sec                  receiver
```
So low speed is not a problem with radio, but with LAN. the POE port is connecting with GB speed (Netgear switch with POE show GB connection) but something is not working ok. Something is blocking the traffic.

userisnotavaliable, 2024-09-20:
Hi, It's been a while, any update here? - looking to update my MR12's to Openwrt 23 but don't want to run into this speed issue...

lexmark3200, 2024-09-20:
Hi.
I don't know, but I think nothing was change.... 
I have sold my meraki a ling time ago. 
You van try reflash, and make some tests.

If you need some help with flashing, I will try to help you.

userisnotavaliable, 2024-09-21:
> Hi. I don't know, but I think nothing was change.... I have sold my meraki a long time ago. You can try reflash, and make some tests.
> 
> If you need some help with flashing, I will try to help you.

Thanks! Tried to upgrade the MR12 via Luci (from 19.07) and it said
`Device mr12 not supported by this image Supported devices: meraki,mr12 mr12 - Image version mismatch: image 2.0, device 1.0. Please wipe config during upgrade (force required) or reinstall. Reason: Partitions differ from ar71xx version of MR12. Image format is incompatible. To use sysupgrade, you must change /lib/update/common.sh::get_image to prepend 128K zeroes to this image, and change the bootcmd in u-boot to "bootm 0xbf0a0000". After that, you can use "sysupgrade -F -n". Make sure you do not keep your old config, as ethernet setup is not compatible either. For more details, see the OpenWrt Wiki: https://openwrt.org/toh/meraki/MR12, or the commit message of the MR12 ath79 port on git.openwrt.org. Invalid image type. Image check failed. `

I forced the upgrade but it bricked it

So then i used serial to flash it back to 19.07 (because the instructions are for that on the openwrt wiki) - how did you flash to >19.07 in the first place? Cheers

lexmark3200, 2024-09-21:
I try to answers you later. I think I have the instruction on my PC. 
The adress of partition are different....