openwrt/openwrt #10369: dnsmasq 2.86 very slow on loading with large configuration files.

sunshinejnjn, 2022-08-02:
Environment: (put here arch, model, OpenWrt version)
ARMv9, Phicomm K3, OpenWRT 22.03.0-rc5
(Generally OpenWRT 22.03)

Description:
After a clean upgrade to version 22.03.0-rc5, the system came with dnsmasq version 2.86. Also tested dnsmasq-full (2.86), the same result.
This version of dnsmasq got very slow on loading two 2MB sized config files (for servers= and address= directives).
On Phicomm K3 with dual-core armv9 (bcm53xx) and 512MB ram, it would take more than 5 minutes to load the said config and it would not run as stable as version 2.85. It has less RAM footprint (~19MB), but client DNS requests fail all the time.

Manually installed dnsmasq 2.85 from package repository from openwrt 21.07.3 (along with libubox, libubus deps), DNSMASQ finally went back to running smoothly. This old version (2.85) takes seconds to load the same configs, although it uses a bit more RAM (~22MB).

Don't know if 2.87(beta) would make any difference as the changelog shows no related modifications, yet.

hideonshadowX, 2022-08-10:
Hi，i have the same problem. how do you do to solve it?

sunshinejnjn, 2022-08-10:
Hi,

  Basically you want to remove the default dnsmasq 2.86 and install version
2.85 from the package repository.
  For example, you can do something like the following with bcm53xx. If you
have another board (you can find your target in /etc/opkg/distfeeds.conf ),
change the packages to your arch/target (check the file URLs at
https://downloads.openwrt.org/releases/21.02.3/packages/ )

cd ~
wget
https://downloads.openwrt.org/releases/21.02.3/packages/arm_cortex-a9/base/libubus20210630_2021-06-30-4fc532c8-2_arm_cortex-a9.ipk
wget
https://downloads.openwrt.org/releases/21.02.3/packages/arm_cortex-a9/base/libubox20210516_2021-05-16-b14c4688-2_arm_cortex-a9.ipk
wget
https://downloads.openwrt.org/releases/21.02.3/packages/arm_cortex-a9/base/dnsmasq-full_2.85-8_arm_cortex-a9.ipk
opkg remove dnsmasq dnsmasq-full
opkg install libubox20210516_2021-05-16-b14c4688-2_arm_cortex-a9.ipk
libubus20210630_2021-06-30-4fc532c8-2_arm_cortex-a9.ipk
dnsmasq-full_2.85-8_arm_cortex-a9.ipk

Best,
  JnJn

Originalimoc, 2022-09-04:
Same, but on a standalone Ubuntu baremetal server. strace shows very weird file syscall which does the loading.

kings-way, 2022-10-06:
@sunshinejnjn from another issue of another repo, I found this: https://github.com/felixonmars/dnsmasq-china-list/issues/227#issuecomment-922443863

And they have made a patch a year ago, merged into 2.87: https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=eb88eed1fc8ed246e9355531c2715fa2f7738afc;hp=8312a3ba4f06141647e86a2c5c4ccfb2e32582ab

Anyway, cause dnsmasq 2.86 does not support nfset, openwrt should really consider update it to 2.87.
Related: #10759 #10820

sunshinejnjn, 2022-10-08:
> @sunshinejnjn from an issue of another repo, I found this: [felixonmars/dnsmasq-china-list#227 (comment)](https://github.com/felixonmars/dnsmasq-china-list/issues/227#issuecomment-922443863)
> 
> And they have made a patch a year ago, merged into 2.87: https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=eb88eed1fc8ed246e9355531c2715fa2f7738afc;hp=8312a3ba4f06141647e86a2c5c4ccfb2e32582ab
> 
> Anyway, cause dnsmasq 2.86 does not support nftset, openwrt should really consider update it to 2.87. Related: #10759 #10820

Yeah, sure. 2.87 got released 10 days ago. I totally agree that OpenWRT should definitely update this broken 2.86 in the next stable release such as 22.03.1.

heiher, 2022-10-13:
What the fuck! So ugly code in v2.87: https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=blob;f=src/domain-match.c;h=f7db0fe88e04ed4c77c854182ba364afdcf80c22;hb=c4b9bc63e0029cf1beaf8bdcbd92fa09f33b599d#l671

v2.87 is still broken, very very very ... slow on load large rules.

ynezz, 2022-10-13:
@heiher please calm down, such behavior it not accepted here, check our [project rules](https://openwrt.org/rules) again, the important part is `12. Be nice to each other.`

> v2.87 is still broken, very very very ... slow on load large rules.

It might be, indeed, we're all doing what we can, it's FOSS, so [patches are welcome](https://openwrt.org/submitting-patches).

ttimasdf, 2023-01-04:
I'd suggest you not using dnsmasq for massive list-matching purpose. Since you use Phicomm product, I assume you are Chinese. See https://github.com/kuoruan/luci-app-v2ray/issues/451#issuecomment-1370413685 for my suggestion on this case.

If you are utilizing host lists for AD blocking, [Pi-hole](https://pi-hole.net/) is better.