openwrt/openwrt #10370: kmod-usb-net-lan78xx: Fails in `package/install` with `Unknown package 'kmod-usb-net-lan78xx'` and missing dependency and architecture incompatibility in 22.03.

dreirund, 2022-08-02:
### Maintainer:

*I don't know. I don't know where the `Makefile` for package `kmod-usb-net-lan78xx` resides.*

### Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` from `git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-07-29, ca. 09:00 UTC  
    *(I give up on posting latest commit date and hash, since `git log` gives me some old dates, and I don't know how I can make `git log` printing out the last operation. I don't know much about `git`.)*
* Building on GNU/Linux on x86_64-architecture (Distribution: [&rarr;&nbsp;Artix Linux](http://artixlinux.org)).
* OpenWrt configuration: [&rarr;&nbsp;`.config`](http://ix.io/46q6),
* `grep -i lan78 .config`:  
  ```
  CONFIG_PACKAGE_kmod-usb-net-lan78xx=y
  ```
* `grep -i mdio .config`:  
  ```
  CONFIG_PACKAGE_kmod-mdio-gpio=y
  CONFIG_PACKAGE_kmod-of-mdio=m
  CONFIG_PACKAGE_kmod-switch-bcm53xx-mdio=m
  CONFIG_PACKAGE_kmod-mdio-netlink=m
  CONFIG_PACKAGE_mdio-tools=m
  ```

### Description:

`make -j6` fails in above configuration on `package/install`.
An explicit run of  
`make -j1 V=sc`  
fails with dependency errors for `kmod-usb-net-lan78xx`:  
```
[...]
Unknown package 'kmod-usb-net-lan78xx'.
[...]
 * pkg_hash_check_unresolved: cannot find dependency kmod-of-mdio for kmod-usb-net-lan78xx
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-usb-net-lan78xx found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-usb-net-lan78xx.
[...]
```

<details><summary>More `make -j1 V=sc` output context (*click* to open):</summary>

```
[...]
SHELL= flock /[...]/openwrt-22.03/tmp/.root-copy.flock -c 'cp -fpR /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/packages/.pkgdir/kmod-usb-net-lan78xx/. /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/'
touch /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/stamp/.kmod-usb-net-lan78xx_installed
echo "kmod-usb-net-lan78xx" >> /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
mkdir -p /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/stamp
[...]
Installing kmod-usb-net-hso (5.10.131-1) to root...
Installing kmod-usb-net-kaweth (5.10.131-1) to root...
Unknown package 'kmod-usb-net-lan78xx'.
Installing kmod-usb-net-pegasus (5.10.131-1) to root...
Installing kmod-usb-net-mcs7830 (5.10.131-1) to root...
[...]
[...]
Configuring kmod-ipt-offload.
Configuring luci-app-olsr-viz.
Configuring Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kmod-of-mdio for kmod-usb-net-lan78xx
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-usb-net-lan78xx found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-usb-net-lan78xx.
[...]
Configuring collectd-mod-snmp6.
Configuring luci-proto-ncm.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/[...]/openwrt-22.03'
make[1]: *** [package/Makefile:111: /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/[...]/openwrt-22.03'
make: *** [/[...]/openwrt-22.03/include/toplevel.mk:230: world] Error 2
```

[&rarr;&nbsp;Full `make` output](https://github.com/openwrt/packages/files/9240909/make.log).

</details>

dreirund, 2022-08-02:
&hellip; it seems to be a missing dependency on `PACKAGE_kmod-of-mdio`:  
If I manually set `CONFIG_PACKAGE_kmod-of-mdio=y` (or `CONFIG_PACKAGE_kmod-usb-net-lan78xx=m` instead of `=y`) it `package/install`s fine.

shuganth, 2025-06-01:
`#!/bin/bash

# RTL8152/RTL8153 USB Ethernet Driver Installer for OpenWrt
# This script installs the RTL8152 driver for USB ethernet adapters like Honeywell adapters
# Works around dependency and architecture issues in standard OpenWrt builds

set -e  # Exit on any error

# Colors for output
RED='\033[0;31m'
GREEN='\033[0;32m'
YELLOW='\033[1;33m'
BLUE='\033[0;34m'
NC='\033[0m' # No Color

# Function to print colored output
print_status() {
    echo -e "${BLUE}[INFO]${NC} $1"
}

print_success() {
    echo -e "${GREEN}[SUCCESS]${NC} $1"
}

print_warning() {
    echo -e "${YELLOW}[WARNING]${NC} $1"
}

print_error() {
    echo -e "${RED}[ERROR]${NC} $1"
}

# Function to check if running as root
check_root() {
    if [ "$EUID" -ne 0 ]; then
        print_error "This script must be run as root"
        exit 1
    fi
}

# Function to detect OpenWrt version and kernel
detect_system() {
    if [ ! -f /etc/openwrt_release ]; then
        print_error "This script is designed for OpenWrt systems only"
        exit 1
    fi
    
    KERNEL_VERSION=$(uname -r)
    OPENWRT_VERSION=$(grep DISTRIB_RELEASE /etc/openwrt_release | cut -d"'" -f2)
    ARCH=$(grep DISTRIB_ARCH /etc/openwrt_release | cut -d"'" -f2)
    TARGET=$(grep DISTRIB_TARGET /etc/openwrt_release | cut -d"'" -f2)
    
    print_status "Detected OpenWrt $OPENWRT_VERSION, Kernel $KERNEL_VERSION, Architecture $ARCH"
}

# Function to check if RTL8152 module is already loaded
check_existing_driver() {
    if lsmod | grep -q r8152; then
        print_success "RTL8152 driver is already loaded!"
        print_status "Current USB ethernet interfaces:"
        ip link show | grep -E "(eth[1-9]|usb[0-9])" || print_warning "No additional ethernet interfaces found"
        exit 0
    fi
}

# Function to check for RTL8153 USB devices
check_usb_devices() {
    print_status "Checking for RTL8153 USB ethernet devices..."
    
    FOUND_RTL=false
[... 242 lines cut ...]
    check_usb_devices
    download_package
    install_driver
    load_driver
    setup_autoload
    verify_installation
    
    echo ""
    print_success "RTL8152 driver installation completed successfully!"
    echo ""
    print_status "Your USB ethernet adapter should now be working."
    print_status "You can configure it in LuCI web interface at: http://192.168.1.1"
    print_status "Go to Network → Interfaces to set up your new ethernet interface."
    echo ""
    
    cleanup_and_exit 0
}

# Trap to ensure cleanup on script exit
trap cleanup_and_exit EXIT INT TERM

# Run main function with all arguments
main "$@"`

this worked for me