openwrt/openwrt #10371: Fritzbox 3390: 2 LAN Ports (still) not working - “OpenWrt 22.03.0-rc6 Changelog”

frank-44, 2022-08-02:
Hello 

I was happy to read from "OpenWrt 22.03.0-rc6 Changelog" about
fix 506432a lantiq: fix network port GPIO settings for Fritzbox 3390
and installed RC6.

But unfortunately ports LAN3 and LAN4 are still not working on my device.
LAN1 and LAN2 are fine.

The ports get link, but no traffic.
Maybe the log could help to find the reason:

root@OpenWrt-3390-1:~# dmesg | fgrep -i gswip
[    1.156499] gswip 1e108000.switch: dsa switch register failed: -517
[    1.646879] gswip 1e108000.switch lan3 (uninitialized): PHY [1e108000.switch-mii:00] driver [Qualcomm Atheros AR8035] (irq=POLL)
[    1.736876] gswip 1e108000.switch lan4 (uninitialized): PHY [1e108000.switch-mii:01] driver [Qualcomm Atheros AR8035] (irq=POLL)
[    1.752463] gswip 1e108000.switch lan2 (uninitialized): PHY [1e108000.switch-mii:11] driver [Intel XWAY PHY11G (xRX v1.2 integrated)] (irq=POLL)
[    1.769285] gswip 1e108000.switch lan1 (uninitialized): PHY [1e108000.switch-mii:13] driver [Intel XWAY PHY11G (xRX v1.2 integrated)] (irq=POLL)
[    1.785100] gswip 1e108000.switch: probed GSWIP version 21 mod 0
[    7.972557] gswip 1e108000.switch lan1: configuring for phy/internal link mode
[   10.036517] gswip 1e108000.switch lan1: Link is Up - 100Mbps/Full - flow control off
[   12.873266] gswip 1e108000.switch lan1: Link is Down
[   42.363896] gswip 1e108000.switch lan1: configuring for phy/internal link mode
[   42.536323] gswip 1e108000.switch lan2: configuring for phy/internal link mode
[   42.606378] gswip 1e108000.switch lan3: configuring for phy/rgmii link mode
[   42.662572] gswip 1e108000.switch lan4: configuring for phy/rgmii link mode
[   44.436862] gswip 1e108000.switch lan1: Link is Up - 100Mbps/Full - flow control off
[   45.797108] gswip 1e108000.switch lan4: Link is Up - 1Gbps/Full - flow control rx/tx
[   50.038844] gswip 1e108000.switch lan3: Link is Up - 100Mbps/Full - flow control rx/tx
[  140.595299] gswip 1e108000.switch lan4: Link is Down
[  143.475680] gswip 1e108000.switch lan2: Link is Up - 1Gbps/Full - flow control rx/tx
[  386.446786] gswip 1e108000.switch lan3: Link is Down
[  386.461715] gswip 1e108000.switch lan3: configuring for phy/rgmii link mode
[  386.536387] gswip 1e108000.switch lan4: configuring for phy/rgmii link mode
[  393.780406] gswip 1e108000.switch lan3: Link is Up - 100Mbps/Full - flow control rx/tx
[  473.332904] gswip 1e108000.switch lan3: Link is Down

Regards,
Frank

chunkeey, 2022-08-04:
Hm looks like you tested 1Gbps + 100Mbits but neither worked, is this correct? 

There isn't much information in the logs there. From what I can tell, the PHY has power since it was correctly identified as a "Qualcomm Atheros AR8035" (if it had no power it would have complained there.). The RGMII link mode is sensible too. The AR8035 support RGMII/RMII/MII and RGMII is the only one for 1Gbps support.

(Just in case, you did try a fresh openwrt install with no previous network configuration.)

frank-44, 2022-08-04:
It's correct I tested with 1Gbps + 100Mbits.

I also have Fritzbox 7360SL and 3370 working.
Here their logs of initialisation for information:

```
root@OpenWrt-3370-3:~# logread | fgrep -i gswi 
Thu Aug  4 15:58:47 2022 kern.err kernel: [    1.154186] gswip 1e108000.switch: dsa switch register failed: -517
Thu Aug  4 15:58:47 2022 kern.info kernel: [    1.566984] gswip 1e108000.switch lan3 (uninitialized): PHY [1e108000.switch-mii:00] driver [Intel XWAY PHY11G (PEF 7071/PEF 7072) v1.3] (irq=POLL)
Thu Aug  4 15:58:47 2022 kern.info kernel: [    1.584253] gswip 1e108000.switch lan4 (uninitialized): PHY [1e108000.switch-mii:01] driver [Intel XWAY PHY11G (PEF 7071/PEF 7072) v1.3] (irq=POLL)
Thu Aug  4 15:58:47 2022 kern.info kernel: [    1.601290] gswip 1e108000.switch lan2 (uninitialized): PHY [1e108000.switch-mii:11] driver [Intel XWAY PHY11G (xRX v1.1 integrated)] (irq=POLL)
Thu Aug  4 15:58:47 2022 kern.info kernel: [    1.618179] gswip 1e108000.switch lan1 (uninitialized): PHY [1e108000.switch-mii:13] driver [Intel XWAY PHY11G (xRX v1.1 integrated)] (irq=POLL)
Thu Aug  4 15:58:47 2022 kern.info kernel: [    1.634075] gswip 1e108000.switch: probed GSWIP version 0 mod 1
Thu Aug  4 15:58:47 2022 kern.info kernel: [    8.184001] gswip 1e108000.switch lan1: configuring for phy/internal link mode
Thu Aug  4 15:59:06 2022 kern.info kernel: [   38.757624] gswip 1e108000.switch lan1: configuring for phy/internal link mode
Thu Aug  4 15:59:06 2022 kern.info kernel: [   38.901876] gswip 1e108000.switch lan2: configuring for phy/internal link mode
Thu Aug  4 15:59:06 2022 kern.info kernel: [   39.029851] gswip 1e108000.switch lan3: configuring for phy/rgmii link mode
Thu Aug  4 15:59:06 2022 kern.info kernel: [   39.120388] gswip 1e108000.switch lan4: configuring for phy/rgmii link mode
Thu Aug  4 15:59:09 2022 kern.info kernel: [   42.186617] gswip 1e108000.switch lan3: Link is Up - 1Gbps/Full - flow control off
Thu Aug  4 15:59:09 2022 kern.info kernel: [   42.274316] gswip 1e108000.switch lan4: Link is Up - 1Gbps/Full - flow control rx/tx
Thu Aug  4 17:34:00 2022 kern.info kernel: [  445.864695] gswip 1e108000.switch lan4: Link is Down
Thu Aug  4 17:34:03 2022 kern.info kernel: [  448.984871] gswip 1e108000.switch lan4: Link is Up - 1Gbps/Full - flow control rx/tx
```

```
root@OpenWrt-7360SL:~# logread | fgrep -i gswi 
Thu Aug  4 13:05:42 2022 kern.err kernel: [    1.107130] gswip 1e108000.switch: dsa switch register failed: -517
Thu Aug  4 13:05:42 2022 kern.info kernel: [    1.529879] gswip 1e108000.switch lan3 (uninitialized): PHY [1e108000.switch-mii:00] driver [Qualcomm Atheros AR8030] (irq=POLL)
Thu Aug  4 13:05:42 2022 kern.info kernel: [    1.543982] gswip 1e108000.switch lan4 (uninitialized): PHY [1e108000.switch-mii:01] driver [Qualcomm Atheros AR8030] (irq=POLL)
Thu Aug  4 13:05:42 2022 kern.info kernel: [    1.559637] gswip 1e108000.switch lan2 (uninitialized): PHY [1e108000.switch-mii:11] driver [Intel XWAY PHY11G (xRX v1.2 integrated)] (irq=POLL)
Thu Aug  4 13:05:42 2022 kern.info kernel: [    1.576518] gswip 1e108000.switch lan1 (uninitialized): PHY [1e108000.switch-mii:13] driver [Intel XWAY PHY11G (xRX v1.2 integrated)] (irq=POLL)
Thu Aug  4 13:05:42 2022 kern.info kernel: [    1.591400] gswip 1e108000.switch: probed GSWIP version 21 mod 0
Thu Aug  4 13:05:42 2022 kern.info kernel: [    7.579646] gswip 1e108000.switch lan1: configuring for phy/internal link mode
Thu Aug  4 13:05:42 2022 kern.info kernel: [   10.682567] gswip 1e108000.switch lan1: Link is Up - 1Gbps/Full - flow control rx/tx
Thu Aug  4 13:05:42 2022 kern.info kernel: [   12.502020] gswip 1e108000.switch lan1: Link is Down
Thu Aug  4 13:06:06 2022 kern.info kernel: [   43.766394] gswip 1e108000.switch lan1: configuring for phy/internal link mode
Thu Aug  4 13:06:06 2022 kern.info kernel: [   43.987918] gswip 1e108000.switch lan2: configuring for phy/internal link mode
Thu Aug  4 13:06:06 2022 kern.info kernel: [   44.053902] gswip 1e108000.switch lan3: configuring for phy/rmii link mode
Thu Aug  4 13:06:07 2022 kern.info kernel: [   44.105086] gswip 1e108000.switch lan4: configuring for phy/rmii link mode
Thu Aug  4 13:06:08 2022 kern.info kernel: [   46.043022] gswip 1e108000.switch lan2: Link is Up - 1Gbps/Full - flow control rx/tx
Thu Aug  4 13:06:09 2022 kern.info kernel: [   46.202903] gswip 1e108000.switch lan3: Link is Up - 100Mbps/Full - flow control rx/tx
Thu Aug  4 13:06:09 2022 kern.info kernel: [   46.923500] gswip 1e108000.switch lan1: Link is Up - 1Gbps/Full - flow control rx/tx

```

chunkeey, 2022-08-04:
yeah, this likely needs more investigation. But I think this requires a device.

If you can assign your connected device a static IP and test if you can pass any sort of traffic in either direction. This could help to debug the problem. Because if it's DHCP that's failing, but bigger packages can sometime pass though and show up on the stats counter... this would point to a incorrect/missing rx/tx delay.

This would be "easy to change" since the knob is the phy-mode of lan3+lan4 in the dts. Apart from just "rgmii", there are special rgmii-(rx/tx)id modes and setting the correct one can make the difference.

@kestrel1974 is this something you experience too?

frank-44, 2022-08-04:
Your assumption "incorrect/missing rx/tx delay" sounds logical.

I assigned a static IP  (192.168.1.1) for LAN4 and connected a PC with 192.168.1.2
Obviously the 3390-device can transmit packets but not receive via LAN4 port

[image]

Plugging the network cable produces the expected log entries;

[image]

chunkeey, 2022-08-05:
Do you see those VLAN'd packaged ping requests in tcpdump/wireshark on the 192.168.1.2 PC of yours? (and the FB3390 isn't getting the ping replies now?)

If you are [able to compile your images](https://openwrt.org/docs/guide-developer/toolchain/beginners-build-guide), you could give this patch a try:

---
```
diff --git a/target/linux/lantiq/files/arch/mips/boot/dts/lantiq/vr9_avm_fritz3390.dts b/target/linux/lantiq/files/arch/mips/boot/dts/lantiq/vr9_avm_fritz3390.dts
index 5a3f2e5836..db0dca1a63 100644
--- a/target/linux/lantiq/files/arch/mips/boot/dts/lantiq/vr9_avm_fritz3390.dts
+++ b/target/linux/lantiq/files/arch/mips/boot/dts/lantiq/vr9_avm_fritz3390.dts
@@ -170,14 +170,14 @@
 	port@0 {
 		reg = <0>;
 		label = "lan3";
-		phy-mode = "rgmii";
+		phy-mode = "rgmii-id";
 		phy-handle = <&phy0>;
 	};
 
 	port@1 {
 		reg = <1>;
 		label = "lan4";
-		phy-mode = "rgmii";
+		phy-mode = "rgmii-rxid";
 		phy-handle = <&phy1>;
 	};
```
---

This tests two of the three delay settings at once. lan3 now has both rx/tx delays set. lan4 only has rx delay.
If one of the ports now start working, then we likely found the culprit.

If not, you can change one of the two to "rgmii-txid" and try again.

You can check during boot if the changes got applied, the entries:
```
[ 42.606378] gswip 1e108000.switch lan3: configuring for phy/rgmii link mode
[ 42.662572] gswip 1e108000.switch lan4: configuring for phy/rgmii link mode
```
should change to the respective phy/rgmii(-(rx|tx)id) mode.

kestrel1974, 2022-08-05:
@frank-44 @chunkeey Interestingly I had the same issue with 7490, while the avm stock firmware says rgmii, it turned out that the bootloader actually sets the flags for rgmii-rxid in the registers. I found that out by adding tracing to the at803x.c and printing the registers before they were set by the driver.
I just tried rgmii-rxid and it works with static ip and dhcp. I just tried ssh and it worked. I have not made the effort to test without rgmii-rxid for now.

kestrel1974, 2022-08-05:
I have created [a fix and a PR](https://github.com/openwrt/openwrt/pull/10391) for this issue.

chunkeey, 2022-08-05:
I found this [RGMII Interface Timing Considerations](https://ethernetfmc.com/rgmii-interface-timing-considerations/) helpful in the past to explain what's going on.

The short version is that it's related to the PCB trace layout for the RGMII signal lines. Ideally all lines should all have equal length, but to account for less-than-ideal situations the MAC or PHY (or both!) can specify delays to account for this.

commit 8756a047 (2022-08-05): lantiq: fix lan port 3+4 phy-mode settings for Fritzbox 3390

There are forum reports that 2 LAN ports are still not working,
the phy-mode settings are adjusted to fix the problem.

Fixes: #10371
commit c2c32bdc (2022-08-06): lantiq: fix lan port 3+4 phy-mode settings for Fritzbox 3390

There are forum reports that 2 LAN ports are still not working,
the phy-mode settings are adjusted to fix the problem.

Fixes: #10371

(cherry picked from commit 8756a047874bf688138a81898b6973f196cd1d36)
closed 2022-08-05 by jow-