openwrt/openwrt #10372: openwrt-22.03.0-rc6：Error compiling x86x64 firmware

openwrtdiy, 2022-08-03:
make[3]: Entering directory '/home/kokang/openwrt-22.03.0-rc6/package/kernel/button-hotplug'
touch /home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/.prepared_32c83835befb6cbfb6a2e663930b75ed_6664517399ebbbc92a37c5bb081b5c53_check
true
[ ! -d ./src/ ] || cp -fpR ./src/. /home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug
touch /home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/.prepared_32c83835befb6cbfb6a2e663930b75ed_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/.configured_*
rm -f /home/kokang/openwrt-22.03.0-rc6/staging_dir/target-x86_64_musl/stamp/.button-hotplug_installed
(cd /home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/./; if [ -x ./configure ]; then find /home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/ -name config.guess | xargs -r chmod u+w; find /home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/ -name config.guess | xargs -r -n1 cp --remove-destination /home/kokang/openwrt-22.03.0-rc6/scripts/config.guess; find /home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/ -name config.sub | xargs -r chmod u+w; find /home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/ -name config.sub | xargs -r -n1 cp --remove-destination /home/kokang/openwrt-22.03.0-rc6/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug=button-hotplug -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug=button-hotplug -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -DCONFIG_BUTTON_HOTPLUG=1  " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug=button-hotplug -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -DCONFIG_BUTTON_HOTPLUG=1  " CPPFLAGS="-I/home/kokang/openwrt-22.03.0-rc6/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/usr/include -I/home/kokang/openwrt-22.03.0-rc6/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/include/fortify -I/home/kokang/openwrt-22.03.0-rc6/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/include " LDFLAGS="-L/home/kokang/openwrt-22.03.0-rc6/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/usr/lib -L/home/kokang/openwrt-22.03.0-rc6/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/.built
touch /home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/.built_check
make -C "/home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.134" KCFLAGS="-fmacro-prefix-map=/home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/kokang/openwrt-22.03.0-rc6/staging_dir/host/include -I/home/kokang/openwrt-22.03.0-rc6/staging_dir/hostpkg/include -I/home/kokang/openwrt-22.03.0-rc6/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Jul 31 15:12:47 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/kokang/openwrt-22.03.0-rc6/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/kokang/openwrt-22.03.0-rc6/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="" KERNELRELEASE=5.10.134 M="/home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug" EXTRA_CFLAGS=" -DCONFIG_BUTTON_HOTPLUG=1  " CONFIG_BUTTON_HOTPLUG=m modules
make[4]: Entering directory '/home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.134'
make[5]: *** No rule to make target '/home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/button-hotplug.o', needed by '/home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/button-hotplug.mod'.  Stop.
Makefile:1831: recipe for target '/home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug' failed
make[4]: *** [/home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug] Error 2
make[4]: Leaving directory '/home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.134'
Makefile:50: recipe for target '/home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/.built' failed
make[3]: *** [/home/kokang/openwrt-22.03.0-rc6/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/.built] Error 2
make[3]: Leaving directory '/home/kokang/openwrt-22.03.0-rc6/package/kernel/button-hotplug'
time: package/kernel/button-hotplug/compile#0.48#0.46#0.95
    ERROR: package/kernel/button-hotplug failed to build.
package/Makefile:114: recipe for target 'package/kernel/button-hotplug/compile' failed
make[2]: *** [package/kernel/button-hotplug/compile] Error 1
make[2]: Leaving directory '/home/kokang/openwrt-22.03.0-rc6'
package/Makefile:110: recipe for target '/home/kokang/openwrt-22.03.0-rc6/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/kokang/openwrt-22.03.0-rc6/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kokang/openwrt-22.03.0-rc6'
/home/kokang/openwrt-22.03.0-rc6/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

aparcar, 2022-08-07:
@jow- didn't you provide a fix which fixes this hotplug issue some time ago?

openwrtdiy, 2022-08-07:
I got the same error when compiling openwrt-22.03-rc5 before!

aparcar, 2022-08-07:
Is it there with SNAPSHOTS, too?

openwrtdiy, 2022-08-08:
> Is it there with SNAPSHOTS, too?

master-snapshots compiles no problem!

openwrtdiy, 2022-08-12:
ln -sf zerotier-one zerotier-idtool
ln -sf zerotier-one zerotier-cli
make[4]: Leaving directory '/home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6'
touch /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/.built
rm -rf /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/.pkgdir/zerotier.installed /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/.pkgdir/zerotier
mkdir -p /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/.pkgdir/zerotier
install -d -m0755 /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/.pkgdir/zerotier/usr/bin
install -m0755 /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/zerotier-one /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/.pkgdir/zerotier/usr/bin/
ln -sf zerotier-one /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/.pkgdir/zerotier/usr/bin/zerotier-cli
ln -sf zerotier-one /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/.pkgdir/zerotier/usr/bin/zerotier-idtool
cp -fpR ./files/* /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/.pkgdir/zerotier/
touch /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/.pkgdir/zerotier.installed
mkdir -p /home/kokang/openwrt-22.03/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/kokang/openwrt-22.03/tmp/.root-copy.flock -c 'cp -fpR /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/.pkgdir/zerotier/. /home/kokang/openwrt-22.03/staging_dir/target-x86_64_musl/root-x86/'
touch /home/kokang/openwrt-22.03/staging_dir/target-x86_64_musl/root-x86/stamp/.zerotier_installed
mkdir -p /home/kokang/openwrt-22.03/bin/targets/x86/64/packages /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/ipkg-x86_64/zerotier/CONTROL /home/kokang/openwrt-22.03/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/ipkg-x86_64/zerotier/usr/bin
install -m0755 /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/zerotier-one /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/ipkg-x86_64/zerotier/usr/bin/
ln -sf zerotier-one /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/ipkg-x86_64/zerotier/usr/bin/zerotier-cli
ln -sf zerotier-one /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/ipkg-x86_64/zerotier/usr/bin/zerotier-idtool
cp -fpR ./files/* /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/ipkg-x86_64/zerotier/
find /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/ipkg-x86_64/zerotier -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/kokang/openwrt-22.03/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/kokang/openwrt-22.03/scripts/strip-kmod.sh" PATCHELF="/home/kokang/openwrt-22.03/staging_dir/host/bin/patchelf" /home/kokang/openwrt-22.03/scripts/rstrip.sh /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/ipkg-x86_64/zerotier
rstrip.sh: /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/ipkg-x86_64/zerotier/usr/bin/zerotier-one: executable
(cd /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/ipkg-x86_64/zerotier/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm; echo "$V_Package_zerotier_conffiles" > conffiles;  )
install -d -m0755 /home/kokang/openwrt-22.03/bin/packages/x86_64/packages
/home/kokang/openwrt-22.03/staging_dir/host/bin/fakeroot /home/kokang/openwrt-22.03/staging_dir/host/bin/bash /home/kokang/openwrt-22.03/scripts/ipkg-build -m "" /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/ipkg-x86_64/zerotier /home/kokang/openwrt-22.03/bin/packages/x86_64/packages
Packaged contents of /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/ipkg-x86_64/zerotier into /home/kokang/openwrt-22.03/bin/packages/x86_64/packages/zerotier_1.8.6-2_x86_64.ipk
echo "zerotier" >> /home/kokang/openwrt-22.03/staging_dir/target-x86_64_musl/pkginfo/zerotier.default.install
make[3]: Leaving directory '/home/kokang/openwrt-22.03/feeds/packages/net/zerotier'
time: package/feeds/packages/zerotier/compile#77.67#7.61#84.89
make[3]: Entering directory '/home/kokang/openwrt-22.03/package/kernel/button-hotplug'
rm -f /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/.built
touch /home/kokang/openwrt-22.03/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/.built_check
[... cut ...]
make[2]: *** [package/kernel/button-hotplug/compile] Error 1
make[2]: Leaving directory '/home/kokang/openwrt-22.03'
package/Makefile:110: recipe for target '/home/kokang/openwrt-22.03/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/kokang/openwrt-22.03/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kokang/openwrt-22.03'
/home/kokang/openwrt-22.03/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2
kokang@virtual-machine:~/openwrt-22.03$

openwrtdiy, 2022-09-05:
install -d -m0755 /home/kokang/openwrt-22.03.0/bin/packages/x86_64/packages
/home/kokang/openwrt-22.03.0/staging_dir/host/bin/fakeroot /home/kokang/openwrt-22.03.0/staging_dir/host/bin/bash /home/kokang/openwrt-22.03.0/scripts/ipkg-build -m "" /home/kokang/openwrt-22.03.0/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/ipkg-x86_64/zerotier /home/kokang/openwrt-22.03.0/bin/packages/x86_64/packages
Packaged contents of /home/kokang/openwrt-22.03.0/build_dir/target-x86_64_musl/ZeroTierOne-1.8.6/ipkg-x86_64/zerotier into /home/kokang/openwrt-22.03.0/bin/packages/x86_64/packages/zerotier_1.8.6-2_x86_64.ipk
echo "zerotier" >> /home/kokang/openwrt-22.03.0/staging_dir/target-x86_64_musl/pkginfo/zerotier.default.install
make[3]: Leaving directory '/home/kokang/openwrt-22.03.0/feeds/packages/net/zerotier'
time: package/feeds/packages/zerotier/compile#84.66#9.16#93.42
make[3]: Entering directory '/home/kokang/openwrt-22.03.0/package/kernel/button-hotplug'
rm -f /home/kokang/openwrt-22.03.0/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/.built
touch /home/kokang/openwrt-22.03.0/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/.built_check
make -C "/home/kokang/openwrt-22.03.0/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.138" KCFLAGS="-fmacro-prefix-map=/home/kokang/openwrt-22.03.0/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/kokang/openwrt-22.03.0/staging_dir/host/include -I/home/kokang/openwrt-22.03.0/staging_dir/hostpkg/include -I/home/kokang/openwrt-22.03.0/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Sep  3 02:55:34 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/kokang/openwrt-22.03.0/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/kokang/openwrt-22.03.0/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="" KERNELRELEASE=5.10.138 M="/home/kokang/openwrt-22.03.0/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug" EXTRA_CFLAGS=" -DCONFIG_BUTTON_HOTPLUG=1  " CONFIG_BUTTON_HOTPLUG=m modules
make[4]: Entering directory '/home/kokang/openwrt-22.03.0/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.138'
make[5]: *** No rule to make target '/home/kokang/openwrt-22.03.0/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/button-hotplug.o', needed by '/home/kokang/openwrt-22.03.0/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/button-hotplug.mod'.  Stop.
Makefile:1832: recipe for target '/home/kokang/openwrt-22.03.0/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug' failed
make[4]: *** [/home/kokang/openwrt-22.03.0/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug] Error 2
make[4]: Leaving directory '/home/kokang/openwrt-22.03.0/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.138'
Makefile:50: recipe for target '/home/kokang/openwrt-22.03.0/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/.built' failed
make[3]: *** [/home/kokang/openwrt-22.03.0/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/.built] Error 2
make[3]: Leaving directory '/home/kokang/openwrt-22.03.0/package/kernel/button-hotplug'
time: package/kernel/button-hotplug/compile#0.48#0.23#0.74
    ERROR: package/kernel/button-hotplug failed to build.
package/Makefile:114: recipe for target 'package/kernel/button-hotplug/compile' failed
make[2]: *** [package/kernel/button-hotplug/compile] Error 1
make[2]: Leaving directory '/home/kokang/openwrt-22.03.0'
package/Makefile:110: recipe for target '/home/kokang/openwrt-22.03.0/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/kokang/openwrt-22.03.0/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kokang/openwrt-22.03.0'
/home/kokang/openwrt-22.03.0/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2
kokang@virtual-machine:~/openwrt-22.03.0$

openwrtdiy, 2022-09-12:
After deleting the old source code, re-downloading the source code and compiling again solved the problem!

openwrtdiy, 2022-09-13:
Had the same problem on the openwrt forum, but no one helped!
https://forum.openwrt.org/t/22-03-build-fails-on-button-hotplug-package/133349

**22.03 build fails on button-hotplug package**

Freshly cloned 22.03 compiles fine with a bare x86 config (make menuconfig, and only select x86_64 target), but after I select packages in menuconfig it fails to build button-hotplug. It looks like a build byproduct isn't being cleaned up, so it doesn't re-make the generated makefile that includes the definition for button-hotplug.o. I made another build tree to compare side-by-side and it looks like .button-hotplug.o.cmd and .button-hotplug.mod.cmd never get re-generated.

Running make clean doesn't work, only make distclean works to clean it up and it will build after that if I use a bare config. As soon as I copy in the .config with all the packages selected, it fails. Any help would be appreciated.

Here's the end of the build output where it fails (V=sc):

ecrasy, 2022-10-16:
I got the same error on x86_64 config built with 22.03,
but building raspberry pi 4b with 22.03 was ok though.

ecrasy, 2022-10-17:
i solved the error by seperating the compile jobs like this:

          make buildinfo
          make diffconfig buildversion feedsversion
          make target/compile -j$(nproc)
          make package/kernel/button-hotplug/compile -j1 V=s 
          make package/compile -j$(nproc)
          make package/index

command "make package/kernel/button-hotplug/compile -j1 V=s " help me to work around the problem.

openwrtdiy, 2022-10-22:
The development version of openwrt-22.03 on October 21, 2022 can already compile x86x64 firmware normally! I repeated the compilation process twice and no error message appeared again.

ynezz, 2022-10-22:
Probably related https://github.com/openwrt/openwrt/pull/10917 ? What is your build host?

openwrtdiy, 2022-10-22:
> Probably related #10917 ? What is your build host?

Here is a screenshot of my build host
[image]
[image]

ynezz, 2022-10-22:
> Here is a screenshot of my build host [image] [image]

Thanks, but by build host, I mean a system on which you're building OpenWrt firmware image, where you're seeing this issue.

openwrtdiy, 2022-10-22:
> > Here is a screenshot of my build host
> 
> Thanks, but by build host, I mean a system on which you're building OpenWrt firmware image, where you're seeing this issue.

Yes! Had the same problem when building firmware for openwrt-22.03-snapshots, openwrt-22.03.0, openwrt-22.03.1 before!
When you compile it for the first time without any integrated plugins, you can compile it successfully! But when compiling for the second time and integrating the plug-in, there will be an error message indicating that the compilation was unsuccessful!

fannwong001, 2023-04-01:
> Yes! Had the same problem when building firmware for openwrt-22.03-snapshots, openwrt-22.03.0, openwrt-22.03.1 before! When you compile it for the first time without any integrated plugins, you can [...]

Problem still exists in 22.03.

miaozilong, 2023-04-11:
> 

希望能有用，我来试试！
太难了  可能是package/kernel/button-hotplug/compile这个包不能多线程编译

fannwong001, 2023-04-12:
> > 
> 
> 
> 
> 希望能有用，我来试试！
> 
> 太难了  可能是package/kernel/button-hotplug/compile这个包不能多线程编译

一般是在增删package后出现，还是依赖的问题。

openwrtdiy, 2023-04-14:
The latest openwrt stable version 22.03.4 still has the same problem!
最新openwrt稳定版22.03.4,依旧存在相同问题！

sdelafond, 2023-04-18:
I can confirm the issue is still present in 22.03.4, with  `kmod-button-hotplug` failing to build as described at the top of this issue. I have to use @ecrasy's workaround to build a custom image:

```
make buildinfo
make diffconfig buildversion feedsversion
make target/compile -j$(nproc)
make package/kernel/button-hotplug/compile -j1 V=s 
make package/compile -j$(nproc)
make package/index
```

The build host is a regular Debian bookworm install.

openwrtdiy, 2023-06-03:
https://forum.openwrt.org/t/solved-problem-compiling-22-03-4/157221/3

Installing the following software temporarily solves the problem! You can compile and see
**sudo apt-get install libelf-dev**