openwrt/openwrt #10374: EdgeSwitch 8XP - "Failed to connect to the switch..."

lister-wrt, 2022-08-03:
This device hasn't been working in OpenWrt for ~1 year now (in my experience). I worked around it by reverting to the factory image (all those months ago). 

I tried 21.02 and had the same problem. I have not tested 23.03rc yet but can if required. At the time I assumed the kernel had grown too large and I've been putting off looking into it ever since.

Now I've replaced it, I can finally see what's going on.

This "**Failed to connect to the switch. Use the "list" command to see which switches are available**." error is not something I've seen before. Hoping someone can enlighten me. 

Same result with the default/OOTB build configuration (5.10, GCC11, binutils 2.37). 

This model uses a BCM53128 switch. Port lights still function (link, no traffic) but I can't reach the switch via SSH.

Factory firmwares work normally when recovered.

Full boot log + build config below... Happy to try any suggestions. 

```
U-Boot 1.1.4.2-s559 (May 23 2012 - 13:31:49)

Board: Ubiquiti Networks XM board (rev 2.4 e702)
DRAM:  64 MB
Flash:  8 MB
Net:   eth0, eth1
Hit any key to stop autoboot:  0
## Booting image at 9f050000 ...
   Image Name:   MIPS OpenWrt Linux-5.10.134
   Created:      2022-07-31  18:30:20 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2220721 Bytes =  2.1 MB
   Load Address: 80060000
   Entry Point:  80060000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 5.10.134 (*@*) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.1.0 r20212-beeb49740b) 12.1.0, GNU ld (GNU Binutils) 2.38) #0 Sun Jul 31 18:30:20 2022
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] MIPS: machine is Ubiquiti EdgeSwitch 8XP
[    0.000000] SoC: Atheros AR7242 rev 1
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 56244K/65536K available (5416K kernel code, 595K rwdata, 1184K rodata, 1180K init, 200K bss, 9292K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 390.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 9801335621 ns
[    0.000013] sched_clock: 32 bits at 195MHz, resolution 5ns, wraps every 11012737021ns
[    0.007987] Calibrating delay loop... 259.27 BogoMIPS (lpj=1296384)
[    0.084174] pid_max: default: 32768 minimum: 301
[    0.089022] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096316] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.107412] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.120203] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.130050] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[... 179 lines cut ...]
# CONFIG_PACKAGE_kmod-crypto-cmac is not set
# CONFIG_PACKAGE_kmod-crypto-ctr is not set
# CONFIG_PACKAGE_kmod-crypto-gcm is not set
# CONFIG_PACKAGE_kmod-crypto-gf128 is not set
# CONFIG_PACKAGE_kmod-crypto-ghash is not set
# CONFIG_PACKAGE_kmod-crypto-hmac is not set
# CONFIG_PACKAGE_kmod-crypto-manager is not set
# CONFIG_PACKAGE_kmod-crypto-null is not set
# CONFIG_PACKAGE_kmod-crypto-rng is not set
# CONFIG_PACKAGE_kmod-crypto-seqiv is not set
# CONFIG_PACKAGE_kmod-crypto-sha256 is not set
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
# CONFIG_PACKAGE_kmod-mac80211 is not set
# CONFIG_PACKAGE_kmod-ppp is not set
# CONFIG_PACKAGE_libiwinfo is not set
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_nano=y
# CONFIG_PACKAGE_odhcp6c is not set
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_terminfo=y
# CONFIG_PACKAGE_wireless-regdb is not set
# CONFIG_PACKAGE_wpad-basic-wolfssl is not set
# CONFIG_WPA_WOLFSSL is not set
```

lister-wrt, 2022-08-06:
Confirmed the issue is present in 22.03rc6. I can't contact the device on any port.

Logs from 22.03rc6
```
Board: Ubiquiti Networks XM board (rev 2.4 e702)
DRAM:  64 MB
Flash:  8 MB
Net:   eth0, eth1
Hit any key to stop autoboot:  0
## Booting image at 9f050000 ...
   Image Name:   MIPS OpenWrt Linux-5.10.134
   Created:      2022-07-31  15:12:47 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2242984 Bytes =  2.1 MB
   Load Address: 80060000
   Entry Point:  80060000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 5.10.134 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19590-042d558536) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Sun Jul 31 15:12:47 2022
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] MIPS: machine is Ubiquiti EdgeSwitch 8XP
[    0.000000] SoC: Atheros AR7242 rev 1
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 56172K/65536K available (5845K kernel code, 608K rwdata, 744K rodata, 1240K init, 209K bss, 9364K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 390.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 9801335621 ns
[    0.000014] sched_clock: 32 bits at 195MHz, resolution 5ns, wraps every 11012737021ns
[    0.008001] Calibrating delay loop... 259.27 BogoMIPS (lpj=1296384)
[    0.084175] pid_max: default: 32768 minimum: 301
[    0.089024] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096313] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.107311] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.120141] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.129990] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.136912] pinctrl core: initialized pinctrl subsystem
[    0.148883] NET: Registered protocol family 16
[    0.154565] thermal_sys: Registered thermal governor 'step_wise'
[    0.237332] clocksource: Switched to clocksource MIPS
[    0.250284] NET: Registered protocol family 2
[    0.255024] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.263601] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.272084] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.279750] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.286779] TCP: Hash tables configured (established 1024 bind 1024)
[    0.293346] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.299934] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.307381] NET: Registered protocol family 1
[    0.311761] PCI: CLS 0 bytes, default 32
[    0.325335] workingset: timestamp_bits=14 max_order=14 bucket_order=0
[... 117 lines cut ...]
[   56.174565] device eth0.1 entered promiscuous mode
[   56.179460] device eth0 entered promiscuous mode
[   56.373595] br-lan: port 1(eth0.1) entered blocking state
[   56.379107] br-lan: port 1(eth0.1) entered forwarding state
[   56.442522] eth1: link up (1000Mbps/Full duplex)
[   56.447800] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   57.147509] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

BusyBox v1.35.0 (2022-07-31 15:12:47 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 22.03.0-rc6, r19590-042d558536
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:/#
```

lister-wrt, 2022-08-06:
22.02.3 logs
```
U-Boot 1.1.4.2-s559 (May 23 2012 - 13:31:49)

Board: Ubiquiti Networks XM board (rev 2.4 e702)
DRAM:  64 MB
Flash:  8 MB
Net:   eth0, eth1
Hit any key to stop autoboot:  0
## Booting image at 9f050000 ...
   Image Name:   MIPS OpenWrt Linux-5.10.134
   Created:      2022-07-31  15:12:47 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2242984 Bytes =  2.1 MB
   Load Address: 80060000
   Entry Point:  80060000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 5.10.134 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19590-042d558536) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Sun Jul 31 15:12:47 2022
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] MIPS: machine is Ubiquiti EdgeSwitch 8XP
[    0.000000] SoC: Atheros AR7242 rev 1
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 56172K/65536K available (5845K kernel code, 608K rwdata, 744K rodata, 1240K init, 209K bss, 9364K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 390.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 9801335621 ns
[    0.000014] sched_clock: 32 bits at 195MHz, resolution 5ns, wraps every 11012737021ns
[    0.008001] Calibrating delay loop... 259.27 BogoMIPS (lpj=1296384)
[    0.084175] pid_max: default: 32768 minimum: 301
[    0.089024] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096313] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.107311] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.120141] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.129990] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.136912] pinctrl core: initialized pinctrl subsystem
[    0.148883] NET: Registered protocol family 16
[    0.154565] thermal_sys: Registered thermal governor 'step_wise'
[    0.237332] clocksource: Switched to clocksource MIPS
[    0.250284] NET: Registered protocol family 2
[    0.255024] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.263601] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.272084] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.279750] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.286779] TCP: Hash tables configured (established 1024 bind 1024)
[    0.293346] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.299934] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.307381] NET: Registered protocol family 1
[    0.311761] PCI: CLS 0 bytes, default 32
[    0.325335] workingset: timestamp_bits=14 max_order=14 bucket_order=0
[... 313 lines cut ...]
[   16.621645] Backport generated by backports.git v5.10.110-1-0-g1fbde860
[   16.705165] xt_time: kernel timezone is -0000
[   17.085599] PPP generic driver version 2.4.2
[   17.099263] NET: Registered protocol family 24
[   17.241088] kmodloader: done loading kernel modules from /etc/modules.d/*
[   17.412122] urngd: v1.0.2 started.
[   17.712143] crng init done

BusyBox v1.33.2 (2022-04-16 12:59:34 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 21.02.3, r16554-1d4dea6d4f
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:/#
```

rogerpueyo, 2022-09-04:
Hi,

I'm facing the same issue you described. I could observe that, after flashing `v22.03.0-rc6`, `swconfig` does not report any switch:
```
[...]
 -----------------------------------------------------
 OpenWrt 22.03.0-rc6, r19590-042d558536
 -----------------------------------------------------
[...]
--------------------------------------------------
root@OpenWrt:~# swconfig list
root@OpenWrt:~# 
```

while `v21.02.3` did:
```
 -----------------------------------------------------
 OpenWrt 21.02.3, r16554-1d4dea6d4f
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# swconfig list
Found: switch0 - bcm53128
```

My bet is that the switch is not detected and configured with 22.03/snapshot and only warm boot works because it is not reset. 

Entering in failsafe mode doesn't help in 22.03. 

In both 21.02 and 22.03, I see the `b53_mdio` and `b53_common` modules are loaded.

lister-wrt, 2022-09-05:
Glad to see I'm not the only one. I was starting to think it might be my build host.

The last thing I tried was a 21.02 snapshot, it had the same issue. I did select the latest gcc/binutils available though. I'll try without that when I get home. 

Thanks for the motivation 😄

rogerpueyo, 2022-09-05:
> Glad to see I'm not the only one. I was starting to think it might be my build host.

It was very good to find your bug report too, at first I thought my device was damaged and I needed to replace it. :smiley:

> The last thing I tried was a 21.02 snapshot, it had the same issue. I did select the latest gcc/binutils available though. I'll try without that when I get home.

Did you compile it yourself? The binary from https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/ath79/generic/ works well for me, `swconfig list`does find the switch.

I've been building several images to try to narrow down which commit introduced the bug/regression. If I find it, I'll report it here.

rogerpueyo, 2022-09-06:
Hi, @lister-wrt,

Well, this is getting worse. I built commit [`r11962-b6946542f8`](https://github.com/openwrt/openwrt/commit/b6946542f85014096c08f58c3c0acbbfe623a734), which adds support for the device. To my surprise, the switch is not detected anymore:

```
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r11962-b6946542f8
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# swconfig list
root@OpenWrt:~# uname -a
Linux OpenWrt 4.19.93 #0 Sun Jan 12 13:12:50 2020 mips GNU/Linux
root@OpenWrt:~# 
```

rogerpueyo, 2022-09-06:
Thanks, you're right. No kernel module == no switch :-)

I'll keep investigating.

rogerpueyo, 2022-09-06:
Hi,

I narrowed down the bug to the following commits:

- `r17692-`[`e4e410733f`](https://github.com/openwrt/openwrt/commit/e4e410733f3b0cad09d3a235eec0b04f4068c331)` (kernel: export switch_generic_set_link() symbol) => **switch not detected -- not working**
- `r17691-`[`2e4193f3cf`](https://github.com/openwrt/openwrt/commit/2e4193f3cfa23f6c85f479ef10bf779406538f3c)` (kernel: 5.10: dsa: don't set skb->offload_fwd_mark when not offloading bridge) => compilation fails
- `r17690-`[`b2cfed48f6`](https://github.com/openwrt/openwrt/commit/b2cfed48f62622efa8577372942bbc268afe4677)`(Revert "swconfig: fix Broadcom b53 support") => compilation fails
- `r17689-`[`fd71ef34b7`](https://github.com/openwrt/openwrt/commit/fd71ef34b75c81646d17d21d80dc3a5b5e2e6bb8)` (kernel: backport bgmac upstream commits from 5.15 / for 5.16) => **switch detected -- working**

So it seems b2cfed48f6 breaks the compilation, e4e410733f fixes the compilation, but something is left that makes the switch not being detected anymore.

rogerpueyo, 2022-09-06:
Quick update:

I got the switch working with current `master`, after reverting first e4e410733f3b0cad09d3a235eec0b04f4068c331 and second b2cfed48f62622efa8577372942bbc268afe4677.

```
BusyBox v1.35.0 (2022-09-06 21:59:17 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r20499+2-33c11442b2
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# swconfig list
Found: switch0 - bcm53128
root@OpenWrt:~# uname -a
Linux OpenWrt 5.10.139 #0 Tue Sep 6 21:59:17 2022 mips GNU/Linux
```

@lister-wrt, do you feel like digging further into it and creating a patch to fix it?

lister-wrt, 2022-09-06:
@rogerpueyo nice work tracking that down. Unfortunately I haven't found the time to look at it again (I've been on-call).

Having just read through the discussion linked in the revert, I think it's beyond me anyway. 

Maybe @rmilecki could give us some background?

rogerpueyo, 2022-09-07:
Hi, @rmilecki,

Quick summary here. The `bcm53128` switch in the Ubiquiti EdgeSwitch 8XP is no longer detected. Commits related, from old to new:
- 8f9cd1af0f9c325a902dbd0e79e12015372e6bb0 (swconfig: fix Broadcom b53 support) => seems to be working correctly
- b2cfed48f62622efa8577372942bbc268afe4677 (Revert "swconfig: fix Broadcom b53 support") => reverts the above, but breaks compilation
- e4e410733f3b0cad09d3a235eec0b04f4068c331 (kernel: export switch_generic_set_link() symbol) => fixes b2cfed48f62622efa8577372942bbc268afe4677 and compilation works again, but the switch is no longer detected

rmilecki, 2022-09-07:
Nice work @rogerpueyo.

So BCM53128 in Ubiquiti EdgeSwitch 8XP was supported for a moment by a code that **may work** but is **incorrect** in general. It causes lockups on other devices and that's why it was reverted.

There are two **correct** solutions to this issue:
1. Make OpenWrt's downstream `b53` driver (swconfig based) a proper MDIO driver (replace `phy_driver` with `mdio_driver` in the `target/linux/generic/files/drivers/net/phy/b53/b53_mdio.c`).
2. Switch Ubiquiti EdgeSwitch 8XP to use upstream `b53` driver (DSA based)

Personally I think the later option is a better and long term solution.

rmilecki, 2022-09-07:
I see that `target/linux/ath79/dts/ar7242_ubnt_edgeswitch-8xp.dts` includes binding for the BCM53128 switch. Can someone try compiling Ubiquiti EdgeSwitch 8XP firmware with upstream `b53` driver (`CONFIG_B53=y` and `# CONFIG_SWCONFIG_B53 is not set`).

lister-wrt, 2022-09-13:
@rmilecki where should I change this? I see where CONFIG_B53/CONFIG_SWCONFIG_B53 exist in config-5.** files for other targets but not ath79

rogerpueyo, 2022-09-14:
@lister-wrt 

I did this:
```
diff --git a/target/linux/ath79/config-5.10 b/target/linux/ath79/config-5.10
index bdc4b74508..dd90cbd8be 100644
--- a/target/linux/ath79/config-5.10
+++ b/target/linux/ath79/config-5.10
@@ -186,6 +186,8 @@ CONFIG_SPI_MEM=y
 CONFIG_SRCU=y
 CONFIG_SWCONFIG=y
 CONFIG_SWCONFIG_LEDS=y
+CONFIG_B53=y
+# CONFIG_SWCONFIG_B53 is not set
 CONFIG_SWPHY=y
 CONFIG_SYSCTL_EXCEPTION_TRACE=y
 CONFIG_SYS_HAS_CPU_MIPS32_R2=y
```
I don't have the device at hand. Would you like to test this sysupgrade image? [openwrt-ath79-generic-ubnt_edgeswitch-8xp-squashfs-sysupgrade.bin.gz](https://github.com/openwrt/openwrt/files/9565887/openwrt-ath79-generic-ubnt_edgeswitch-8xp-squashfs-sysupgrade.bin.gz). It is gzipped.

Cheers!

lister-wrt, 2022-09-17:
@rogerpueyo same result here.

kmod-switch-bcm53xx, kmod-switch-bcm53xx-mdio and kmod-swconfig are in the manifest... Maybe they, or at least swconfig need to be excluded? On other DSA devices I have, swconfig is not present. 

I'm not sure what @rmilecki was expecting us to see. Nothing stands out to me.
```
U-Boot 1.1.4.2-s559 (May 23 2012 - 13:31:49)

Board: Ubiquiti Networks XM board (rev 2.4 e702)
DRAM:  64 MB
Flash:  8 MB
Net:   eth0, eth1
Hit any key to stop autoboot:  0
## Booting image at 9f050000 ...
   Image Name:   MIPS OpenWrt Linux-5.15.67
   Created:      2022-09-16   7:17:24 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2345554 Bytes =  2.2 MB
   Load Address: 80060000
   Entry Point:  80060000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 5.15.67 (@) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r20648-8010d3da03) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 Fri Sep 16 07:17:24 2022
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] MIPS: machine is Ubiquiti EdgeSwitch 8XP
[    0.000000] SoC: Atheros AR7242 rev 1
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 55788K/65536K available (5762K kernel code, 589K rwdata, 1248K rodata, 1224K init, 207K bss, 9748K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 390.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 9801335621 ns
[    0.000002] sched_clock: 32 bits at 195MHz, resolution 5ns, wraps every 11012737021ns
[    0.007988] Calibrating delay loop... 259.27 BogoMIPS (lpj=1296384)
[    0.084177] pid_max: default: 32768 minimum: 301
[    0.089125] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096440] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.107692] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.120358] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.130226] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.137175] pinctrl core: initialized pinctrl subsystem
[    0.144692] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.177363] clocksource: Switched to clocksource MIPS
[    0.183859] NET: Registered PF_INET protocol family
[    0.189186] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.197462] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.205845] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.213607] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.221268] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.228323] TCP: Hash tables configured (established 1024 bind 1024)
[    0.234861] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[... 100 lines cut ...]
[   15.274169] procd: - ubus -
[   15.485998] random: ubusd: uninitialized urandom read (4 bytes read)
[   15.494901] random: ubusd: uninitialized urandom read (4 bytes read)
[   15.515570] random: ubusd: uninitialized urandom read (4 bytes read)
[   15.534898] procd: - init -
Please press Enter to activate this console.
[   16.865931] random: jshn: uninitialized urandom read (4 bytes read)
[   18.327449] kmodloader: loading kernel modules from /etc/modules.d/*
[   19.015052] xt_time: kernel timezone is -0000
[   19.258090] kmodloader: done loading kernel modules from /etc/modules.d/*
[   19.454320] urngd: v1.0.2 started.
[   19.842547] random: crng init done
[   19.845996] random: 20 urandom warning(s) missed due to ratelimiting
[   35.845074] eth0: link up (1000Mbps/Full duplex)
[   35.870289] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   35.881297] br-lan: port 1(eth0.1) entered blocking state
[   35.886768] br-lan: port 1(eth0.1) entered disabled state
[   35.892758] device eth0.1 entered promiscuous mode
[   35.897652] device eth0 entered promiscuous mode
[   35.951700] br-lan: port 1(eth0.1) entered blocking state
[   35.957165] br-lan: port 1(eth0.1) entered forwarding state
[   36.256356] eth1: link up (1000Mbps/Full duplex)
[   36.271610] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   36.837498] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
```

rogerpueyo, 2022-11-18:
Hi @lister-wrt, @rmilecki,

Sorry for the delay, I've been very busy recently. Good news: I got the switch to work using the upstream DSA-based `b53` driver.

The patch is at https://github.com/rogerpueyo/openwrt/commits/ath79-ubnt-edgeswitch-8xp-b53-dsa_wip (still work in progress).

You may want to give a try to this image attached (current snapshot + the commit above): [openwrt-ath79-generic-ubnt_edgeswitch-8xp-squashfs-sysupgrade.bin.gz](https://github.com/openwrt/openwrt/files/10036863/openwrt-ath79-generic-ubnt_edgeswitch-8xp-squashfs-sysupgrade.bin.gz)

I hope to be submitting a proper pull request at some point.

rogerpueyo, 2022-11-22:
Hi, @lister-wrt,

I'll be submitting a pull request with the code from https://github.com/rogerpueyo/openwrt/commits/ath79-ubnt-edgeswitch-8xp-b53-dsa soon. I'm attaching a sysupgrade image, in case you'd like to give it another try: [openwrt-ath79-generic-ubnt_edgeswitch-8xp-squashfs-sysupgrade.bin.gz](https://github.com/openwrt/openwrt/files/10065694/openwrt-ath79-generic-ubnt_edgeswitch-8xp-squashfs-sysupgrade.bin.gz).

Cheers!

lister-wrt, 2022-11-22:
@rogerpueyo I'll be able to test it when I'm home on Friday.

lister-wrt, 2022-12-11:
@rogerpueyo life had other plans. I tested your patches work against current master, all seems to be working!

```dmesg
[    0.000000] Linux version 5.15.82 (*@*) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.2.0 r21399-644175c29c) 12.2.0, GNU ld (GNU Binutils) 2.39) #0 Sun Dec 11 02:30:14 2022
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] MIPS: machine is Ubiquiti EdgeSwitch 8XP
[    0.000000] SoC: Atheros AR7242 rev 1
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 55592K/65536K available (5928K kernel code, 583K rwdata, 1296K rodata, 1208K init, 207K bss, 9944K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 390.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 9801335621 ns
[    0.000002] sched_clock: 32 bits at 195MHz, resolution 5ns, wraps every 11012737021ns
[    0.007995] Calibrating delay loop... 259.27 BogoMIPS (lpj=1296384)
[    0.084177] pid_max: default: 32768 minimum: 301
[    0.089164] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096483] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.108018] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.120731] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.130602] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.137552] pinctrl core: initialized pinctrl subsystem
[    0.145083] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.178416] clocksource: Switched to clocksource MIPS
[    0.184879] NET: Registered PF_INET protocol family
[    0.190219] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.198496] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.206879] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.214625] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.222284] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.229340] TCP: Hash tables configured (established 1024 bind 1024)
[    0.235878] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.242480] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.249917] NET: Registered PF_UNIX/PF_LOCAL protocol family
[    0.255624] PCI: CLS 0 bytes, default 32
[    0.264471] workingset: timestamp_bits=14 max_order=14 bucket_order=0
[    0.278947] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.284765] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.298065] pinctrl-single 18040028.pinmux: 64 pins, size 8
[    0.305358] Serial: 8250/16550 driver, 1 ports, IRQ sharing disabled
[    0.312859] printk: console [ttyS0] disabled
[    0.317243] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 9, base_baud = 12187500) is a 16550A
[    0.326033] printk: console [ttyS0] enabled
[    0.334444] printk: bootconsole [early0] disabled
[    0.361308] gpio_spi@0 enforce active low on chipselect handle
[    0.380548] spi-nor spi0.0: mx25l6405d (8192 Kbytes)
[    0.385668] 5 fixed-partitions partitions found on MTD device spi0.0
[    0.392175] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.398910] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.405990] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.412752] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.420126] Creating 5 MTD partitions on "spi0.0":
[... 127 lines cut ...]
[   25.391547] bcm53xx mdio.0:1e lan4: Link is Up - 100Mbps/Full - flow control rx/tx
[   25.421082] br-lan: port 4(lan4) entered blocking state
[   25.426382] br-lan: port 4(lan4) entered disabled state
[   25.461865] device lan4 entered promiscuous mode
[   25.548725] br-lan: port 4(lan4) entered blocking state
[   25.554025] br-lan: port 4(lan4) entered forwarding state
[   25.594343] bcm53xx mdio.0:1e lan5: configuring for phy/gmii link mode
[   25.629223] br-lan: port 5(lan5) entered blocking state
[   25.634517] br-lan: port 5(lan5) entered disabled state
[... cut ...]
[   25.982114] device lan7 entered promiscuous mode
[   26.094618] bcm53xx mdio.0:1e lan8: configuring for phy/gmii link mode
[   26.129216] br-lan: port 8(lan8) entered blocking state
[   26.134515] br-lan: port 8(lan8) entered disabled state
[   26.142233] device lan8 entered promiscuous mode
[   28.738947] bcm53xx mdio.0:1e lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[   28.746462] br-lan: port 1(lan1) entered blocking state
[   28.751796] br-lan: port 1(lan1) entered forwarding state```

areqq, 2023-11-02:
any chance for working latest openwrt on Ubiquiti EdgeSwitch 8XP?

rogerpueyo, 2023-11-03:
Not really, the current b53 switch driver (as it is now) is broken and won't be fixed. The proper way to go is using the DSA b53 driver. It works when the driver is built into the kernel, but not when built as a module. Any help will be more than appreciated.

areqq, 2023-11-03:
so, why not built  the driver into the kernel ?

rogerpueyo, 2023-11-03:
This was my first approach, but it means that the driver is added to the images of all the `ath79/generic` devices. Since this driver is 8xp-specific, and not used by any other device, it is a bit overkill.

PR #11680 (2024-10-19): ath79: Use upstream b53 DSA driver for Ubiquiti EdgeSwitch 8XP
The swconfig-based b53 driver for the BCM53128 switch stopped working after commits b2cfed4 (Revert "swconfig: fix Broadcom b53 support") and e4e4107 (kernel: export switch_generic_set_link() symbol).

This commit enables DSA for the ath79/generic target/subtarget, adds the upstream b53 MDIO switch driver to the kernel config, removes the kmod-switch-bcm53xx-mdio package from the image Makefile and updates the 02_network init script with the new interfaces. Finally, it reenables image building for the device.

Fixes #10374.

Tested [... cut]