openwrt/openwrt #10384: Option metric in /etc/config/network not correctly processed.

epmsmid, 2022-08-04:
The metric is a 32 bit unsigned integer, however the value seems to be maximized at '0x7fffffff'.
This causes problems when using redistribute kernel in the FRR BGP module, where the upper byte is used as distance and the lower 3 bytes are used as the metric.

Configuration in "/etc/config/network":

  config interface 'wan'                        
          option device 'eth0.2'                
          option proto 'dhcp'                   
          option metric '0xd200000f'

Output of "ip route":
  default via 172.25.3.1 dev eth0.2 proto static src 172.25.3.2 metric 2147483647 (This is '0x7fffffff')

Configuration in "/etc/config/network":
  config interface 'wan'                        
          option device 'eth0.2'                
          option proto 'dhcp'                   
          option metric '0x7f00000f'

Output of "ip route":
  default via 172.25.3.1 dev eth0.2 proto static src 172.25.3.2 metric 2130706447 (This is '0x7f00000f')

When I manually add a route with "ip route add metric" metric is correct. E.g.:
  ip route add 8.8.8.8/32 via 172.25.3.2 metric 0xd200000f

Output of "ip route":
  8.8.8.8 via 172.25.3.2 dev eth0.2 metric 3523215375 (This is '0xd200000f')  

Model: Netgear Nighthawk X4S R7800
Architecture: ARMv7 Processor rev 0 (v7l)
Target Platform: ipq806x/generic
Firmware Version: OpenWrt 21.02.2 r16495-bf0c965af0 / LuCI openwrt-21.02 branch git-22.167.28411-ee8170b
Kernel Version:5.4.179

aparcar, 2022-08-07:
@nbd168 this look like an issue of netifd right?

john-tho, 2022-09-06:
It looks to me like this comes from netifd using an int32 to store metric, and uci using strtol to initially store this metric.

On my ipq40xx arm device, LONG_MAX is 0x7fff ffff
metric can overflow INT32_MAX, and is then set to that max by strtol

from `config_parse_interface(struct uci_section *s, bool alias)`
https://lxr.openwrt.org/source/netifd/config.c#L188
```c
uci_to_blob(&b, s, &interface_attr_list);

iface = interface_alloc(s->e.name, b.head, false);
```

`uci_to_blob` eventually uses `uci_attr_to_blob`
https://lxr.openwrt.org/source/uci/blob.c#L45
```c
case BLOBMSG_TYPE_INT32:
    intval = strtol(str, &err, 0);
```

From man strtol:
If  an overflow occurs, strtol() returns LONG_MAX.
       In both cases, errno is set to ERANGE

The [iproute2 utils do check metric is u32](https://git.kernel.org/pub/scm/network/iproute2/iproute2.git/tree/ip/iproute.c?id=f1d037ab4ad3d4af7b77ae4f8969d658e04b5d3f#n1221)
```c
			if (get_u32(&metric, *argv, 0))
				invarg("\"metric\" value is invalid\n", *argv);
			addattr32(&req.n, sizeof(req), RTA_PRIORITY, metric);
```

A quick and nasty patch. Would probably want to add a ceiling of UINT32_MAX somewhere.
```patch
From 6ccb71234e764726a96d8545b1fcd54823cfa945 Mon Sep 17 00:00:00 2001
From: John Thomson <git@johnthomson.fastmail.com.au>
Date: Mon, 5 Sep 2022 20:05:10 +1000
Subject: [PATCH] interface: store metric int64 to allow metric>INT32_MAX

A metric value > INT32_MAX (0x80000000-1) stored as INT32 overflows strtol,
which used by uci/blob.c in uci_attr_to_blob. strtol then uses INT32_MAX as the metric. Use BLOBMSG_TYPE_INT64 to allow metric values up to UINT32_MAX.

---
 interface-ip.c | 14 +++++++-------
 interface-ip.h |  4 ++--
 interface.c    |  4 ++--
 interface.h    |  2 +-
 4 files changed, 12 insertions(+), 12 deletions(-)

diff --git a/interface-ip.c b/interface-ip.c
index ab4a5cf..fadace8 100644
--- a/interface-ip.c
+++ b/interface-ip.c
@@ -52,7 +52,7 @@ static const struct blobmsg_policy route_attr[__ROUTE_MAX] = {
 	[ROUTE_TARGET] = { .name = "target", .type = BLOBMSG_TYPE_STRING },
 	[ROUTE_MASK] = { .name = "netmask", .type = BLOBMSG_TYPE_STRING },
 	[ROUTE_GATEWAY] = { .name = "gateway", .type = BLOBMSG_TYPE_STRING },
-	[ROUTE_METRIC] = { .name = "metric", .type = BLOBMSG_TYPE_INT32 },
+	[ROUTE_METRIC] = { .name = "metric", .type = BLOBMSG_TYPE_INT64 },
 	[ROUTE_MTU] = { .name = "mtu", .type = BLOBMSG_TYPE_INT32 },
 	[ROUTE_TABLE] = { .name = "table", .type = BLOBMSG_TYPE_STRING },
 	[ROUTE_VALID] = { .name = "valid", .type = BLOBMSG_TYPE_INT32 },
@@ -451,7 +451,7 @@ interface_ip_add_route(struct interface *iface, struct blob_attr *attr, bool v6)
 	}
 
 	if ((cur = tb[ROUTE_METRIC]) != NULL) {
-		route->metric = blobmsg_get_u32(cur);
+		route->metric = blobmsg_get_u64(cur);
 		route->flags |= DEVROUTE_METRIC;
 	}
 
[... 81 lines cut ...]
 		iface->dns_metric = blobmsg_get_u32(cur);
 
 	if ((cur = tb[IFACE_ATTR_METRIC]))
-		iface->metric = blobmsg_get_u32(cur);
+		iface->metric = blobmsg_get_u64(cur);
 
 	if ((cur = tb[IFACE_ATTR_IP6ASSIGN]))
 		iface->assignment_length = blobmsg_get_u32(cur);
diff --git a/interface.h b/interface.h
index 9343ade..4d565ee 100644
--- a/interface.h
+++ b/interface.h
@@ -154,7 +154,7 @@ struct interface {
 	struct vlist_tree host_routes;
 	struct vlist_tree host_neighbors;
 
-	int metric;
+	unsigned int metric;
 	int dns_metric;
 	unsigned int ip4table;
 	unsigned int ip6table;
-- 
2.37.2

```