openwrt/openwrt #10385: DIR-853 A3 wifi signal strength

sunyoung1, 2022-08-04:
I flashed the latest release candidate for this router and it only have 9dBm (7mW) transmit power. I changed the country code to anything and didn't work. also i don't know if it is because of the signal strength but i can't connect to my openwrt mesh network.

ahsaan51486, 2022-09-12:
I have also tried latest release and encoutered same issue, is there any update on this ?

21Z, 2023-11-26:
same issue here, when i get out of the room i have bad signal strength which i didnt have with [padavan](https://github.com/hanwckf/rt-n56u) or stock firmware.
https://forum.openwrt.org/t/dlink-dir-853-a3/70548/498

davegermiquet, 2024-02-16:
Same issue for me as well

davegermiquet, 2024-02-16:
Based on specs:

Transmitter powerIn accordance with the standards, the maximum power for a 2.4 GHz network is set at 20 dBM, for a 5 GHz network 24 dBM. The standards were introduced with the aim of limiting the influence of the electromagnetic field on the human body. | 15 dBm
-- | --

noes14155, 2025-01-31:
Same issue. Any update on this?

ahsaan51486, 2025-02-01:
Mine working fine since then, its just UI Bug shows wrong information, There is no difference in coverage or network quality just shows low transmit power.

noes14155, 2025-02-01:
But download speed is very low, only getting 70-90mbps. I have a 300mbps connection. Once i lower the 2.4GHz power to 15dbm 5GHz speed increases.

planetmh711-git, 2026-02-24:
@Lahore7861 thanks for the proposed solution. while the patch may enhance link speeds, its impact on signal strength is anticipated to be minimal.

A relevant patch for the same Wi-Fi chip has been identified within the Git repository. I will provide the details of this patch, which requires integration into a specific directory during the source code compilation process.

Subject: [PATCH] mt7615: fix exPA TX target power value
---
 mt7615/eeprom.c | 9 ---------
 mt7615/init.c   | 2 +-
 2 files changed, 1 insertion(+), 10 deletions(-)

--- a/mt7615/eeprom.c
+++ b/mt7615/eeprom.c
@@ -204,15 +204,6 @@ int mt7615_eeprom_get_target_power_index
 	if (chain_idx > 3)
 		return -EINVAL;
 
-	/* TSSI disabled */
-	if (mt7615_ext_pa_enabled(dev, chan->band)) {
-		if (chan->band == NL80211_BAND_2GHZ)
-			return MT_EE_EXT_PA_2G_TARGET_POWER;
-		else
-			return MT_EE_EXT_PA_5G_TARGET_POWER;
-	}
-
-	/* TSSI enabled */
 	if (chan->band == NL80211_BAND_2GHZ) {
 		index = MT_EE_TX0_2G_TARGET_POWER + chain_idx * 6;
 	} else {
--- a/mt7615/init.c
+++ b/mt7615/init.c
@@ -285,7 +285,7 @@ void mt7615_init_txpower(struct mt7615_d
 	    (MT_EE_RATE_POWER_EN | MT_EE_RATE_POWER_SIGN))
 		delta += rate_val & MT_EE_RATE_POWER_MASK;
 
-	if (!is_mt7663(&dev->mt76) && mt7615_ext_pa_enabled(dev, band))
+	if (!is_mt7663(&dev->mt76) && !mt7615_ext_pa_enabled(dev, band))
 		target_chains = 1;
 	else
 		target_chains = n_chains;

azdxb, 2026-09-04:
### **Confirmed on a DIR-853 A3 (OpenWrt 25.12.4, kernel 6.12.87). Adding fresh evidence + related-ticket references.**

**Verified on this unit (commands below):**
- ubus call system board → model "D-Link DIR-853 A3", board dlink,dir-853-a3, OpenWrt 25.12.4, kernel 6.12.87.
- iw phy phy1 info → every 5GHz channel reports 9.0 dBm (channels 36–165).
- iw reg get → country 00; regdb allows 20 dBm on 5GHz UNII-1 (5170–5250) and UNII-3 (5735–5835) - so regulatory is not the limit.
- iw dev phy1-ap0 set txpower fixed 2300 → succeeds with no error, radio stays at 9 dBm (driver ignores the setting).
- mt76 debugfs EEPROM (/sys/kernel/debug/ieee80211/phy0/mt76/eeprom): 0x0F3=04 (ext-PA 5G target), 0x0D5=C5, 0x058=24 (2.4GHz per-chain target → runs 20 dBm).
- 2.4GHz on the same chip: 20 dBm, working.
- Measured (~5 m, one wall): 5GHz 30.3 down / 122 up; 2.4GHz 28/26. The 5GHz down/up gap is the signature of an AP-side TX-power cap, not a regulatory or ISP issue.

**How this fits existing reports (all verified open/matching at URLs below):**
- [openwrt/mt76 #531](https://github.com/openwrt/mt76/issues/531
) — MT7615DN: Phicomm K2P with external PA capped at 9 dBm vs JCG Y2 with internal PA reaching 23 dBm. Same chip; the external-PA path is the clamp. This matches DIR-853 A3 exactly.
- [openwrt/mt76 #633](https://github.com/openwrt/mt76/issues/633
) — umbrella: TX power not working across MT7610/7613/7615/7603 (open).
- [openwrt/mt76 #657](https://github.com/openwrt/mt76/issues/657
) — mt76 reports 4–10 dBm below OEM/MediaTek drivers across MT7621+MT76 combos (closed).
- [openwrt/mt76 #804 ](https://github.com/openwrt/mt76/issues/804
)— MT7621/MT7615N 5GHz problems (open).
- [ forum.openwrt.org/t/107392 ](https://forum.openwrt.org/t/solved-openwrt-on-mt7621-mt7615n-devices-with-5ghz-problems/107392)— "MT7621/MT7615N 5GHz problems" — contains the mt7615: fix exPA TX target power value patch discussion (external-PA boards → ~18–20 dBm). Patch is community-posted, not yet merged; not independently tested here.

**Request:** please review/upstream the mt7615 exPA TX target power fix (mt76 #804 / thread 107392) and consider honoring a user-set txpower ceiling for external-PA devices bounded by regdb (20 dBm) — matching the vendor firmware's behavior.