openwrt/openwrt #10389: ujail: changes to resolv.conf may not be reflected in jail

janh, 2022-08-04:
The `/etc/resolv.conf` within a jail is a bind mount of the real `/etc/resolv.conf`. If `/etc/resolv.conf` (or the destination of the symlink) is replaced, then that change won't be visible within the jail (it still sees the old file).

As long dnsmasq is installed and used as local resolver that is not an issue, as `/tmp/resolv.conf` doesn't change in this case (it specifies 127.0.0.1 and ::1, so only dnsmasq has to care about the real nameservers).

However, it is a problem if there is no local resolver, as the link then points to `/tmp/resolv.conf.d/resolv.conf.auto` which can get replaced during normal operation, for example when `/etc/init.d/network` is started or stopped. This means that if a jailed service is started while the network is not (yet) configured, it will never have working DNS (`resolv.conf.auto` is empty in that case, so DNS queries will always be sent to localhost, and fail if there is no local resolver).

The actual example where I found this issue is sysntpd, which as a result won't ever synchronize the time if it is started before nameservers are configured.

jow-, 2022-08-05:
The autogenerated resolv files got moved into the `/tmp/resolv.conf.d/` for exactly this reason, so that jails can bind mount the parent directory.

janh, 2022-08-08:
> The autogenerated resolv files got moved into the `/tmp/resolv.conf.d/` for exactly this reason, so that jails can bind mount the parent directory.

This works for dnsmasq as it is specifically interested in the autogenerated resolv file.

But I'm not sure if bind mounting `/tmp/resolv.conf.d/` is a proper solution for a regular jailed service that just wants DNS to work. It seems like that wouldn't be enough to make jailed and unjailed services have a consistent view of `/etc/resolv.conf` (since `/tmp/resolv.conf` doesn't always link to the autogenerated resolv file).

Hurricos, 2022-11-22:
In particular, `/tmp/resolv.conf` is not linked to the autogenerated resolv file at boot when, for example, `sysntpd` starts. The only reason the jailing of `ntpd` hasn't broken NTP across all OpenWrt users is that most of them happen to also have a DNS server running at 127.0.0.1:53, which Busybox `ntpd` defaults to if it cannot read `/etc/resolv.conf`.

Hurricos, 2022-11-23:
So, how are we going to get jails to point their copy of `/etc/resolv.conf` to their bind-mounted `/tmp/resolv.conf.d` directory?

webknjaz, 2026-10-02:
For people landing here from search engines, the workaround for fixing ntp is disabling jail for it (until the underlying race condition is fixed):
```console
$ mv /etc/capabilities/ntpd.json /etc/capabilities/ntpd.json-disabled
$ service sysntpd restart
```
*(credit: https://forum.openwrt.org/t/when-will-sysntpd-sync-time/109382/6?u=webknjaz)*

---

Since I'm making my own ImageBuilder images, I've put the above into `/etc/uci-defaults/99-sysntpd-workaround` so that it runs on the first put upon flashing a new dumb AP.