openwrt/openwrt #10393: How does openwrt populate IPv6 default route without receiving router advertisements?

shotahino, 2022-08-05:
Apologies for the lack of more details. But it appears that openwrt can populate IPv6 default route for it without receiving router advertisements.  This is reported for IPv6 + PPPoE setup. 

openwrt appears to have IPv6 connectivity despite it not seeing RAs. In the packet capture, I see openwrt sent router solicitation 3 times (, which I believe is in accordance with RFC2461), but there are no RAs. So, I am wondering what mechanism openwrt uses to populate default router so it has IPv6 connectivity.

Note that the segment I am looking at is between upstream network and openwrt router. I am _not_ looking at downstream clients.

jow-, 2022-08-05:
Via PPPoE IPV6CP. Please don't use the issue tracker for support questions.

closed 2022-08-05 by jow-

shotahino, 2022-08-05:
@jow- Thanks. Posted a question in the openwrt forum - https://forum.openwrt.org/t/how-does-openwrt-set-ipv6-default-route-without-receiving-ras-from-upstream/133916