openwrt/openwrt #10402: opkg_install_cmd: Cannot install package busybox

atyouspace, 2022-08-07:
Hi all,

I am compiling openwrt 22.03.0 r6.

**"make menuconfig":
Base system  --->
<*> busybox
[*] Customize busybox options
      Networking Utilities  --->
      [*] ifup (14 kb)                                                          
  │ │ [*] ifdown (13 kb)                                                             
  │ │            ()    Absolute path to ifstate file (NEW)                                         
  │ │            [ ]   Use ip tool (else ifconfig/route is used) (NEW)                            
  │ │            [ ]   Support IPv4 (NEW)                                                          
  │ │            [ ]   Support IPv6 (NEW)                                                          
  │ │            [ ]   Enable mapping support (NEW)                                               
  │ │            [ ]   Support external DHCP clients (NEW)**

If I uncheck ifup and ifdown, the issues are still there.

Issues are following,
==
Collected errors:
 * check_data_file_clashes: Package busybox wants to install file /home/v22030-rc6/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/root-imx/sbin/ifdown
	But that file is already provided by package  * netifd
 * check_data_file_clashes: Package busybox wants to install file /home/v22030-rc6/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/root-imx/sbin/ifup
	But that file is already provided by package  * netifd
 * opkg_install_cmd: Cannot install package busybox.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/v22030-rc6/openwrt'
make[1]: *** [package/Makefile:111: /home/v22030-rc6/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/v22030-rc6/openwrt'
make: *** [/home/v22030-rc6/openwrt/include/toplevel.mk:230: world] Error 2
==

Thanks for all.

Best regards,
Tao

aparcar, 2022-08-07:
I assume netifd provides the files already, is there a Special reason you need the busybox versions?

jow-, 2022-08-09:
This build configuration is not supported.

closed 2022-08-09 by jow-