openwrt/openwrt #10405: SSH stops working on Belkin RT3200/Linksys E8450 with -rc6

richb-hanover, 2022-08-07:
_As reported in: https://forum.openwrt.org/t/ssh-stops-working-on-belkin-rt3200-linksys-e8450-with-rc6/133911_

I installed OpenWrt 22.03.0-rc6 on a Belkin RT3200 router. The Wi-Fi and traffic all seem fine, and I can SSH in. But...

Shortly after a reboot, the SSH sessions freeze. At that point, I cannot establish any new SSH logins. This occurs within a few minutes of a reboot, or perhaps as long as 20 minutes. This is repeatable. (This also happened with -rc4 - I updated to -rc6 before reporting the problem.)

A reboot clears up the SSH problem - I can log in as expected (for a while). In the meantime, even when the SSH process is frozen, the LuCI GUI works as expected, and the router passes traffic normally.

**Update:** My current test is to ssh into the router and run htop. The Uptime: value shows how long the SSH session runs before freezing which tends to be about 3-10 minutes.

**Update 2:** I have some evidence that the problem is being caused by being SSH'd in. The router runs fine without any SSH connections, but if I wait a while (hours) then SSH in, the SSH session will hang shortly thereafter.

What other troubleshooting information could I provide? Thanks.

aparcar, 2022-08-10:
@dangowrt ideas?

richb-hanover, 2022-08-10:
Oh dear. I forgot I posted this report here, too. I have lots more info at the OpenWrt forum posting at: https://forum.openwrt.org/t/ssh-over-wifi-stops-working-on-rt3200-e8450-with-22-03-0-rc6/133911/18

hnyman, 2022-08-10:
You might edit your report (and title) here to clearly specify that the bug seems to be wifi-only, and wired is ok for you. 
(based on forum discussion)

That suggests that wifi connections occasionally break in a way that breaks the TCP stream session for SSH, or something similar.

richb-hanover, 2022-08-10:
Good point. I fixed both the title and added the Wi-Fi vs Ethernet note to the original report. Thanks.

richb-hanover, 2022-08-11:
More data collected on the forum (see the OP): included here for completeness...

Tons of new evidence (no `tcpdump` yet):

- I'm still running RC1 on the Belkin RT3200
- I decided to try a couple more devices running `htop`, so I fired up my old MacBook Pro ("oMBP") and an old Win10 laptop. My primary machine is a newer MBP. All three were connected via Wi-Fi and running `htop` successfully.
- After ~1h 15 minutes, I got tired of waiting, so I stopped `htop` and exited the SSH sessions on oMPB and the Win10 machine .
- Within 5 minutes, `htop` froze on the third computer (MBP). I could not re-establish a new SSH connection. (oMBP and Win10 were still disconnected from SSH at that time.)
- While the new MBP was in that bad state, I **was** able to ssh in to the router and run `htop` on oMBP and Win10. After checking that `htop` worked, I stopped it and exited the SSH session on those machines.
- I turned off Wi-Fi on the new MBP, then turned it back on, and was immediately able to reconnect to the router.

My summary of the evidence:

- Something is interfering with SSH & Wi-Fi. Running `htop` over Wi-Fi freezes within 5-20 minutes, and that computer cannot re-connect to SSH.
- If multiple computers were connected via Wi-Fi and running `htop`, no freeze was observed (I waited 1h 15m, when a freeze normally occurs within 20 minutes)
- (I didn't try it in this round of experiments, but...) SSH over Ethernet seems always to work
- When one computer is in the "frozen-Wi-Fi" state, another computer can SSH in via Wi-Fi
- When I turned Wi-Fi off and back on for the affected computer, it could immediately SSH back in.

What's the next experiment? `tcpdump`? Thanks

dangowrt, 2022-08-11:
Sounds like a problem with connection tracking or more likely flow offloading which results in a stale DROP entry in either hardware or software tables. Starting with WED I would try to one by one disable offloading features and see if the problem persists.

richb-hanover, 2022-08-11:
Hi Daniel, Thanks for your thoughts. I am using a plain-vanilla config - just install the RC, and opkg install htop/nano - no other configs save the LAN subnet and the Wi-Fi credentials.

I am not sure how to disable offloading. Could you give me a quick rundown? Thanks again.

richb-hanover, 2022-08-11:
@aparcar 
> Is this related? https://github.com/openwrt/openwrt/pull/10422

Hmmm... Maybe. Has that fix been merged? Should I wait for the next nightly snapshot to see if it changes the behavior?

aparcar, 2022-08-11:
Not merged yet. @hauke do you think it could be relevant?

csharper2005, 2022-08-12:
Confirm this. Various mt7621 devices running 22.03-rc4, rc5.

richb-hanover, 2022-08-12:
Good news over on the Forum. https://forum.openwrt.org/t/ssh-over-wifi-stops-working-on-rt3200-e8450-with-22-03-0-rc6/133911/55 And it failed when I reverted the `/etc/config/wireless` to default. 

My next step is to re-flash RC6, then try jow's nftables fix (https://forum.openwrt.org/t/ssh-over-wifi-stops-working-on-rt3200-e8450-with-22-03-0-rc6/133911/38)

csharper2005, 2022-08-12:
> My next step is to re-flash RC6, then try jow's nftables fix

I got the issue even the firewall was disabled (dumb AP).

richb-hanover, 2022-08-13:
**Update:**  The `nft` commands also seem to have solved the problem (but I'm going to let the `htop` test run overnight). Here's a summary from the Forum: https://forum.openwrt.org/t/ssh-over-wifi-stops-working-on-rt3200-e8450-with-22-03-0-rc6/133911/62

hauke, 2022-08-24:
I do not see the problem on my Linksys E8450 running almost OpenWrt v22.03.0-rc6 (own build with additional packages from commit before tag) without using any mentioned workarounds. 
This is my production device so I prefer not to do debugging there. 
The device has a uptime of 24 days and I can connect to it over ssh using wifi without problems.
The SSH connection stays up for at least 10 minutes of doing nothing.
I can also access OpenWrt devices behind this E8450.

I am not using special offloading functionality. I have multiple SSIDs for multiple networks.

I have some questions:
# Could someone please provide a minimal configuration of a device where he sees this problem and describe how to run into it with OpenWrt 22.03.0-rc6 or later. Preferable without the need of a WAN connection. 
# Did someone run into this problem on devices other combinations than MT7622 + MT7915? 
# Did someone see this problem in OpenWrt master?
# Maybe the problem is related Mediatek WED offloading. 
## Could someone please try this patch on top of openwrt 22.03: https://github.com/hauke/openwrt/commit/e480bf26f668d4e774d8056e3016bda5792d8d0e

richb-hanover, 2022-08-24:
Hi Hauke,

I made the initial report. Here's the summary of the report from the Forum: https://forum.openwrt.org/t/ssh-over-wifi-stops-working-on-rt3200-e8450-with-22-03-0-rc6/133911/63

---
Summary of problem and workarounds

When connected to my 22.03.0-rc6 router via Wi-Fi, I would see an SSH connection freeze after a while (minutes to perhaps an hour). I would run htop, and the output from the router would freeze. Furthermore, I could no longer log into the router, even from another SSH session over Wi-FI.

I was surprised to see that the router would continue to pass traffic from my computer to the internet, and I could continue to use the LuCI web interface even when the SSH sessions were hung up. And SSH connections via Ethernet worked just fine(!)

The symptoms seem to be caused by a combination of a bug in the MT76 driver and DSCP markings ("af21") recently introduced by dropbear. This group has proposed workarounds (#1 and #2 below), both of which seem to address the problem, but the real fix seems to require the MT76 driver update.

Another workaround seems to be simply turning Wi-Fi off and back on on my MacBook Pro - that immediately allowed me to SSH back in.
---

Other observations:

- I could reproduce the bug with 22.03.0-rc6. After installation, I used LuCI to install only nano and htop. I also reproduced on RC1, and RC4, I think. I might have tried a snapshot somewhere along the line - same failure. I could not try 21.02 - it does not support this router.

- My test procedure was; 
	- Using Ethernet cable, install a fresh copy of OpenWrt, don't keep settings.
	- Use LuCI to install nano and htop. 
	- Use LuCI to enable Wi-Fi (I was using a 2.4GHz channel; 5GHz was disabled). 
	- Disconnect Ethernet, connect to the Wi-Fi
	- ssh into the router, start htop and note the uptime
	- Sometime later, I would come back to that terminal window and see that uptime was no longer counting up. The difference showed how long it ran.
	- The freeze would occur between 5 and 90 minutes
	- Even when the SSH session (over Wi-Fi) was frozen, Wi-Fi still worked: I could use the LuCI web GUI, and the router would pass traffic at full speed.
	- And a SSH connection over Ethernet worked fine if I turned off the laptop's Wi-Fi and connect the Ethernet

- This seemed to be related to the duration of my laptop connection (not uptime). I could connect, disconnect after a few minutes, then re-connect hours later without problem. But if I stayed connected (running htop) on Wi-Fi the uptime shown on the screen would indicate how long the connection lasted before freezing.

- When I first noticed the problem, the connection would fail in 5-15 minutes. Around the time I posted the summary, it had run for over an hour before the SSH session froze. I have no explanation for this.

- I only saw it on my MacBook Pro (10.15.7). I didn't really try to make it happen with a Win10 or Linux laptop.

- Workarounds:
	#1: https://forum.openwrt.org/t/ssh-over-wifi-stops-working-on-rt3200-e8450-with-22-03-0-rc6/133911/38
	#2: https://forum.openwrt.org/t/ssh-over-wifi-stops-working-on-rt3200-e8450-with-22-03-0-rc6/133911/49

Thanks for thinking about this. Let me know if you have other questions.

Rich

ynezz, 2022-08-26:
> I got the issue even the firewall was disabled (dumb AP).

@csharper2005 can you try following patch?

> 1.2. Could someone please try this patch on top of openwrt 22.03: https://github.com/hauke/openwrt/commit/e480bf26f668d4e774d8056e3016bda5792d8d0e

neheb, 2022-08-26:
the nft commands work but are not a long term solution.

This issue happens with mt7622 and mt7915. It does not happen with mt7621 and mt7915 DBDC.

The only difference I can see between both setups is Wireless Ethernet Dispatch, which is exclusive to mt7622. Maybe @nbd168 knows more.

csharper2005, 2022-08-26:
@ynezz @hauke patch https://github.com/hauke/openwrt/commit/e480bf26f668d4e774d8056e3016bda5792d8d0e doesn't help.

MTS WG420223 (mt7621, mt7615 dbdc). 22.03-HEAD.

nbd168, 2022-08-26:
Was this issue reproduced on the 5 GHz wifi as well?

richb-hanover, 2022-08-26:
> Was this issue reproduced on the 5 GHz wifi as well?

I have not tried the 5GHz channel. (I did not enable it at all...) I will enable a 5GHz SSID and report back.

csharper2005, 2022-08-26:
@nbd168 yeah, 2g and 5g both.

nbd168, 2022-08-26:
@csharper2005 but only on a device with mt7615 DBDC, right?
Based on what I've read so far, it seems to me that the issue is specific to the mt7615 driver (which also handles mt7622)

nbd168, 2022-08-26:
I managed to reproduce the issue on MT7615 by forcibly restarting aggregation on TID3 and quickly found the bug afterwards. Commit ec7d32f3769fbd815f72a7471e4bb7a07aee359d should fix it, please test.

richb-hanover, 2022-08-26:
> Commit https://github.com/openwrt/openwrt/commit/ec7d32f3769fbd815f72a7471e4bb7a07aee359d should fix it, please test.

For someone who has not yet learned to build the OpenWrt software, how can I get an image (a snapshot?) that contains this commit? Thanks.

csharper2005, 2022-08-26:
> but only on a device with mt7615 DBDC, right?

@nbd168, It seems to me that the problem also was reproduced on the device with mt7613, but I'm not sure. I can't test it right now. [WiFire S1500.NBN](https://openwrt.org/toh/wifire/s1500_nbn) with mt7602, mt7612 is not affected.

nbd168, 2022-08-26:
mt7613 is handled by the same driver as well. mt7602/mt7612 is a different one

nbd168, 2022-08-26:
@richb-hanover when you're not building images for yourself, just wait for the next snapshot build and flash it. I think your test should work fine if it was able to reproduce the issue before.

csharper2005, 2022-08-26:
> I managed to reproduce the issue on MT7615 by forcibly restarting aggregation on TID3 and quickly found the bug afterwards. Commit [ec7d32f](https://github.com/openwrt/openwrt/commit/ec7d32f3769fbd815f72a7471e4bb7a07aee359d) should fix it, please test.

@nbd168 that's a win! 22.03-HEAD with your commit is working without the issue.

nbd168, 2022-08-26:
great, thanks for testing!

closed 2022-08-26 by nbd168