openwrt/openwrt #10416: Large sets with "counter" enabled fail and break subsequent nft operations.

djadair, 2022-08-08:
Both 22.03-rc6 and master w/ 5.15 kernel fail when attempting to load tables containing large
block lists with counters.  The same list works fine with Ubuntu 5.17 kernel ( loads w/o errors, counts blocked traffic ).  Also if "counter" is removed from large sets file loads properly.  Loading sets one element at a time sometimes works but that is both really really slow and non-atomic.

- nft reports netlink / ENOBUFS errors. ( atomic update )
- kernel WARNING backtraces. ( randomly after failures ).
- nft_counter module refcount -1 ( sometimes after load, always for table delete w/ large set counters )

The refcount -1 is fatal and requires a reboot to get nft to work again.

Note:  Netlink appears to be a red herring.  Seems like nf_tables api throwing errors not the netlink layer.

Reproduction:
Using 22.03-rc6 x86-64 image file in vmware 2 core machine.
nft -f banIP.save.nft  >> fails.
remove "counter" from sets >> works.

Have also seen one case where chains stopped being processed despite showing up in nft list.  Flushing and re-creating table did not fix but deleting table and re-creating did.  Unfortunately can not reproduce that one but I think it may be related to errors following use of "dormant" flag.

The backtrace from main references "module_put" which is consistent with the refcount -1.  Not sure if the accounting error is cause or effect but it seems likely this patch is related to it not occurring 5.17 which was first version that removed nft_counter as a separate module:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/net/netfilter?h=v5.18.16&id=023223dfbfb34fcc9b7dd41e21fbf9a5d5237989

[banIP.save.nft.gz](https://github.com/openwrt/openwrt/files/9279287/banIP.save.nft.gz)

From 22.03-rc6:
```
[   78.868725] ------------[ cut here ]------------
[   78.869865] WARNING: CPU: 1 PID: 4187 at 0xffffffff81133891
[   78.873156] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack slhc r8169 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c igc forcedeth e1000e crc_ccitt bnx2 i2c_dev nfsv3 nfs nfs_ssc ixgbe e1000 amd_xgbe mdio oid_registry vfat fat lockd sunrpc grace dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 ena igb ramoops reed_solomon pstore button_hotplug tg3 ptp realtek pps_core mii
[   78.906278] CPU: 1 PID: 4187 Comm: nft Tainted: G        W         5.10.134 #0
[   78.908083] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 11/12/2020
[   78.910764] RIP: 0010:0xffffffff81133891
[   78.911789] Code: 31 c0 eb f3 0f 1f 44 00 00 48 85 ff 74 17 8b 87 b0 02 00 00 89 c2 83 ea 01 78 0f f0 0f b1 97 b0 02 00 00 75 ef c3 cc cc cc cc <0f> 0b eb f7 66 66 2e 0f 1f 84 00 00 00 00 00 55 48 89 e5 e8 c7 ff
[   78.916382] RSP: 0018:ffffc9000240f918 EFLAGS: 00010297
[   78.917735] RAX: 0000000000000000 RBX: ffffffffa02ba000 RCX: 0000000000000011
[   78.919558] RDX: 00000000ffffffff RSI: 0000000000000286 RDI: ffffffffa02ba0c0
[   78.921405] RBP: ffffc9000240f938 R08: 0000000000000000 R09: 00000000000001ec
[   78.923209] R10: fff0000000000fff R11: 00000000000002ba R12: ffff888005821000
[   78.924984] R13: ffff8880055349f0 R14: ffffffff82500ca0 R15: ffffffff824ffe80
[   78.927008] FS:  00007fd9f34e5b48(0000) GS:ffff88807be40000(0000) knlGS:0000000000000000
[   78.929129] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   78.930714] CR2: 00007ffe15e6a508 CR3: 000000000585a005 CR4: 00000000000606e0
[   78.932632] Call Trace:
[   78.933303]  ? 0xffffffffa028be01 [nf_tables@0000000001b1d63b+0x27000]
[   78.934978]  0xffffffffa0297b8e [nf_tables@0000000001b1d63b+0x27000]
[   78.936583]  0xffffffffa0297f21 [nf_tables@0000000001b1d63b+0x27000]
[   78.938131]  0xffffffffa0240abc [nfnetlink@00000000aa1f8fb0+0x3000]
[   78.939905]  ? 0xffffffff811f0ae0
[   78.940769]  ? 0xffffffff811d27d0
[   78.941640]  ? 0xffffffff813cca4b
[   78.942494]  ? 0xffffffff811e9209
[   78.943353]  0xffffffffa0240f88 [nfnetlink@00000000aa1f8fb0+0x3000]
[   78.944899]  0xffffffff818b8865
[   78.945795]  0xffffffff818b8b40
[   78.946622]  0xffffffff81831f82
[   78.947458]  ? 0xffffffff818b8930
[   78.948336]  ? 0xffffffff813b412b
[   78.949215]  0xffffffff81833734
[   78.950122]  ? 0xffffffff8120c70c
[   78.951007]  ? 0xffffffff813b7bf0
[   78.951861]  ? 0xffffffff8183821c
[   78.952716]  ? 0xffffffff81206cca
[   78.953571]  ? 0xffffffff811b2db2
[   78.954419]  ? 0xffffffff81a37aa9
[   78.955418]  ? 0xffffffff81838cda
[   78.956300]  ? 0xffffffff818392a2
[   78.957178]  0xffffffff81833840
[   78.958016]  0xffffffff81833898
[   78.958852]  0xffffffff81a2c5e5
[   78.959710]  0xffffffff81c000a9
[   78.960632] RIP: 0033:0x7fd9f34c436c
[   78.961594] Code: c3 8b 07 85 c0 75 24 49 89 fb 48 89 f0 48 89 d7 48 89 ce 4c 89 c2 4d 89 ca 4c 8b 44 24 08 4c 8b 4c 24 10 4c 89 5c 24 08 0f 05 <c3> e9 ff d0 ff ff 41 54 b8 02 00 00 00 49 89 f4 be 00 88 08 00 55
[   78.966219] RSP: 002b:00007ffe15e6a058 EFLAGS: 00000246 ORIG_RAX: 000000000000002e
[   78.968050] RAX: ffffffffffffffda RBX: 00007fd9f34e5b48 RCX: 00007fd9f34c436c
[   78.969860] RDX: 0000000000000000 RSI: 00007ffe15e6a0a8 RDI: 0000000000000003
[   78.971655] RBP: 0000000000000003 R08: 0000000000000000 R09: 0000000000000000
[   78.973427] R10: 0000000000000000 R11: 0000000000000246 R12: 000000000000002e
[   78.975362] R13: 00007ffe15e6a510 R14: 00007ffe15e7b560 R15: 00007ffe15e7b870
[   78.977222] ---[ end trace 5520b1270a0d6134 ]---
```

djadair, 2022-08-08:
Tried "netfilter: nf_tables: make counter support built-in" patch with main and confirmed that it does get rid of kernel warnings and refcount failure.  Thats better because subsequent nft commands without large sets work after the failing command but something else is causing the kernel allocations to fail (aka ENOBUFS) with openwrt while they work with Ubuntu.

jow-, 2022-08-09:
When you say it works with Ubuntu kernel but fails with an Ubuntu one, are you testing this on the same environment with the same amount of available RAM?

djadair, 2022-08-10:
Sorry I will edit to clarify a bit.  This is what I see using the same VM config ( generic linux 5.x, 2 cpus, 2G memory):
openwrt 22.03-rc6:  fails
openwrt: main from last week:  fails.
ubuntu jammy:  works.
same file w/o set counters:  works everywhere.

Note: Ubuntu is a bloated live image so it has a lot less free memory.

Here is a bit more detail on the working version.  I noticed that the majority of the memory used by counters was due to per_cpu with large #cpus.  ( 16 * min(NR_CPUS, nr_cpu_ids) * counters ).  In the working case that is the majority of the space used when it works (58MB).  Booting with nr_cpus=4 reduces memory use but does not result in working with openwrt.

Also worth noting that the percpu allocation failures appear to have warnings for the first 10 failures -- I never see any of those.  The extra use appears to trigger the problem but not directly cause it.

```
root@ubuntu:~# grep cpu /proc/meminfo
Percpu:           101376 kB
root@ubuntu:~# grep Mem /proc/meminfo
MemTotal:        1989528 kB
MemFree:           82372 kB
MemAvailable:     832124 kB
root@ubuntu:~# uname -a
Linux ubuntu 5.15.0-25-generic #25-Ubuntu SMP Wed Mar 30 15:54:22 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
root@ubuntu:~# nft -v
nftables v1.0.2 (Lester Gooch)
root@ubuntu:~# nft -f /home/ubuntu/banIP.save.nft 
root@ubuntu:~# grep cpu /proc/meminfo 
Percpu:           168960 kB
root@ubuntu:~# grep Mem /proc/meminfo
MemTotal:        1989528 kB
MemFree:          275608 kB
MemAvailable:     768220 kB

about 16K set counters ( 2/line )
root@ubuntu:~# nft list ruleset | wc -l
8849
root@ubuntu:~# nft -t list ruleset | wc -l
24

```

djadair, 2022-08-13:
Still no idea why working system is different but I have confirmed which path is failing.

The initial error is actually ENOMEM.  It gets converted to ENOBUFS after running through the whole batch.
```
Thread 1 hit Breakpoint 6, nfnetlink_rcv_batch (skb=0xffff888002ee24c0, skb@entry=0xffff888002ee27c0, nlh=0xffffc90001a74608, nlh@entry=0xffffc90001a47000, subsys_id=10, genid=genid@entry=0) at net/netfilter/nfnetlink.c:519
519				if (err == -EAGAIN) {
(gdb) p err
$20 = -12
```
Not *completly* sure but it appears the initial cause is the GFP_ATOMIC allocation in nft_counter_clone. I think the set element creation can happen in the filter path so atomic likely on purpose but in this case it is called right after a kzalloc with GFP_KERNEL in nft_set_elem_expr_clone.

```
Thread 1 hit Breakpoint 10, nft_counter_clone (dst=0xffff8880082a9770, src=<optimized out>) at net/netfilter/nft_counter.c:239
239		if (cpu_stats == NULL)
(gdb) p cpu_stats
$26 = (struct nft_counter *) 0x0 <fixed_percpu_data>
(gdb) bt
#0  nft_counter_clone (dst=0xffff8880082a9770, src=<optimized out>) at net/netfilter/nft_counter.c:239
#1  0xffffffffa028c21e in nft_expr_clone (dst=<optimized out>, src=0xffff88800460b4e0) at net/netfilter/nf_tables_api.c:2830
#2  0xffffffffa028fa71 in nft_set_elem_expr_setup (num_exprs=1, expr_array=0xffffc9000178f768, ext=0xffff8880082a9758, 
    ctx=0xffffc9000178f940) at net/netfilter/nf_tables_api.c:5498
#3  nft_add_set_elem (ctx=ctx@entry=0xffffc9000178f940, set=set@entry=0xffff888009290e00, attr=attr@entry=0xffffc90001ac4720, 
    nlmsg_flags=<optimized out>) at net/netfilter/nf_tables_api.c:5977
#4  0xffffffffa02903ce in nf_tables_newsetelem (skb=0xffff888002ee24c0, info=0xffffc9000178f9f8, nla=<optimized out>)
    at net/netfilter/nf_tables_api.c:6085
#5  0xffffffffa0219d14 in nfnetlink_rcv_batch (skb=0xffff888002ee24c0, skb@entry=0xffff888002ee27c0, nlh=0xffffc90001ac46e4, 
    nlh@entry=0xffffc90001a47000, subsys_id=10, genid=genid@entry=0) at net/netfilter/nfnetlink.c:513
#6  0xffffffffa021a128 in nfnetlink_rcv_skb_batch (nlh=<optimized out>, skb=0xffff888002ee27c0) at net/netfilter/nfnetlink.c:634
#7  nfnetlink_rcv (skb=0xffff888002ee27c0) at net/netfilter/nfnetlink.c:652
#8  0xffffffff818d637a in netlink_unicast_kernel (ssk=0xffff888005c45000, skb=0xffff888002ee27c0, sk=0xffff888004826800)
```

djadair, 2022-08-15:
I've convinced myself that Ubuntu is working by accident rather than a code differences.  It has:
- PREEMPT_VOLUNTARY enabled  and
- A huge amount of percpu memory allocated due to GUI and/or snaps cruft.
My sample set loads because it fits in holes but a large enough table would have to fail because the GFP_ATOMIC allocations require a failure to trigger allocation of a modest amount of new space.  With a large enough set the free space reclamation would kick in before it was possible to run nft enough times to have it succeed.  ( not to mention that takes forever and can trigger module refcount errors ).

The attached patch changes the netlink triggered allocations to GFP_KERNEL so the allocations are limited by free memory rather than PCPU_EMPTY_POP_PAGES_LOW.  I don't know enough however to be very comfortable with this, specifically:
- I don't know or have a test case for dynset which appears to need atomic allocations ??
- I don't know why the initial commits for cloning referenced flow tables.  Could not figure out how/why they are using clone or what their atomic requirements are.
- Seems like the default path should be GFP_KERNEL_ACCOUNT but we don't have the rest of those changes yet.
- Could not find the refcount error.  Patch hides it by not failing.  Create, update, delete work reliably.
[125-netfilter_clone_gfp.patch.txt](https://github.com/openwrt/openwrt/files/9336118/125-netfilter_clone_gfp.patch.txt)

djadair, 2022-08-18:
Ok so Openwrt unlikely to fix upstream bugs but for completeness I think this is the refcount leak:

nft_expr_clone  -- conditionally calls __module_get(src->ops->type->owner); on success.
nft_expr_destroy -- unconditionally calls module_put(type->owner);

The err < 0 path should just kfree.

```
int nft_set_elem_expr_clone(const struct nft_ctx *ctx, struct nft_set *set,
			    struct nft_expr *expr_array[])
{
	struct nft_expr *expr;
	int err, i, k;

	for (i = 0; i < set->num_exprs; i++) {
		expr = kzalloc(set->exprs[i]->ops->size, GFP_KERNEL_ACCOUNT);
		if (!expr)
			goto err_expr;

		err = nft_expr_clone(expr, set->exprs[i]);
		if (err < 0) {
			nft_expr_destroy(ctx, expr);
			goto err_expr;
		}
		expr_array[i] = expr;
	}

	return 0;

err_expr:
	for (k = i - 1; k >= 0; k--)
		nft_expr_destroy(ctx, expr_array[k]);

	return -ENOMEM;
}
```