openwrt/openwrt #10417: Bug in Unrar 6.10 beta 3

Kallek57, 2022-08-08:
There is a bug in UNRAR 6.10 beta 3 that is currently used in entware. It breaks direct unpack in sabnzbd. Can you please update Unrar to 6.10 stable. I was asked to post this here.

https://github.com/Entware/Entware/issues/828

BKPepe, 2022-08-08:
@Kallek57 You were pointed in the referenced issue to [the packages feed](https://github.com/openwrt/packages/blob/openwrt-21.02/utils/unrar/Makefile), where the unrar is located, but..

- OpenWrt 19.07 has unrar 5.7.3
- OpenWrt 21.02 has unrar 6.0.3
- OpenWrt 22.03 and OpenWrt master have unrar 6.1.3.

There is no unrar 6.10 beta3. You should ask Entware which OpenWrt version you are using and if they can update it in their fork.

Safihre, 2022-08-08:
This is the strange labeling way of unrar (Rarlabs), 6.1.3 is their way of indicating 6.10 Beta 3. They just keep increasing the "patch" version for their Linux releases. 
For example, the stable 6.11 version is released as:
https://www.rarlab.com/rar/unrarsrc-6.1.7.tar.gz

ynezz, 2022-08-10:
This package is provided by packages feed, so should be handled there.

closed 2022-08-10 by ynezz