openwrt/openwrt #10428: error when building the base toolchain for ATH79

artynet, 2022-08-09:
Hello folks,

any ideas about this strange error about the toolchain build step for **ATH79** architecture on stable tag **v22.03.0-rc6**. The build system is running on a freshly installed **Debian 11 bullseye** Linux distribution. Of course, all the prerequisites have been satisfied as in the following guide:

https://openwrt.org/docs/guide-developer/toolchain/install-buildsystem#debianubuntu

the errors seems to take place in the **initial** gcc compiler build step.

```
/mnt/volume_fra1_01/openwrt-stm32mp/build_dir/toolchain-mips_24kc_gcc-11.2.0_musl/gcc-11.2.0/gcc/c/c-typeck.c
/bin/bash: line 2: 2884785 Killed                  build/genautomata /mnt/volume_fra1_01/openwrt-stm32mp/build_dir/toolchain-mips_24kc_gcc-11.2.0_musl/gcc-11.2.0/gcc/common.md /mnt/volume_fra1_01/openwrt-stm32mp/build_dir/toolchain-mips_24kc_gcc-11.2.0_musl/gcc-11.2.0/gcc/config/mips/mips.md insn-conditions.md > tmp-automata.c
make[4]: *** [Makefile:2459: s-automata] Error 137
make[4]: *** Waiting for unfinished jobs....
rm gcc.pod
make[4]: Leaving directory '/mnt/volume_fra1_01/openwrt-stm32mp/build_dir/toolchain-mips_24kc_gcc-11.2.0_musl/gcc-11.2.0-initial/gcc'
make[3]: *** [Makefile:4416: all-gcc] Error 2
make[3]: Leaving directory '/mnt/volume_fra1_01/openwrt-stm32mp/build_dir/toolchain-mips_24kc_gcc-11.2.0_musl/gcc-11.2.0-initial'
make[2]: *** [Makefile:37: /mnt/volume_fra1_01/openwrt-stm32mp/build_dir/toolchain-mips_24kc_gcc-11.2.0_musl/gcc-11.2.0-initial/.built] Error 2
make[2]: Leaving directory '/mnt/volume_fra1_01/openwrt-stm32mp/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#191.53#40.25#250.48
    ERROR: toolchain/gcc/initial failed to build.
make[1]: *** [toolchain/Makefile:97: toolchain/gcc/initial/compile] Error 1
make[1]: Leaving directory '/mnt/volume_fra1_01/openwrt-stm32mp'
make: *** [/mnt/volume_fra1_01/openwrt-stm32mp/include/toplevel.mk:230: toolchain/install] Error 2
```
any ideas to solve it ?

Thanks in advance

hnyman, 2022-08-09:
the actual error might be somewhat earlier, especially if that is a parallel build.

Or is that log from "-j 1" , no parallel ???

jow-, 2022-08-09:
> /bin/bash: line 2: 2884785 Killed                  build/genautomata 

This particular process step of the gcc build requires a lot of memory. Most likely you're running out of ram or swap on your build system.

closed 2022-08-09 by jow-