openwrt/openwrt #10442: Base packages build errors when compile with '-O2' flag

alexeys85, 2022-08-13:
Where are base packages which always compile with `-Os` flag. I tried to build them with `-O2` and got errors due to `-Werror` flag. Few examples.
1. rpcd:
```
/media/enzo/Data/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl_eabi/rpcd-2022-05-31-1c482573/sys.c:244:25: error: 'strncpy' specified bound 128 equals destination size [-Werror=stringop-truncation]
  244 |                         strncpy(ver, p1, sizeof(ver));
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
2. ubox
```
/media/enzo/Data/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl_eabi/ubox-2022-07-17-46a33b8b/kmodloader.c:752:20: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
  752 |         options[0] = 0;
      |         ~~~~~~~~~~~^~~
/media/enzo/Data/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl_eabi/ubox-2022-07-17-46a33b8b/kmodloader.c:745:19: note: destination object of size 0 allocated by 'malloc'
  745 |         options = malloc(len);
      |                   ^~~~~~~~~~~
```
3. fstools
```
/media/enzo/Data/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl_eabi/fstools-2022-07-17-ebf7e902/blockd.c:118:26: error: '%s' directive argument is null [-Werror=format-overflow=]
  118 |                 ULOG_ERR("failed to spawn %s %s %s\n", *argv, action, device);
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/media/enzo/Data/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl_eabi/fstools-2022-07-17-ebf7e902/blockd.c: In function 'blockd_startup':
/media/enzo/Data/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl_eabi/fstools-2022-07-17-ebf7e902/blockd.c:118:49: note: format string is defined here
  118 |                 ULOG_ERR("failed to spawn %s %s %s\n", *argv, action, device);
      |                                                 ^~
```
4. May be some other packages, as seems most of them from [here](https://git.openwrt.org/?a=project_list;pf=project) has `-Os`.

P.S. To override `-Os` I add
```
define Build/Configure
	find $(PKG_BUILD_DIR)/ -type f -name "CMakeLists.txt" -exec sed -i 's/-Os/-O2/g' {} +
	$(call Build/Configure/Default)
endef
```
to Makefiles, probably there is some cleaner way?

ynezz, 2022-08-24:
> 5. rpcd

I've looked at it and I can confirm this and indeed, it's quite interesting, that gcc reports this only with -O2/-O3 flags, but not with -Os flag. BTW clang-15/scan-build-15 doesn't report any such issues.

alexeys85, 2022-08-24:
`Reported-by: Alexey Smirnov <s.alexey@gmail.com>`