openwrt/openwrt #10450: arm-trusted-firmware: fails to compile with glibc (bl31.elf has a LOAD segment with RWX permissions)

PolynomialDivision, 2022-08-14:
Arm-trusted firmware fails using glibc with `bl31.elf has a LOAD segment with RWX permissions`.
```
Image Type:   MediaTek BootROM Loadable Image
Boot Media: Serial NAND
Page Size:    2KB
Spare Size:   64B
Load Address: 00201000
Architecture: ARM64
Building mt7622
  OD      /home/nick/openwrt/build_dir/target-aarch64_cortex-a53_glibc/arm-trusted-firmware-mediatek-mt7622-snand-1ddr/arm-trusted-firmware-mediatek-2021-05-08-d2c75b21/build/mt7622/release/bl2/bl2.dump
  LD      /home/nick/openwrt/build_dir/target-aarch64_cortex-a53_glibc/arm-trusted-firmware-mediatek-mt7622-snand-1ddr/arm-trusted-firmware-mediatek-2021-05-08-d2c75b21/build/mt7622/release/bl31/bl31.elf
aarch64-openwrt-linux-gnu-ld.bfd: warning: /home/nick/openwrt/build_dir/target-aarch64_cortex-a53_glibc/arm-trusted-firmware-mediatek-mt7622-snand-1ddr/arm-trusted-firmware-mediatek-2021-05-08-d2c75b21/build/mt7622/release/bl31/bl31.elf has a LOAD segment with RWX permissions
make[4]: *** [Makefile:1147: /home/nick/openwrt/build_dir/target-aarch64_cortex-a53_glibc/arm-trusted-firmware-mediatek-mt7622-snand-1ddr/arm-trusted-firmware-mediatek-2021-05-08-d2c75b21/build/mt7622/release/bl31/bl31.elf] Error 1
make[4]: Leaving directory '/home/nick/openwrt/build_dir/target-aarch64_cortex-a53_glibc/arm-trusted-firmware-mediatek-mt7622-snand-1ddr/arm-trusted-firmware-mediatek-2021-05-08-d2c75b21'
make[3]: *** [Makefile:103: /home/nick/openwrt/build_dir/target-aarch64_cortex-a53_glibc/arm-trusted-firmware-mediatek-mt7622-snand-1ddr/arm-trusted-firmware-mediatek-2021-05-08-d2c75b21/.built] Error 2
make[3]: Leaving directory '/home/nick/openwrt/package/boot/arm-trusted-firmware-mediatek'
time: package/boot/arm-trusted-firmware-mediatek/mt7622-snand-1ddr/compile#0.47#0.16#0.68
    ERROR: package/boot/arm-trusted-firmware-mediatek failed to build (build variant: mt7622-snand-1ddr).
make[2]: *** [package/Makefile:116: package/boot/arm-trusted-firmware-mediatek/compile] Error 1
make[2]: Leaving directory '/home/nick/openwrt'
make[1]: *** [package/Makefile:110: /home/nick/openwrt/staging_dir/target-aarch64_cortex-a53_glibc/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/nick/openwrt'
make: *** [/home/nick/openwrt/include/toplevel.mk:230: world] Error 2
```

ping @dangowrt

dangowrt, 2022-08-15:
Please try picking the two commits below into your tree and see if the problem persists:
https://git.openwrt.org/?p=openwrt/staging/dangole.git;a=commit;h=ec6271910f56319f796116877e5d0d97dd55a07b
https://git.openwrt.org/?p=openwrt/staging/dangole.git;a=commit;h=bcb927f551109ee20800e95420adfc26ba43561f

PolynomialDivision, 2022-08-25:
I just compiled everything again from a fresh git clone and now it works with following settings:
```
./scripts/diffconfig.sh 
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_mt7622=y
CONFIG_TARGET_mediatek_mt7622_DEVICE_linksys_e8450-ubi=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_LIBC="glibc"
CONFIG_LIBC_USE_GLIBC=y
# CONFIG_LIBC_USE_MUSL is not set
CONFIG_PACKAGE_librt=y
CONFIG_STRIP_ARGS="--strip-all"
CONFIG_TARGET_SUFFIX="gnu"
CONFIG_USE_GLIBC=y
CONFIG_USE_STRIP=y
```

So I will close. Sorry.

closed 2022-08-25 by PolynomialDivision