openwrt/openwrt #10452: binutils 2.39 package fails to compile

hnyman, 2022-08-14:
cc @polynomialdivision @hauke

The recently upgraded binutils 2.39 package fails to compile for several targets.

There are already several reports in the upgrade commit itself.
https://github.com/openwrt/openwrt/commit/cc24c4ed5eae66c061e5787bb172fd2a91861f25#comments

I wonder if the new package 2.39 also requires binutils 2.39 in the toolchain?  
(there the default is still an older version 2.37)

Also buildbot notices the build failure.
E.g.
https://downloads.openwrt.org/snapshots/faillogs/aarch64_generic/base/binutils/compile.txt
```
In file included from dispatcher.c:35:
/builder/shared-workdir/build/sdk/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/sys/signal.h:1:2: warning: #warning redirecting incorrect #include <sys/signal.h> to <signal.h> [-Wcpp]
    1 | #warning redirecting incorrect #include <sys/signal.h> to <signal.h>
      |  ^~~~~~~
In file included from dispatcher.c:30:
./../src/collector_module.h:78:13: error: duplicate member 'pwrite'
   78 |   ssize_t (*pwrite64)();
      |             ^~~~~~~~
dispatcher.c: In function 'collector_timer_create':
dispatcher.c:578:8: error: 'struct sigevent' has no member named '_sigev_un'
  578 |   sigev._sigev_un._tid = __collector_gettid ();
      |        ^
dispatcher.c: In function 'init_interposition_intf':
dispatcher.c:690:27: warning: implicit declaration of function 'dlvsym'; did you mean 'dlsym'? [-Wimplicit-function-declaration]
  690 |   __real_pthread_create = dlvsym (dlflag, "pthread_create", SYS_PTHREAD_CREATE_VERSION);
      |                           ^~~~~~
      |                           dlsym
dispatcher.c:690:25: warning: assignment to 'void *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
  690 |   __real_pthread_create = dlvsym (dlflag, "pthread_create", SYS_PTHREAD_CREATE_VERSION);
      |                         ^
make[9]: *** [Makefile:639: libgp_collector_la-dispatcher.lo] Error 1
```

Ps. also the iproute2 upgrade from the same author seems to fails for the same targets. Not sure if that is related.
```
make[4]: Entering directory '/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/iproute2-ipfull/iproute2-5.19.0'
Usage: configure [OPTIONS]
	--include_dir <dir>		Path to iproute2 include dir
	--libdir <dir>			Path to iproute2 lib dir
	--libbpf_dir <dir>		Path to libbpf DESTDIR
	--libbpf_force <on|off>		Enable/disable libbpf by force. Available options:
					  on: require link against libbpf, quit config if no libbpf support
					  off: disable libbpf probing
	--prefix <dir>			Path prefix of the lib files to install
	-h | --help			Show this usage info
make[4]: *** No rule to make target 'config.mk', needed by 'all'.  Stop.
make[4]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/iproute2-ipfull/iproute2-5.19.0'
```

PolynomialDivision, 2022-08-14:
> (Reverting the 2.39 enabling commit enables the mt7622/E8450 build to succeed again.)

But the commit only enables it. Weird. The default binutils should be still 2.37
https://github.com/openwrt/openwrt/blob/8456d06aa678c649d178053a5734d0d4c40a058c/toolchain/binutils/Config.in#L5

PolynomialDivision, 2022-08-14:
> (Reverting the 2.39 enabling commit enables the mt7622/E8450 build to succeed again.)

@hnyman Are you speaking about the [toolchain commit](https://github.com/openwrt/openwrt/commit/81ecce27c740604c0d9470be2270b71f22db5351) or the [package update](https://github.com/openwrt/openwrt/commit/cc24c4ed5eae66c061e5787bb172fd2a91861f25) commit?

hnyman, 2022-08-14:
I meant that reverting the 2.39 bump in the devel packages enables the compilation to succeed (with 2.38).
The toolchain 2.39 enabling commit does not need to be touched.

PolynomialDivision, 2022-08-14:
Okay, the error is introduced here:
https://github.com/bminor/binutils-gdb/commit/bb368aad297fe3ad40cf397e6fc85aa471429a28

We should just disable this module `--disable-gprofng` (like done [here](https://lore.kernel.org/all/20220807222613.1173589-1-raj.khem@gmail.com/) and it should compile again.

hauke, 2022-08-14:
binutils update was reverted in a9eb5cce7e5c4996fa6ffc081b5323bfd4735dc2.

closed 2022-08-14 by hauke