openwrt/openwrt #10459: X550 cannot autoneg on 2.5Gbps

Shinwar, 2022-08-16:


hauke, 2022-08-16:
OpenWrt does not apply any patches on ixgbe, I would suggest to report this regression to the upstream kernel maintainers, this is probably a general problem in the upstream kernel. They can fix it in the Linux stable tree and then OpenWrt will get it automatically.

abhishek-rawal, 2022-08-23:
@Shinwar : I haven't looked into the commits between 5.4 and 5.10 src of ixgbe upstream code. However, the [readme.md of ixgbe](https://downloadmirror.intel.com/738735/readme.txt) says : `By default, devices based on the Intel(R) Ethernet Controller x550 do not advertise 2.5 Gbps or 5 Gbps.` You might have to advertise it : `# ethtool -s <ifname> advertise 0x800000000000`

Shinwar, 2022-08-26:
> @Shinwar : I haven't looked into the commits between 5.4 and 5.10 src of ixgbe upstream code. However, the [readme.md of ixgbe](https://downloadmirror.intel.com/738735/readme.txt) says : `By [...]

Tried this command and X550 can autoneg to 2.5Gbps again, Thanks
According to https://github.com/torvalds/linux/commit/a296d665eae1e8ec6445683bfb999c884058426a ixgbe of kernel 5.10  introduced suppression of the advertisement of NBASE-T speeds by default. Now X550 in linux works as stupid as it worked in windows

closed 2022-08-28 by Shinwar

shenlijun, 2023-01-31:
https://github.com/shenlijun/openwrt-x550-nbase-t