openwrt/openwrt #10460: Unify 6 LR - LED not working

John-H-Smith, 2022-08-16:
While booting the AP, the LED is flashing. When it's booted up, the LED is out even if configured to be always on.

riptidewave93, 2022-08-19:
Which firmware are you using? There is a build for the older devices that had an RGB LED, and newer units that only have a blue/white LED.

https://github.com/openwrt/openwrt/blob/31d86a1a119265393db02aa66e6bc6518ee7b905/target/linux/mediatek/dts/mt7622-ubnt-unifi-6-lr-v1.dtsi
https://github.com/openwrt/openwrt/blob/31d86a1a119265393db02aa66e6bc6518ee7b905/target/linux/mediatek/dts/mt7622-ubnt-unifi-6-lr-v2.dtsi

John-H-Smith, 2022-08-19:
> Which firmware are you using? There is a build for the older devices that had an RGB LED, and newer units that only have a blue/white LED. [...]

Thanks for the links!
I'm running on OpenWrt 21.02.3 r16554-1d4dea6d4f.
How can I test if my hardware is a newer or a older one?

John-H-Smith, 2022-09-06:
Update: I just updated the firmware.
Now running on OpenWrt 22.03.0 r19685-512e76967f / LuCI openwrt-22.03 branch git-22.245.77528-487e58a

It would be okay that only a white/blue led is on, but currently the led is not lighting at all, even if its set as followed:
[image]

John-H-Smith, 2022-09-12:
Another update: I tried installing the _kmod-leds-ubnt-ledbar_ package, but that doesn't change anything on the behavior. LED is still always off.

joed74, 2024-11-24:
I'm running OpenWrt 23.05.5 r24106-10cc5fcd00 on a Unifi 6 LR v1 and get the following messages if i want to change colors:

[205836.528305] ubnt-ledbar 0-0030: Error initializing LED transaction: a8
[205836.535083] leds blue: Setting an LED's brightness failed (-22)