openwrt/openwrt #10461: GL-iNet gl-mt1300 (Beryl) flash corruption

mpannen1979, 2022-08-16:
I have been attempting to install regular OpenWRT on my gl-mt1300 for quite a while.  I've tried varius recent releases/snapshots/rc but I am having hardware problems.

I was able to get initial install that seemed to work, but after making minor changes to config such as wifi, etc to get basic networking working with my other network devices, it would seem that after a reboot, things fall apart.

I could ping LAN interface, but web interface no longer works, wifi broken, etc.

I was able to SSH to the unit and dmesg output had errors such as "jffs corrupt".  Sorry, I do not have and copy paste of that output. 

More recently, I have been trying other firmwares that seem to be quite close to gl-mt1300, such as:

mediatek_ap-mt7621a-v60
mediatek_mt7621-eval-board

And most recently:

oraybox_x3a

The first two devices don't have wifi, so that 'kind of worked'.  The oraybox_x3a actually did work with wifi, etc.  All 3 of these choices never gave me issues with "jffs corruption" messages, and I could make changes, reboot, etc, and things would come back to life.

So, I decided to look close at what the differences might be between the 'oraybox x3a' and the 'gl-mt-1300' that could reasonably cause a problem with flash storage.  Specifically, I have taken a closer look at the ".dts" files that look like they define flash storage, comparing these two files for differences.  These files can be found at the source path in the master:

target/linux/ramips/dts/
mt7621_glinet_gl-mt1300.dts
mt7621_oraybox_x3a.dts

A few differences that I could infer would be that the glinet has an spi0 section like this:

-------------------------------------------------------------------------
&spi0 {
	status = "okay";

	flash@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <80000000>;
		m25p,fast-read;
		broken-flash-reset;

----------------------------------------------------------------------------
And the oraybox has a section like this:

-----------------------------------------------------------------------------
&spi0 {
	status = "okay";

	flash@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <30000000>;
------------------------------------------------------------------

The oraybox uses a lower "spi-max-frequency", does not use "fast-read", and whatever "broken-flash-reset" is.

I followed an online guide for OpenWRT custom build, and built an install .bin for my mt1300 using 21.02.3 OpenWRT code.  I manually changed the mt7621_glinet_gl-mt1300.dts file so the top of the spi0 section looks like the one from the oraybox--I commented out the "m25p,fast-read;" the "broken-flash-reset;", and changed "spi-max-frequency" down to 30000000.

I flashed this modified firmware, and I am fully able to make changes, save changes, reboot, etc and no longer seeing things break, no more "jffs corrupt" kernel messages with my gl-mt1300.

I am curious if this "corruption" has been observed by anyone else?

mpannen1979, 2022-08-16:
Also, I would like to add, I've not had any problems with the original "GL-iNet" firmware like this, I just don't want to use their candied up firmware, and I would like something newer.  

I also looked at GL-iNet's own code repository for the dts file:

https://github.com/gl-inet/openwrt/blob/openwrt-19.07.8/target/linux/ramips/dts/GL-MT1300.dts

Any their dts file for spi0 looks like this:
-----------------------------------------------------------------
&spi0 {
	status = "okay";

	m25p80@0 {
		compatible = "mx25l25635f", "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <10000000>;
-------------------------------------------------------------------

This also leads me to belive that one or more of the 3 changes I made above could be a reasonable fix for the flash issues I was having.

mpannen1979, 2022-08-16:
Leaving the "broken-flash-reset" seems to be OK .

I suspect it is the spi-max-frequency being too high?

Trying this right now: (leaving the 

-----------------------------------------------------------

&spi0 {
status = "okay";

flash@0 {
	compatible = "jedec,spi-nor";
	reg = <0>;
	spi-max-frequency = <30000000>;
	broken-flash-reset;

----------------------------------------------------------

Linaro1985, 2022-08-17:
Hi @mpannen1979. 

Try spi-max-frequency = <25000000>;
Do not change other parameters.

mpannen1979, 2022-08-17:
OK got it.

I've made a new binary for openwrt 22.03.0-rc6 for this device with JUST spi-max-frequency lowered from 80MHz to 25MHz.

So far it is working without issues, no jffs corruption messages.  I've installed some packages, reboots, etc, and still has sign of life.

I have confirmed that the SPI max is set to 25MHz:

--------------------------------------------------------------
root@OpenWrt-Garage:/sys/firmware/devicetree/base/palmbus@1e000000/spi@b00/flash@0# ls -l
-r--r--r--    1 root     root             0 Aug 17 14:45 broken-flash-reset
-r--r--r--    1 root     root            14 Aug 17 14:45 compatible
-r--r--r--    1 root     root             0 Aug 17 14:45 m25p,fast-read
-r--r--r--    1 root     root             6 Aug 17 14:45 name
drwxr-xr-x    6 root     root             0 Aug 17 14:45 partitions
-r--r--r--    1 root     root             4 Aug 17 14:45 reg
-r--r--r--    1 root     root             4 Aug 17 14:45 spi-max-frequency
root@OpenWrt-Garage:/sys/firmware/devicetree/base/palmbus@1e000000/spi@b00/flash@0# cat spi-max-frequency | hexdump -
0000000 7d01 4078
0000004

-------------------------------------------------------------------------

mpannen1979, 2022-09-07:
This slight tweak and all has been working well for me 22+ days.

gmertes, 2022-09-09:
Been running 22.03 on my MT1300 for 5 days with the default SPI frequency (80 MHz) - no corruption so far. Installed some packages and rebooted a few times, everything is running smooth and flash memory is persisting. I'll continue monitoring to see if things fall apart in the future 🤞. Perhaps bad luck with the silicon? 

I did see the following error on the first boot after flashing: `mount_root: failed to sync jffs2 overlay` , the same issue as documented [here](https://forum.openwrt.org/t/solved-getting-error-failed-to-sync-jffs2-overlay/47742). However, that error went away after a reboot so I'm not sure if it's related. And if I understand it correctly the conclusion in that forum thread is that it's not a breaking error. 

The jffs messages in my log are all normal now:

```
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.588636] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    8.561234] jffs2: notice: (574) jffs2_build_xattr_subsystem: complete building xattr subsystem, 26 of xdatum (19 unchecked, 7 orphan) and 33 of xref (7 dead, 0 orphan) found.
[    8.582474] mount_root: switching to jffs2 overlay
```

mpannen1979, 2022-09-10:
Maybe different devices have different flash chips?  I'm not sure if I want to put the effort into opening this unit up to look physically.  I bought it new maybe 2 months ago.  I can live with it.  :)

mpannen1979, 2022-09-10:
I would accept that it could be a off grade flash chip on my unit.  I am glad I could find a workaround.

learnzalot, 2022-10-21:
I've just purchased a gl-mt1300 and I'm having similar issues to those described in this thread. I'm wondering if it's possible for me to get a copy of the new binary that you created for openwrt 22.03.0-rc6 for this device, please?

mlyle, 2022-10-29:
I'm in the same category: flaky jffs2 on gl-mt1300.  Can we please consider turning down the speed in devicetree?

mlyle, 2022-10-29:
@mpannen1979 would you consider reopening this given that others are experiencing the same problem and would benefit from downclocking SPI?

mlyle, 2022-10-30:
@learnzalot https://drive.google.com/file/d/1ZiO9eowfpqXgGPuQEZ973yssPYsg2j_c/view?usp=sharing is 22.03.2 with downclocked SPI.

mpannen1979, 2022-10-30:
Sorry I don't have the firmware saved.  I don't have the knowledge to submit a patch, it would be great if someone could do it though.

learnzalot, 2022-10-30:
> @learnzalot https://drive.google.com/file/d/1ZiO9eowfpqXgGPuQEZ973yssPYsg2j_c/view?usp=sharing is 22.03.2 with downclocked SPI.

Thank you, I will check it out.

learnzalot, 2022-10-30:
> > @learnzalot https://drive.google.com/file/d/1ZiO9eowfpqXgGPuQEZ973yssPYsg2j_c/view?usp=sharing is 22.03.2 with downclocked SPI.
> 
> Thank you, I will check it out.

Loaded it up and so far all seems to  be working fine. The problems with wifi have disappeared. Thanks again!

mpannen1979, 2022-10-30:
I've made a new master firmware with those line changes for the 50MHz  SPI, no fast-read, for my unit and it is running now just fine as expected.

mlyle, 2022-10-30:
I am also at 50MHz with the code in the pull request and it is working well for me.

learnzalot, 2022-10-31:
https://drive.google.com/file/d/1FmipYapkGzggyN66E4NodFg1-qgaALPe/view?usp=sharing 

...... works just fine for me too! Thank you.

Linaro1985, 2022-10-31:
> no fast-read

It's better to leave fast-read if you don't want to get a race-condition when updating the firmware in the future.

mpannen1979, 2022-10-31:
My experimentation a few months ago, the only change I made was to lower the SPI frequency.  I left the fast-read opcode as it was, FYI.

SPI at 50Mhz works for me, the 80MHz was too high.

Here is another file with just SPI lowered to 50MHz:

https://drive.google.com/file/d/1mk0bvgzxy-oL-KGK0MXG35LLQyidm94w/view?usp=share_link

mpannen1979, 2022-11-01:
My unit has a "Winbond 25Q256JVFQ" flash chip.  On the PCB board is "V1.1".

The back sticker has:

Device ID: rsc15ba
FCC ID: 2AFIW-MT1300
IC: 23019-MT1300

mlyle, 2022-11-01:
@mpannen1979 Which in principle supports fast read at a high clock rate.  But apparently the board doesn't manage this on our devices (or the chips are fake, etc.)

closed 2022-11-07 by openwrt-bot

Zaelnorth, 2023-02-17:
Greetings, thanks a lot for all your efforts, total noob here.

I got my MT1300 yesterday and first of all installed the stock OpenWRT from the [official repo](https://openwrt.org/toh/gl.inet/gl-mt1300_v1), unfortunately the radio0 had a really weak signal, I saw online that this was a problem back in '21.

Then I came to this bug and thought perhaps that's what's causing the signal instability. I downloaded the bins on this page and did sysupgrade, unfortunately still no change.

Not sure if it's related to the bug here or to another one elsewhere, so apologies if I'm asking in the wrong place. 

I'd be very grateful if anyone could point me in the right direction, thank you!

gmertes, 2023-02-17:
@Zaelnorth I think it's unlikely a weak signal is related to this fix. radio0 is 2.4GHz, radio1 is 5GHz. Did you configure it like that? The webinterface allows you to select 5GHz for radio0, which would cause a weak/unusable signal.

Here are the radio options in my `/etc/config/wireless` in case it helps:

```
config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option txpower '20'
        option cell_density '0'
        option htmode 'HT20'
        option band '2g'
        option channel '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0+1'
        option band '5g'
        option htmode 'VHT80'
        option txpower '20'
        option cell_density '0'
        option channel '40'
```

Zaelnorth, 2023-02-18:
@gmertes thanks for your comments, will check it out and let you know, appreciate it!

commit 961e01fc (2022-11-07): ramips: gl-mt1300: downclock SPI to 50MHz

The SPI max frequency was set to 80MHz, considerably higher than the
vendor clocks it in their firmware (10MHz).  Multiple users reported
jffs2 corruption/instability in GitHub issue #10461.

My unit has a W25Q256; datasheet specifies maximum SPI frequency for
read command of 50MHz.

Thanks to @DragonBlueP for suggesting to eliminate m25p,fast-read;
and @MPannen1979 for identifying the problem.

Fixes: #10461
commit 9155d405 (2022-11-12): ramips: gl-mt1300: downclock SPI to 50MHz

The SPI max frequency was set to 80MHz, considerably higher than the
vendor clocks it in their firmware (10MHz).  Multiple users reported
jffs2 corruption/instability in GitHub issue #10461.

My unit has a W25Q256; datasheet specifies maximum SPI frequency for
read command of 50MHz.

Thanks to @DragonBlueP for suggesting to eliminate m25p,fast-read;
and @MPannen1979 for identifying the problem.

Fixes: #10461

(cherry picked from commit 961e01fc67e7d9e60557df3474fa326216aa4839)