openwrt/openwrt #10471: Completely drop CONFIG_IPV6

f00b4r0, 2022-08-18:
A [patch](http://patchwork.ozlabs.org/project/openwrt/patch/20220816170047.50642-1-hacks@slashdirt.org/) has been submitted to "hide" `CONFIG_IPV6`, the disabling of which is no longer supported (it breaks build).

However, as has been pointed out [before](http://lists.openwrt.org/pipermail/openwrt-devel/2022-March/038249.html) and [now](http://lists.openwrt.org/pipermail/openwrt-devel/2022-August/039273.html), the correct approach is to entirely remove `CONFIG_IPV6` from the openwrt trees.

This requires a coordinated change across the buildroot and the packages repositories (and maybe third-party repositories as well).

This issue is opened to serve as a reminder of what still needs to be done.

HTH

ddmesh, 2023-01-06:
Copied from closed issue #9580 , because it is not solved:

I have to reopen this issue, because **droping "IPv4 only" support is absolutely not acceptable!**
We have a project (Freifunk) that depends on disabling IPv6 in firmware. 

First we do not need this and secondly there is no space in flash. Kernel and all its modules take a lot of space
which does not fit into flash memory of routers with 8mbyte or lower.
In addition we need to implement unused security/network/iptables for IPv6. This is against any idea
behind openwrt. It must be an open platform and it must be possible to have IPv4 only systems as long
Linux supports this and has this included.

The project uses a common code base, so small and new routers can be supported including other non-router 
devices.
When you just drop IPv6 because you just don't want to fix this issue, this will cause huge problems, not just
for Freifunk. 

Openwrt 22.03.0-rc4 is the last version that does work. all other candidates or releases have this blocking issue.
* What has changed since then?

I really appreciate if you can fix this issue. Unless we have an unsolvable problem with our project which is
doomed to be end. A lot of devices are not available in shops and new devices are build. All those can not be 
used in future. 

I really hope that such profound decisions will be reverted and will never happens again . 
There were already a lot of massive changes during upgrades from openwrt 19 to 21 without any upgrade posibility (except factory reset) in an environment, where routers are in field and not accessible physically. 
We could solve this upgrade problem by reverting some patches in openwrt.

Droping IPv4-Only support is not possible.

Stephan (Freifunk)

pkgadd, 2023-01-06:
Given the long standing issues with `CONFIG_IPV6`, this is long overdue, thank you for taking this up.

f00b4r0, 2023-01-06:
> I have to reopen this issue, because **droping "IPv4 only" support is absolutely not acceptable!** We have a project (Freifunk) that depends on disabling IPv6 in firmware.

That's your prerogative. I'm not sure why it should bear particular relevance to the project as a whole?

> First we do not need this

I'm sorry, who's "we"?

> and secondly there is no space in flash. Kernel and all its modules take a lot of space which does not fit into flash memory of routers with 8mbyte or lower. In addition we need to implement unused [...]

And you absolutely can disable IPv6 at the interface level or system-wide through sysctl. Your argument about firewalling is moot.

> The project uses a common code base, so small and new routers can be supported including other non-router devices. When you just drop IPv6 because you just don't want to fix this issue, this will cause huge problems, not just for Freifunk.

We didn't "drop IPv6", in fact it's quite the opposite.
Besides, this is **2023**. IPv6 should have replaced IPv4 a long time ago, so your argument seems rather backwards.

> Openwrt 22.03.0-rc4 is the last version that does work. all other candidates or releases have this blocking issue.
> 
> * What has changed since then?

Feel free to bisect and submit a patch if you can dedicate the resources to fix this.

> I really appreciate if you can fix this issue.

That's not how this [open source] works.

> Unless we have an unsolvable problem with our project which is doomed to be end. A lot of devices are not available in shops and new devices are build. All those can not be used in future.

They certainly can, although not easily. The kernel has already grown so big that very old devices with very small flash can no longer even store the kernel alone. Are you suggesting we should stop updating the kernel so that these ancient devices be supported forever, and that thus newer devices should never be supported?

Besides, 21.02 is still supported, it also caters to this type of use case. And if you can afford to run your own security updates, you can use 19.07 or even older releases.

> I really hope that such profound decisions will be reverted and will never happens again . There were already a lot of massive changes during upgrades from openwrt 19 to 21 without any upgrade [...]

It very much is: see, it _happened_.

ddmesh, 2023-01-06:
Sorry, but you haven't understood my problem. I can not disable IPv6 on router, when there is no IPv6 build-in.
And this is a project requirement. Openwrt is a base for many kind of projects hat it must not break 
a basic functionality. 
You say you do not drop IPv6, but I said you will drop "IPv4 only". THis is when you remove any posibility to disable
IPv6 during compile time.
 And beside of this who "my" project is, it shouldn't matter at all. But for your information some different
freifunk communities are not a small project.

I already spent weeks to investigate the problem, but as you know openwrt is very complex. To find the
specific commit is almost not possible. Only the develper that has changed the sources of kernel modules or
is responsible for updating and including those in openwrt environment could find the issue easier.

You speak about contributing to open source and providing patches for this issue. But when you 
enforce the use of IPv6, I can not disable it any more. Where does a patch go then? 

Openwrt.org has a huge responsibility and is not just a experimental project. 
So I hope you can be more productive to find a solution instead droping IPv6 and tell me I can disable it at runtime
where no IPv6 is present in kernel.

dwfreed, 2023-01-06:
This issue is not about *dropping* IPv6.  This issue is about dropping the ability to disable IPv6 *only at compile time*.  Which means IPv6 *capability* would be built into the kernel, and then one can easily use sysctls to disable IPv6 during runtime.

f00b4r0, 2023-01-06:
> Sorry, but you haven't understood my problem.

I did: it's indeed _your_ problem.
You however do not understand OpenWrt's rationale for doing this.

kthxbye.

ddmesh, 2023-01-06:
@dwfreed 
> This issue is not about **dropping** IPv6. This issue is about dropping the ability to disable IPv6 **only at compile time**. Which means IPv6 **capability** would be built into the kernel, and then one can easily use sysctls to disable IPv6 during runtime.

The freifunk project has to disable IPv6 at compile time because of the already mentioned flash memory restrictions and common code base to be compatible.  So it is simply not possible to keep IPv6 in kernel and later deactivate it in runtime.

champtar, 2023-01-07:
What's the size difference with / without IPv6 ?

jow-, 2023-01-07:
You can still disable IPv6 kernel support through kernel_menuconfig and you can still omit packages not needed for ipv6 functionality (odhcpd, odhcpc, ...).

What will be gone after this change is the ability to compile certain packages without support for IPv6. Many packages don't honor CONFIG_IPV6 anyway and will contain support for dealing with AF_INET6 regardless of the option choice. Some of the packages which do respect the option fail to compile with it being turned off and some will yield usable binaries that simply do not understand AF_INET6.

What is proposed here is the removal of the buildroot level (not kernel!) CONFIG_IPv6 option which is not regarded at all by many packages, broken for others and ineffective for some (e.g. musl libc's `inet_pton()` or `inet_aton()` will still parse and produce IPv6 addresses, regardless of the option being turned off).

ddmesh, 2023-01-07:
@champtar Unfortunately I can not check it because the images are not built because the file system exceeds. 
The freifunk project needs some tools like wireguard,fastd,bmxd. This is the basic functionality. Already because
all those kernel updates (without ipv6 enabled) are too big so I had to change bmxd code completely to get
free space. jffs2 needs at least three free flash sectors that size may differ from device to device.

@jow- 

Thanks for your constructive information. I have no insights in how much code/binary size increases when IPv6
is enabled in openwrt  permanently. When openwrt uses IPv6 and IPv6 code gets enabled in  kernel, there will always 
some kernel modules for IPv6 that are also built and included into the binary. 
I can check, if I can disable those. But in my experience there are some dependencies that can't be disabled. 
the hotplug will then also load and perhaps require those kmods.

The freifunk project already has excluded all tools/libs that are IPv6 (dhcp,iptables,routing...) because of no flash memory and the addtional tools needed for freifunk firmware to cover the functionality of the network.

I can imagine that lots of code does not respect IPv6 and already include this functionality.
Do you have some more information what gets  changed when IPv6 at openwrt level keeps aktivated?
I mean what will be included in the image that can't be disabled?
When I activate IPv6 a lot of dependencies follow and require underlaying support of IPv6, else it would not
work. But I think that this means that also a lot of kernel/module IPv6 dependencies are included which I can't
disable.

When IPv6 is enabled and such dependency chains getting created, then the openwrt team has a lot of work to do,
to ensure that all not needed tools/libs/kmods can still be deactivated. Or openwrt team must ensure, when starting
from basic working openwrt firmware configuration with no optional tools/libs/modules, that any additional selection
of components will only pull the minimal needed dependencies.

It is a difficult task to accomplish all this.

AiyionPrime, 2023-01-07:
Maybe one can take a deep breather, especially if one talks about what *the freifunk project needs*.

If this discussion got really just started because you want OpenWrt developers to support this toggle even longer then they already did, take note that it's an ever increasing challenge, as well as one with ever diminishing returns.
It's not a question about **if**, but **when** this toggle does not make sense anymore.

If you want this toggle to stay, don't tell others why you need this, but show them there's no need to remove it by either continuously fixing the codebase around packages that are not going to respect a IPv4 only domain anymore (a task other developers have done for you so far) or motivate other developers to do so.

But let me tell you: nudging developers by "the freifunk project needs this" is not only pretty impolite and for obvious reasons wrong as other communities have shown for a long time already, but does depict a project and its contributors I really appreciate as ungrateful, demanding and tilting against windmills on others cost.

As such I'd really appreciate if you'd dim the tone to a somewhat more constructive one.
Possibly starting with what asking for documentation on how to help supporting the toggle a little longer, while being aware of, that it will be an annoying task and won't end without giving this toggle up.

I really appreciate you standing up for Freifunk, but imho this is not the right way to do so.

If you want to discuss tiny devices and their Freifunk-Support these years, feel free to join us in [Freifunks Forum topic](https://forum.freifunk.net/t/umgang-mit-tiny-geraeten-in-2022/).

But this place should stay clean of loaded discussions for subprojects requirements.

Opinions are as always mine and neither the views of Freifunk Hannover nor Gluon.

ddmesh, 2023-01-08:
ok, this is not the kind of support an help I expected. I see no one is willing to fix issues, just workaround them. If your only help is, that I have
to fix the issue myself (because you can't) or find people that like to help...
Why have you not told me in your first answer that? 
I"m out.

Spudz76, 2023-01-08:
stop whining or we'll put in systemd and make it non-negotiable too

f00b4r0, 2023-01-08:
> Le 8 janv. 2023 à 01:17, Stephan:
> 
> ﻿
> ok, this is not the kind of support an help I expected. I see no one is willing to fix issues,
> 

There are no “issues”, this is by design. IPv6 is futureproof, IPv4 is not. FWIW the EU (which should concern your alleged “needs”) is pushing forward for massive IPv6 deployment. In that context, asking for a way to entirely remove IPv6 support seems rather backwards.

4/32 devices are no longer supported either. If this is what you’re using, you’re on your own. 16/128 is the current minimum recommendation.

If you want to keep using ancient outdated technologies because of whatever “needs” of yours, so be it: use ancient outdated software as well; but do not expect everyone else to follow suit with your absurd _demands_.

> just workaround them. If your only help is, that I have
> to fix the issue myself (because you can't) or find people that like to help...
> 

Again, that’s not how this works and you have the wrong attitude here.

> Why have you not told me in your first answer that?
> 

We did, apparently you did not want to listen.