openwrt/openwrt #10472: uqmi: hangs on --get-pin-status

dmarion, 2022-08-18:
When Quectel RM510Q-GL 5G modem is used, uqmi hangs on --get-device-status.

That results in qmi.sh waiting forever.

```
21376 root      1452 S    {qmi.sh} /bin/sh ./qmi.sh qmi setup a1 {"proto":"qmi","device":"/dev/cdc-wdm0","delegate":false,"apn":"internet","auth":"none"
21396 root       996 S    uqmi -s -d /dev/cdc-wdm0 --get-pin-status
```

This looks like race condition, as after killing uqmi,  next invocation works fine:

```
root@OpenWrt:~# kill 21396
root@OpenWrt:~# uqmi -s -d /dev/cdc-wdm0 --get-pin-status
"Not supported"
```

```
root@OpenWrt:~# grep VERSION=  /etc/os-release
VERSION="22.03.0-rc6"
```

CodeFetch, 2022-08-18:
This was an issue I ran into about 4 years ago. I think there is a probably "resolved" bug report somewhere.

dmarion, 2022-08-19:
> This was an issue I ran into about 4 years ago. I think there is a probably "resolved" bug report somewhere.

It doesn't look like resolved. Shouldn't uqmi timeout if card is not responding instead of waiting forever....

Vavooon, 2022-08-27:
Looks like this issue. It also happens to me with SIM7600G-H LTE modem.
https://forum.openwrt.org/t/solved-lte-qmi-troubles/28379/23

j123b567, 2023-01-23:
`uqmi` has parameter `-t` to set response timeout but there are two problems
 - timeout parameter is not used in `qmi.sh` and the default value is infinite
 - timeout parameter does not work in some cases. It seems that it terminates the inner thread in `uqmi` but the whole executable hangs indefinitely

I have created wrapper script replacing `uqmi` which calls original `uqmi` with `timeout` command, like
`timeout -s KILL $timeout /sbin/uqmi-raw` which finally resolves this for me. Clean solution would be to fix `uqmi` to properly timeout and to have reasonable default timeout value.

Vavooon, 2023-04-26:
Since `uqmi` might randomly freeze while connecting to SIM7600, I've created a more universal script:

`/usr/bin/uqmi` (it will have higher priority than original `uqmi` in OpenWRT installations if you place it here )
```
#!/bin/sh
# uqmi-watchdog: the wrapper is used to kill and restart frozen `uqmi` processes

output=$( mktemp )
pid=0

while true; do
	if [ $pid -eq 0 ]; then
		/sbin/uqmi "$@" > $output &
		pid=$!
		attempts=0
	fi

	if kill -0 $pid > /dev/null 2>&1; then
		if [ $attempts -ge 5 ]; then
			# kill $pid > /dev/null 2>&1
			killall /sbin/uqmi 2>&1
			pid=0
			logger -t uqmi-watchdog -p daemon.warn "UQMI was killed"
		fi
	else
		break
	fi

	attempts=$((attempts + 1))

	sleep 1
	# better to use `usleep`, enable it in 'Base system' > 'Busybox'
	# usleep 50000
done

wait $pid
exitCode=$?
result="$( cat $output )"

echo "$result"
exit $exitCode
```

As you can see from code, it runs original `uqmi` in the background and kills it once it's frozen (if process is running more than 250ms: that's for `usleep`), probably better to limit attempts count just to `1` if you use it with regular `sleep`.

dr-ni, 2023-10-28:
see
https://github.com/openwrt/openwrt/pull/13324
and now
https://github.com/openwrt/openwrt/pull/13781

zhangyoufu, 2024-05-13:
I run into the same issue on Quectel RM520N-GL.
Upon looking into this issue, I found that uqmi cannot handle two requests simultaneously. And that's what happens behind the scene when you enabled both IPv4 and IPv6. `qmi.sh` was invoked twice.

```
root@OpenWrt:~# uqmi -d /dev/cdc-wdm0 --uim-get-sim-state
{
	"card_application_state": "ready",
	"pin1_status": "disabled",
	"pin1_verify_tries": 3,
	"pin1_unlock_tries": 10,
	"pin2_status": "not_verified",
	"pin2_verify_tries": 3,
	"pin2_unlock_tries": 10
}
root@OpenWrt:~# uqmi -d /dev/cdc-wdm0 --uim-get-sim-state & uqmi -d /dev/cdc-wdm0 --uim-get-sim-state &
root@OpenWrt:~# {
	"card_application_state": "ready",
	"pin1_status": "disabled",
	"pin1_verify_tries": 3,
	"pin1_unlock_tries": 10,
	"pin2_status": "not_verified",
	"pin2_verify_tries": 3,
	"pin2_unlock_tries": 10
}

[2]+  Done                       uqmi -d /dev/cdc-wdm0 --uim-get-sim-state
root@OpenWrt:~# jobs
[1]+  Running                    uqmi -d /dev/cdc-wdm0 --uim-get-sim-state
```

dr-ni, 2024-05-13:
Is everything working if you deactivate ipv6?

zhangyoufu, 2024-05-13:
> Is everything working if you deactivate ipv6?

Nope, I checked that disabling IPv6 does not resolve the issue. `uqmi` still could hang somewhere.

I'm using a modified version of Vavooon's /usr/bin/uqmi wrapper, and I'm happy with the reliability right now.

wireless-road, 2024-10-19:
Facing same with `SIM7600E-L1C`.

and `get-pin-status` which looks unsupported with this modem:

```
# uqmi -s -d /dev/cdc-wdm0 --get-pin-status
"Not supported"
```

but from time to time it hangs indefinitely.
Using `-t 2000`  helped me too.

zaventh, 2025-03-16:
This is still an issue with many `uqmi` commands in `v24.10.0`, e.g.:

```
# uqmi -d /dev/cdc-wdm0 --get-pin-status -t2000 -s
Request timed out
"Unknown error"
^C
```

Time `timeout` flag allows the error to be returned, but it does not exit the process itself. This occurs on many commands as well, not just `get-pin-status`.

MakiseKurisu, 2025-03-21:
I'm also affected by this issue, and without timeout it seems to queue further calls, that I keep getting 4 `uqmi` processes running even when I killed some. I'm using `kill $(ps | grep qmi | cut -d ' ' -f 1 | xargs)` for now.