openwrt/openwrt #10485: python3-cryptography 36.0.2 update

Hi-hass, 2022-08-21:
How can I add a new version?  
I want to keep the old version and name it cryptography34, and then create a new cryptography36

Hi-hass, 2022-08-21:
I want to install aiohomekit, it has requirement cryptography==36.0.2 and numpy.

lucize, 2022-08-22:
you need rust that is not in sdk yet..
you can use this
https://github.com/lmahmutov/SGW-Homeassistant

firedevel, 2022-08-26:
I tried last time and I can't remember the mistakes. In short, build the homeassistant package difficult to succeed

lucize, 2022-08-26:
literally you have to follow the very short instruction described on the page, I guess you are not used to build yourself the image ?

firedevel, 2022-08-26:
Last time it took me 4 hours and 20GB of space. I'm too lazy to compile again :(