openwrt/openwrt #10489: TP-link TL-WR902AC v3 - MediaTek MT7610E 802.11nac Wifi client wont authenticate

darkxst, 2022-08-21:
I have been testing 22.03rc6 and having trouble with the second Wifi radio which is setup in client mode failing to authenticate. Reverting back to 21.02 and every works fine.

radio1 is  MediaTek MT7610E

logs:
```
[   51.552713] wlan1: authenticate with ac:29:a8:e6:86:81
[   51.706648] wlan1: send auth to ac:29:a8:e6:86:81 (try 1/3)
[   52.179474] wlan1: authenticate with ac:29:a8:e6:86:81
[   52.184747] wlan1: send auth to ac:29:a8:e6:86:81 (try 1/3)
[   52.515736] wlan1: authenticated
[   52.526291] wlan1: associate with ac:29:a8:e6:86:81 (try 1/3)
[   53.205980] wlan1: associate with ac:29:a8:e6:86:81 (try 2/3)
[   53.315923] wlan1: associate with ac:29:a8:e6:86:81 (try 3/3)
[   53.645894] wlan1: association with ac:29:a8:e6:86:81 timed out
[   64.541605] wlan1: authenticate with ac:29:a8:e6:7e:a9
[   64.693471] wlan1: send auth to ac:29:a8:e6:7e:a9 (try 1/3)
[   66.317393] wlan1: authenticate with ac:29:a8:e6:7e:a9
[   66.322637] wlan1: send auth to ac:29:a8:e6:7e:a9 (try 1/3)
[   66.334110] wlan1: authenticated
[   66.345995] wlan1: associate with ac:29:a8:e6:7e:a9 (try 1/3)
[   66.354566] wlan1: RX AssocResp from ac:29:a8:e6:7e:a9 (capab=0x1011 status=0 aid=3)
[   66.362724] wlan1: associated
[   67.217222] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   67.224843] wlan1: Limiting TX power to 26 (26 - 0) dBm as advertised by ac:29:a8:e6:7e:a9
[   67.232967] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1-1: link becomes ready
[   67.239912] br-lan: port 2(wlan1-1) entered blocking state
[   67.245484] br-lan: port 2(wlan1-1) entered forwarding state

```

wireless config:
```wireless.radio1.type='mac80211'
wireless.radio1.path='pci0000:00/0000:00:00.0/0000:01:00.0'
wireless.radio1.channel='36'
wireless.radio1.band='5g'
wireless.radio1.htmode='VHT80'
wireless.radio1.cell_density='0'
wireless.radio1.country='AU'
wireless.radio1.txpower='12'
wireless.wifinet2=wifi-iface
wireless.wifinet2.device='radio1'
wireless.wifinet2.mode='sta'
wireless.wifinet2.network='wwan'
wireless.wifinet2.ssid='ssid'
wireless.wifinet2.key='thekey'
wireless.wifinet2.encryption='sae-mixed'
```

darkxst, 2022-08-21:
seems to be the same as issue #10243

closed 2022-08-21 by darkxst

AbdulRahmannnnnn, 2023-12-22:
Did you get any solution regarding this problem? because I still experienced the same problem when using the new version 23.05.2