openwrt/openwrt #10496: boot loop zyxel p-2812 with OpenWrt 22.03.0-rc6

mbunal, 2022-08-22:
i am using 21.02 already. i want try new 22.03 with my zyxel p-2812. but it stuck at boot. i have no screen shot. but i recover my device with tftpboot openwrt-lantiq-xrx200-P2812HNUF1-uImage-initramfs and mafkatel firmware. after that release image.

what is wrong?

Mijzelf, 2022-08-22:
This one?
https://forum.openwrt.org/t/zyxel-p2812hnu-f1-u-boot/100281

mbunal, 2022-08-23:
maybe, but i have this:
ROM VER: 1.0.5
CFG 06
NAND
NAND Read OK

U-Boot SPL 2013.10-openwrt5 (Nov 18 2014 - 19:54:01)
SPL: initializing NAND flash
SPL: checking U-Boot image
SPL: loading U-Boot to RAM
SPL: decompressing U-Boot with LZO
SPL: jumping to U-Boot

U-Boot 2013.10-openwrt5 (Jul 06 2021 - 10:04:01) P-2812HNU-Fx

Board: ZyXEL P-2812HNU-Fx
SoC:   Lantiq VRX288 v1.1
CPU:   500 MHz
IO:    250 MHz
BUS:   250 MHz
BOOT:  NAND
DRAM:  128 MiB
NAND:  128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   ltq-eth
Hit any key to stop autoboot:  2  0 

It does not continue beyond this point. reboot again and again . i will try and inform u. thanks...

mbunal, 2022-08-23:
this is from my 2812:
ROM VER: 1.0.5
CFG 06
NAND
NAND Read OK

U-Boot SPL 2013.10-openwrt5 (Nov 18 2014 - 19:54:01)
SPL: initializing NAND flash
SPL: checking U-Boot image
SPL: loading U-Boot to RAM
SPL: decompressing U-Boot with LZO
SPL: jumping to U-Boot

U-Boot 2013.10-openwrt5 (Nov 18 2014 - 19:54:01) P-2812HNU-Fx

Board: ZyXEL P-2812HNU-Fx
SoC:   Lantiq VRX288 v1.1
CPU:   500 MHz
IO:    250 MHz
BUS:   250 MHz
BOOT:  NAND
DRAM:  128 MiB
NAND:  128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   ltq-eth
Hit any key to stop autoboot:  0

i think i have your mod.

mbunal, 2022-08-23:
my problem is:

U-Boot 2013.10-openwrt5 (Nov 18 2014 - 19:54:01) P-2812HNU-Fx

Board: ZyXEL P-2812HNU-Fx
SoC:   Lantiq VRX288 v1.1
CPU:   500 MHz
IO:    250 MHz
BUS:   250 MHz
BOOT:  NAND
DRAM:  128 MiB
NAND:  128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   ltq-eth
Hit any key to stop autoboot:  0
P-2812HNU-Fx # tftpboot openwrt-22.03.0-rc6-lantiq-xrx200-zyxel_p-2812hnu-f1-initramfs-kernel.bin
ltq_phy: addr 0, link 1, speed 1000, duplex 1
ltq_phy: addr 1, link 0, speed 10, duplex 0
ltq_phy: addr 17, link 0, speed 10, duplex 0
ltq_phy: addr 19, link 0, speed 10, duplex 0
ltq_phy: addr 5, link 0, speed 10, duplex 0
Using ltq-eth device
TFTP from server 192.168.1.2; our IP address is 192.168.1.1
Filename 'openwrt-22.03.0-rc6-lantiq-xrx200-zyxel_p-2812hnu-f1-initramfs-kernel.bin'.
Load address: 0x81000000
Loading: #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #######################################
         2.8 MiB/s
done
Bytes transferred = 6288300 (5ff3ac hex)
P-2812HNU-Fx # bootm 0x81000000
## Booting kernel from Legacy Image at 81000000 ...
   Image Name:   MIPS OpenWrt Linux-5.10.134
   Created:      2022-07-31  15:12:47 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    6288236 Bytes = 6 MiB
   Load Address: 80002000
   Entry Point:  80002000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... LZMA: uncompress or overwrite error 7 - must RESET b

Mijzelf, 2022-08-23:
`Uncompressing Kernel Image ... LZMA: uncompress or overwrite error 7 - must RESET b`

That one is typical for the unpatched u-boot. The uncompressed kernel (+initramfs) doesn't fit in the reserved 8MiB. 

*i think i have your mod.*
Nope.
`U-Boot 2013.10-openwrt5 (Nov 18 2014 - 19:54:01) P-2812HNU-Fx`
vs.
`U-Boot 2013.10-openwrt5 (Jul 06 2021 - 10:04:01) P-2812HNU-Fx`

mbunal, 2022-08-23:
date was different because compile time different. but version is same 2013.10-openwrt5. anyway i will try and inform u.

Mijzelf, 2022-08-25:
There is only one version of u-boot available for this box, as provided by ScApi, in 2015. OpenWrt never provided these. I patched ScApi's binary on two places with a hex-editor to change the buffer size from 8 to 16MiB, and also changed the timestamp (human readable text inside the binary) to be able so see if it's the patched one.

mbunal, 2022-08-27:
Do I have to start over to flash the uboot image? i mean Step 1: Type in console

nand erase.chip ....

mbunal, 2022-08-27:
u were right. with your patched uboot it was booted.

Mijzelf, 2022-08-28:
In that case you can close the issue?

closed 2022-08-28 by mbunal