openwrt/openwrt #10530: mpc85xx: PCIe addresses change from kernel 5.10.135 -> 5.10.138

Hurricos, 2022-08-28:
Noticed after an upgrade that the sysfs paths referred to in `/sys/class/net/*` had changed from 5.10.135 to 5.10.138:

```txt
root@ap3825i4:~# uname -a
Linux ap3825i4 5.10.135 #0 SMP Sat Aug 6 19:32:35 2022 ppc GNU/Linux
root@ap3825i4:~# ls -atlr /sys/class/net/labbmesh5*
lrwxrwxrwx    1 root     root             0 Aug 25 17:35 /sys/class/net/labbmesh5 -> ../../devices/platform/ffe0a000.pcie/pcia000:02/a000:02:00.0/a000:03:00.0/net/labbmesh5
```

```txt
root@ap3825i2:~# uname -a
Linux ap3825i2 5.10.138 #0 SMP Sat Aug 27 13:24:58 2022 ppc GNU/Linux
root@ap3825i2:~# ls -atlr /sys/class/net/labbmesh5*
lrwxrwxrwx    1 root     root             0 Aug 27 09:54 /sys/class/net/labbmesh5 -> ../../devices/platform/ffe0a000.pcie/pci0001:02/0001:02:00.0/0001:03:00.0/net/labbmesh5
```

 -- this caused netifd to be unable to find the original radio config:

```txt
Sat Aug 27 09:49:27 2022 daemon.notice netifd: Wireless device 'radio0' set retry=0
Sat Aug 27 09:49:27 2022 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Sat Aug 27 09:49:27 2022 daemon.notice netifd: radio1 (8547): WARNING: Variable 'data' does not exist or is not an array/object
Sat Aug 27 09:49:27 2022 daemon.notice netifd: radio1 (8547): Bug: PHY is undefined for device 'radio1'
Sat Aug 27 09:49:27 2022 daemon.notice netifd: Wireless device 'radio1' is now down
Sat Aug 27 09:49:27 2022 daemon.notice netifd: radio0 (8549): WARNING: Variable 'data' does not exist or is not an array/object
Sat Aug 27 09:49:27 2022 daemon.notice netifd: radio0 (8549): Bug: PHY is undefined for device 'radio0'
Sat Aug 27 09:49:27 2022 daemon.notice netifd: Wireless device 'radio0' is now down
```

.... and caused the preinit system to generate new definitions in `/etc/config/wireless`.

I'm cloning down linux-stable to find the ""regression"" in 5.10.135 to 5.10.138, but the real takeaway is that these sysfs paths aren't always stable, meaning that using them as an anchor might not be the best idea, at least not without some sort of recovery mechanism -- e.g.:

1) "If I am running a different kernel version since last sysupgrade;"
2) "and I recorded a list of netdev types (i.e. vendor / device IDs or similar) from before this most recent sysupgrade,"
3) "and some of the old netifs are at different paths than we last recorded;"
4) "but there are new matching netdevs at different paths;"
5) "then rather than creating new `wifi_device`s let's migrate the old `wifi_device`s to the new, corresponding paths."

Hurricos, 2022-08-28:
FWIW: This impacts pretty much every P1020E platform; tested and confirmed on an Aerohive AP330.

Hurricos, 2022-08-28:
I *think* the change must have been one of @pali's, but I don't want to bisect to find it -- the long and short is that the paths change and we just have to deal with it. It just wasn't something I expected.

BKPepe, 2022-08-28:
> I don't want to bisect to find it

It's related to https://lore.kernel.org/linuxppc-dev/20220706102148.5060-1-pali@kernel.org/t/#u , but we need to backport a few other patches or there is the possibility that it gets backported to LTS kernels.

- https://lore.kernel.org/linuxppc-dev/20220706104308.5390-1-pali@kernel.org/t/#u

- https://lore.kernel.org/linuxppc-dev/20220706101043.4867-1-pali@kernel.org/

And then enable ``CONFIG_PPC_PCI_BUS_NUM_DOMAIN_DEPENDENT``

For ``mpc85xx``, I am using LTS kernel 5.15 (out of the tree as it is not available within OpenWrt), and I can help you there with backporting it.

pali, 2022-08-28:
I guess it is this commit:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=0fe1e96fef0a5c53b4c0d1500d356f3906000f81

Check if you have pci aliases in your device tree or linux,pci-domain or ibm,opal-phbid properties, which do static assignment of PCI domains to PCI(e) controllers.

Originally I sent more patches which makes all those settings configurable by compile time config options, but those patches were not accepted. Hence device tree assignment is always in use.

Anyway, order of PCI devices in `/proc/bus/pci/devices` should be still same as before. And also output of `lspci -A linux-proc` should be same because 32-bit powerpc code does not fully distinguish between domains.

More PCI cleanup and domain patches are on ML to improve and stabilize this situation and make same as on other modern systems (e.g. arm or x86):
https://lore.kernel.org/linuxppc-dev/20220706104308.5390-1-pali@kernel.org/
https://lore.kernel.org/linuxppc-dev/20220817163927.24453-1-pali@kernel.org/

hauke, 2022-08-28:
We probably need something similar to 0bd5aa89fcf20dc5db073d925d5a08e779ae6292

Hurricos, 2022-08-28:
> Check if you have pci aliases in your device tree or linux,pci-domain or ibm,opal-phbid properties, which do static assignment of PCI domains to PCI(e) controllers.

Looks like there are:

```txt
root@ap3825i4:/sys/firmware/devicetree# find $PWD | grep pci | grep -i alias
/sys/firmware/devicetree/base/aliases/pci0
/sys/firmware/devicetree/base/aliases/pci1
```

Thankfully the alias IDs are taken second, so in theory we could use the `linux,pci-domain` property in our device trees to "lock" the addressing (I have not yet tested this).

Hurricos, 2022-08-30:
`linux,pci-domain` works perfectly.  I have pushed [mpc85xx-lock-pci-domain-pr](https://github.com/Hurricos/openwrt/compare/mpc85xx-lock-pci-domain-pr) to apply this to all mpc85xx boards; I'm going to reach out to see if I can find any users of the WDR4900-v1 to test.

@BKPepe: I'm confused. The two patches you list, are you saying that if applied they'd revert the PCI addressing back to how it used to be?

pali, 2022-08-30:
The question is: why you are setting alias with number in DTS which you do then not want to use? If you want to choose PCI domain randomly (like it was before), just drop or adjust alias. With all those next patches, PCI domain assignment would be same as on other platforms and would mimic v4.4 behavior.

Current behavior is to choose PCI domain pseudo-randomly, that is why you see those "big" numbers like 0xa000.

Anyway, the main issue here is that you want to depend on pseudo-random PCI domain assignment which does not have to be stable. Also the issue is that you want to depend on DTS/sysfs paths which are not declared to be stable too.

What should be stable when `CONFIG_PPC_PCI_BUS_NUM_DOMAIN_DEPENDENT` is not enabled is addresses of PCI devices as listed by command `lspci -A linux-proc`.

pali, 2022-08-31:
> This is the OpenWrt standard for tracking and enumerating wireless devices.

So it is broken and it should be fixed then.

> From my perspective, the sysfs path has never changed (which tells me the PCI domain has always been stable).

But it is not marked as stable and in comment https://github.com/openwrt/openwrt/issues/10530#issuecomment-1229449833 it was written that it happens time to time.

Hurricos, 2022-08-31:
OK, I understand this better now.

So, ultimately we will need a migration. Not this release, there isn't time, I'll just have to do some alias deletions for 22.03 and test to confirm that e.g. U-boot is happy without these aliases.

However, it sounds like `CONFIG_PPC_PCI_BUS_NUM_DOMAIN_DEPENDENT` will be enabled in the future, which will change these addresses again (in the interest of keeping the BDF numbers consistent with what older kernels assigned). Once 22.03 is out, I'm guessing mpc85xx can be bumped to 5.15, and we can focus on making sure these patches are in OpenWrt (or stable upstream) and commit from there.

closed 2022-09-02 by jow-